lopsided98/nix-ros-overlay #3: use packages without nix-shell

lelongg, 2019-11-08:
I don't know how to use ROS packages provided by this overlay without relying on nix-shell. An example on how to use these packages when installed with nix-env would be great if such a thing is possible.

lopsided98, 2019-11-25:
Sorry for not responding, but right now this overlay is not designed to work with `nix-env`. ROS relies heavily on environment variables which are conifgured using setup hooks. The best way to make this work in `nix-env` would probably be to have the ROS `buildEnv` function wrap each binary to perform the necessary setup. This will require some more thought and experimentation, but I'll work on it when I have some time.

breakds, 2020-03-04:
Probably not the best place to ask questions ... a sub topic about this, e.g. in kinetic, I think it would be convenient for people to run binaries such as `roscore` (from ros package `roslaunch`) without having to be in the nix-shell environment. One of the use case is to launch a set of ros applications as systemd services.

I found that unlike other packages built with `buildPythonPackage`, `PYTHONPATH` does not seem to be propagated. This prevents the above use case though. I do not know nix well enough to solve it - do you have any idea?

Thanks a lot!

lopsided98, 2020-03-04:
The Python scripts in ROS packages are not wrapped (except ROS 2 packages that use `ament_python`, because those are built with `buildPythonPackage`), so they don't have `PYTHONPATH` set when run outside of `nix-shell`. Almost all ROS binaries also need `ROS_PACKAGE_PATH` to be set to work correctly.

I have experimented with wrapping ROS binaries, but it still needs more work and it will only work for the most basic use cases. Any package that has a dynamic set of dependencies (such as roslaunch, rosrun and rostopic) will not work correctly without something like what I described in my previous comment.

breakds, 2020-03-05:
I see. Thanks for the explanation - I mentioned kinetic because I found that ROS 2 ament_python packages are take care of in `buildRosPackage`, but ROS 1 ones are not.

So what we need to do here is to detect automatically (or set manually) the set of binaries that needs those environment variables, and provide a way to propagate such information via dependency, and use those information to wrap the binaries. I assume `buildPythonPackage` is doing such a thing.

Or maybe we can patch all the python ROS package so that `buildPythonPackage` works for them as well? Still I am not confident of the solution because I am not that familiar with Nix yet.

lopsided98, 2020-03-05:
In ROS 1, all packages, including Python ones, use catkin (CMake) so `buildPythonPackage` won't work. 

I pushed my wrapping experiment to a new branch: https://github.com/lopsided98/nix-ros-overlay/tree/catkin-wrap. It doesn't really work right yet, but it is enough to run roscore outside of a shell. Only Python scripts are wrapped, sometimes things get wrapped twice and too many things get added to `ROS_PACKAGE_PATH`.

As I said before, I don't believe this is the correct long term solution because it cannot work for things like `roslaunch`.

breakds, 2020-03-05:
Got it. Thanks for the work on that branch - I can definitely learn from how you did that (as well as from reading other part of this repo, really appreciate it!).

To me the most important feature would be running `roscore` - we have developed in-house tools to replace all other ROS commands already.

lopsided98, 2020-09-16:
The ROS specific `buildEnv` function now wraps all ROS packages, setting `PATH`, `CMAKE_PREFIX_PATH` and `ROS_PACKAGE_PATH`. This only works right for ROS 1.

I also added some NixOS modules for deploying nodes and launch files. The following config will enable `roscore` from ROS noetic as a systemd service:
```
services.ros = {
  enable = true;
  distro = "noetic";
};
```

closed 2020-09-16 by lopsided98