lopsided98/nix-ros-overlay #12: urdfdom and its headers missing for ros 1 distros

breakds, 2020-01-13:
Trying to package rviz (kinetic) and saw this repo. Thanks a lot for making it - learned a lot.
One of the dependencies for rviz is urdfdom, which seems no longer maintained by ROS any more (ROS recommends lib-urdfdom-dev which only lives in Debian world).
urdfdom is not in kinetic (amd melodic I think) in this repo, so that rviz cannot be successfully build. Can be reproduced by

`nix repl` with `:l default.nix`, and run `rosPackages.kinetic.rviz`

Not a request to fix, just FYI.

lopsided98, 2020-01-13:
`urdfdom` is packaged in my [nixpkgs fork](https://github.com/lopsided98/nixpkgs/commits/nix-ros) (see https://github.com/lopsided98/nixpkgs/commit/63bd9b44b5e96b31f3bcfd449cfcdbfe9cafb92a for the commit that adds `urdfdom`), which is currently required to use this overlay. I am slowly working on upstreaming everything.

closed 2020-01-13 by lopsided98

lopsided98, 2020-01-13:
I made `default.nix` automatically use my fork by default.

breakds, 2020-01-14:
Thanks for the explanation!