lopsided98/nix-ros-overlay #25: Missing packages

ipa-mtb, 2020-04-14:
Hi,
first of all thanks for your work!
I tried to install some packages and it seems like some dependencies are missing, no they are mentioned in the default.nix as argument but do not exist yet. E.g. `cob-obstacle-distance`.
I'm eager to help adding more packages, but would like to know if you already have some tooling to generate nix-files from the ``package.xml`` and if so what's the problem that lead to these missing packages and maybe general remarks for adding more ros packages here.

lopsided98, 2020-04-14:
The packages in this repo are generated with my [fork of superflore](https://github.com/lopsided98/superflore). All published ROS packages that have all their non-ROS dependencies available in nixpkgs or this overlay are generated. The CI automatically updates the packages once a week.

When a ROS package fails to evaluate because one of its dependencies is missing, that means the missing dependency failed to generate because superflore couldn't resolve the rosdep key for one of its non-ROS dependencies, which either means the rosdep mapping needs to be added, or there is no Nix package for the non-ROS package.

In the case of `cob-obstacle-distance`, the missing rosdep key is `libfcl-dev`. FCL is not packaged in nixpkgs or this overlay.

jethrokuan, 2020-09-21:
Bumped into this as well, `moveit-core` is missing, likely also because `libfcl-dev` is missing from nixpkgs and this overlay.

Would it be fruitful/possible to generate the derivation anyway, but mark it as broken with a comment in the derivation as to why (e.g. a particular dep is missing). That we we know how to go about fixing the missing ROS package.

wentasah, 2026-08-29:
This is related to Noetic, which is EOL. Closing.

closed 2026-08-29 by wentasah