lopsided98/nix-ros-overlay #26: Cannot build ROS

rgrunbla, 2020-04-27:
Hi there,
I'm trying to build ROS using your project, but it fails. Here are the error messages I get:

```
Scanning dependencies of target WheelTrackedVehiclePlugin
[ 96%] Built target LookAtDemoPlugin_autogen
Scanning dependencies of target LedPlugin
[ 96%] Building CXX object plugins/CMakeFiles/WheelTrackedVehiclePlugin.dir/WheelTrackedVehiclePlugin_autogen/mocs_compilation.cpp.o
[ 96%] Building CXX object plugins/CMakeFiles/WheelTrackedVehiclePlugin.dir/WheelTrackedVehiclePlugin.cc.o
[ 96%] Building CXX object plugins/CMakeFiles/LedPlugin.dir/LedPlugin_autogen/mocs_compilation.cpp.o
[ 96%] Building CXX object plugins/CMakeFiles/LedPlugin.dir/LedPlugin.cc.o
[ 96%] Linking CXX shared library libLinkPlot3DPlugin.so
[ 96%] Linking CXX shared library libCameraPlugin.so
[ 96%] Built target LinkPlot3DPlugin
Scanning dependencies of target SimpleTrackedVehiclePlugin
[ 96%] Building CXX object plugins/CMakeFiles/SimpleTrackedVehiclePlugin.dir/SimpleTrackedVehiclePlugin_autogen/mocs_compilation.cpp.o
[ 96%] Building CXX object plugins/CMakeFiles/SimpleTrackedVehiclePlugin.dir/SimpleTrackedVehiclePlugin.cc.o
[ 96%] Built target CameraPlugin
Scanning dependencies of target gzclient
[ 96%] Building CXX object gazebo/gui/CMakeFiles/gzclient.dir/gzclient_autogen/mocs_compilation.cpp.o
[ 96%] Building CXX object gazebo/gui/CMakeFiles/gzclient.dir/main.cc.o
[ 96%] Linking CXX executable gazebo
[ 96%] Linking CXX shared library libRayPlugin.so
[ 96%] Linking CXX shared library libMisalignmentPlugin.so
/nix/store/3b3ighb83nhifa1v4n7855hlbdl1mhf9-binutils-2.31.1/bin/ld: gui/libgazebo_gui.so.9.11.0: undefined reference to `gazebo::gui::TerrainEditorPalette::staticMetaObject'
/nix/store/3b3ighb83nhifa1v4n7855hlbdl1mhf9-binutils-2.31.1/bin/ld: gui/libgazebo_gui.so.9.11.0: undefined reference to `vtable for gazebo::gui::IncrementalPlot'
/nix/store/3b3ighb83nhifa1v4n7855hlbdl1mhf9-binutils-2.31.1/bin/ld: gui/libgazebo_gui.so.9.11.0: undefined reference to `gazebo::gui::TimePanel::SetTimeWidgetVisible(bool)'
/nix/store/3b3ighb83nhifa1v4n7855hlbdl1mhf9-binutils-2.31.1/bin/ld: gui/libgazebo_gui.so.9.11.0: undefined reference to `gazebo::gui::EditorItem::qt_metacast(char const*)'
/nix/store/3b3ighb83nhifa1v4n7855hlbdl1mhf9-binutils-2.31.1/bin/ld: gui/libgazebo_gui.so.9.11.0: undefined reference to `vtable for gazebo::gui::InsertModelWidget'
/nix/store/3b3ighb83nhifa1v4n7855hlbdl1mhf9-binutils-2.31.1/bin/ld: gui/libgazebo_gui.so.9.11.0: undefined reference to `gazebo::gui::TopicView::staticMetaObject'
/nix/store/3b3ighb83nhifa1v4n7855hlbdl1mhf9-binutils-2.31.1/bin/ld: gui/libgazebo_gui.so.9.11.0: undefined reference to `vtable for gazebo::gui::ModelRightMenu'
[…]
collect2: error: ld returned 1 exit status
make[2]: *** [gazebo/CMakeFiles/gazebo.dir/build.make:148: gazebo/gazebo-9.11.0] Error 1
make[1]: *** [CMakeFiles/Makefile2:2548: gazebo/CMakeFiles/gazebo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

I'm kind of new to this whole Nix thing, so it may be I'm doing something wrong. Here is the command I used:

```
nix-shell \                                                                                                                                                                                                                                                                                                                                                                                                  -I nix-ros-overlay=https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz \
  --option extra-substituters 'https://ros.cachix.org' \
  --option trusted-public-keys 'cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY= ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' \
  '<nix-ros-overlay/examples/turtlebot3-gazebo.nix>'
warning: ignoring untrusted substituter 'https://ros.cachix.org'
```

I'm using Archlinux.

lopsided98, 2020-04-27:
The latest update to CMake broke Gazebo. It has been fixed upstream, but I haven't gotten a chance to update it yet.

lopsided98, 2020-04-27:
This should be fixed now. Make sure it builds Gazebo 9.13.0. If it is still building 9.11.0, you may need to pass `--option tarball-ttl 0` to force it pull the latest version of my nixpkgs fork.

closed 2020-04-27 by lopsided98