lopsided98/nix-ros-overlay #37: integration with default.nix or shell.nix

bresilla, 2020-07-04:
I am very sorry, this may sound very naive question (and probably is because i still don't know nix very good), but i am trying since few days and i cant figure it out, so i am bothering you here.

I am trying to use this overlay but i want it to use in a default.nix or shell.nix file. and not to type always the command you have given.

Because i heavily use [direnv](https://github.com/direnv/direnv.vim), which when i enter a specific folder that has `shell.nix` file, it automatically invokes nix-shell. 

I was thinking if you could explain how we/me can use your overlay.

What i am doing now, but very unsuccessfully is that, i modified `~/.config/nixpkgs/config.nix` to
```
{
  packageOverrides = pkgs: {
    nur = import (builtins.fetchTarball "https://github.com/nix-community/NUR/archive/master.tar.gz") {
      inherit pkgs;
    };
    ros = import (builtins.fetchTarball "https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz") {
      inherit pkgs;
    };
  };

  nix.binaryCaches = [
    https://cache.nixos.org
    https://aira.cachix.org
    https://ros.cachix.org
  ];

  nix.binaryCachePublicKeys = [
    "cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY="
    "aira.cachix.org-1:/5nHPqhVrtvt7KCk04I8cH/jETANk8BtPHWsEtcwU/M="
    "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo="
  ];
}
```

and i tried this in `shell.nix` file but does not work:
```
with import <nixpkgs> {
  config = {};
  overlays = [
    (import https://github.com/lopsided98/nix-ros-overlay)
  ];
};
stdenv.mkDerivation rec {
  name = "ros";
  buildInputs = [
    stdenv
    pkg-config
    gazebo
  ];
}
```

Thank you in advance.
Feel free to immediately close this issue if you think that is not the place here to ask question, and i am sorry in advance. I really hope to use this overlay. Its amazing.

lopsided98, 2020-07-04:
To use this repo as an overlay, you need to import `overlay.nix`. Alternatively, you can import `default.nix` (not as an overlay), which provides the correct version of nixpkgs with the overlay already applied. You currently need to use my nixpkgs fork because some things haven't been upstreamed. Also, take a look at the example: https://github.com/lopsided98/nix-ros-overlay/blob/master/examples/turtlebot3-gazebo.nix

In your case, this should work, although unfortunately Gazebo 10 does not build right now. Hopefully I can get that fixed soon.
```
with import (builtins.fetchTarball "https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz") {};
with rosPackages.melodic; # change as desired
stdenv.mkDerivation {
  name = "ros";
  buildInputs = [
    pkg-config
    gazebo
  ];
}
```

bresilla, 2020-07-04:
This is awesome. Thanks a lot for all the amazing work.

closed 2020-07-04 by bresilla