lopsided98/nix-ros-overlay #42: SETUPTOOLS_DEB_LAYOUT is not set

akru, 2020-07-31:
**Tracelog**

https://github.com/airalab/robonomics/runs/928312139?check_suite_focus=true#step:4:1373

```
/store/8iw49kdb964892r133j1gd3ys9cbdq5q-ros-noetic-catkin-0.8.6-r1/bin
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --install-layout not recognized
CMake Error at catkin_generated/safe_execute_install.cmake:4 (message):

  execute_process(/tmp/nix-build-ros-noetic-catkin-0.8.6-r1.drv-0/catkin-release-release-noetic-catkin-0.8.6-1/build/catkin_generated/python_distutils_install.sh)
  returned error code
Call Stack (most recent call first):
  cmake_install.cmake:171 (include)
```

Unfortunately, during the build I got these errors.

https://answers.ros.org/question/52021/installing-groovy-from-source-fails/

Could you set `-DSETUPTOOLS_DEB_LAYOUT=OFF` to cmake sentence to prevent it in future?

lopsided98, 2020-07-31:
This is the same bug as #1. IMO, the real fix is to enable sandboxing, but I can add that flag to make it work without sandboxing.

akru, 2020-07-31:
@lopsided98 thank you, sandboxing works for me. No problem.

closed 2020-07-31 by akru