lopsided98/nix-ros-overlay #45: ROS2 example

lelongg, 2020-08-20:
Could someone provide an example on how to use this overlay with ROS2 please ?

lopsided98, 2020-09-16:
ROS2 support is pretty rough right now, mostly because I don't actually use it. I just tried a simple shell for foxy:
```
with import ../. {};
with rosPackages.foxy;

mkShell {
  buildInputs = [
    ros2topic
    ros2node
  ];
}
```
but was unable to get even `ros2 topic list` to work. It gives the following error:
```
>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'type support not from this implementation, at /build/rmw_fastrtps-release-release-foxy-rmw_fastrtps_cpp-1.2.0-1/src/publisher.cpp:75, at /build/rcl-release-release-foxy-rcl-1.1.7-1/src/rcl/node.c:276'

with this new error message:

  'rcl node's rmw handle is invalid, at /build/rcl-release-release-foxy-rcl-1.1.7-1/src/rcl/node.c:428'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1600297455.820599941] [rcl]: Failed to fini publisher for node: 1
Unknown error creating node: rcl node's rmw handle is invalid, at /build/rcl-release-release-foxy-rcl-1.1.7-1/src/rcl/node.c:428
```

I don't know enough about ROS2 to fix this easily, but I think it might be due to a missing dependency somewhere. `eloquent` seems to work though, so maybe try that?

What specific problem did you run into?

alexhenning, 2021-03-04:
Adding ` RMW_IMPLEMENTATION = "rmw_fastrtps_dynamic_cpp";` to the shell and `rmw-fastrtps-dynamic-cpp` to the list of packages seems to resolve the issue for me.

From https://github.com/ros2/rmw_fastrtps#two-different-rmw-implementations
> rmw_fastrtps actually provides not one but two different ROS 2 middleware implementations, both of them using Fast DDS as middleware layer: rmw_fastrtps_cpp and rmw_fastrtps_dynamic_cpp (note that directory rmw_fastrtps_shared_cpp just contains the code that the two implementations share, and does not constitute a layer on its own).

With a local version with ifconfig fixed, the following lets me run the turtlsim example with foxy. Note, it's using niv to manage the overlay and using a local nix-ros-overlay that has python3-ifconfig installed to fix ros2doctor. That still needs to make its way into  `nixpkgs` for that to work with this repo.
```nix
{ sources ? import ./nix/sources.nix }:     # import the sources
let
  overlay = _: pkgs: {
    niv = import sources.niv {};    # use the sources :)
  };
  pkgs = import sources.nixpkgs {};
  nix-ros-overlay = import sources.nix-ros-overlay {};
  ros = nix-ros-overlay.rosPackages.foxy;
in
pkgs.mkShell {
  buildInputs = with ros; [
    pkgs.glibcLocales
    (buildEnv {
      paths = [
        ros-base
        rmw-fastrtps-dynamic-cpp
        turtlesim
        rviz2
      ];
    })
  ];

  # HACK: This is necessary right now as the equivalent of source /opt/ros/foxy/setup.sh
  shellHook = ''
    source $(echo $buildInputs | sed 's/ /\n/g' | tail -n1)/share/*/local_setup.sh
  '';
  ROS_VERSION = 2;
  ROS_PYTHON_VERSION = 3;
  ROS_DISTRO = "foxy";

  # This seems to be necessary to dynamically parse the types for (de)serialization
  RMW_IMPLEMENTATION = "rmw_fastrtps_dynamic_cpp";
}
```

lopsided98, 2021-03-18:
Most of those hacks in your shell shouldn't be needed now, as I have fixed the environment setup when using `buildEnv`. Unfortunately, `buildEnv` is mostly useless for ROS 2 right now, as the individual package paths get added to `AMENT_PREFIX_PATH`, rather than just the single path to the environment with all of them combined.

This happens because I source the ament generated `local_setup.sh` for each package, which doesn't know about the environment generated by `buildEnv`. I need to figure out how to generate a `setup.sh` for the environment, ideally using the same mechanism `colcon` does, but I haven't even been able to find the code responsible for it in colcon/ament.

I added a very simple example so show how you need to set `RMW_IMPLEMENTATION = "rmw_fastrtps_dynamic_cpp";`. I haven't been able to figure out yet why `rmw_fastrtps_cpp` doesn't work.

acowley, 2021-07-02:
A basic flake version of the wisdom shared here is,

```nix
{
  description = "Testing ROS on NixOS";
  inputs = {
    flake-utils.url = "github:numtide/flake-utils";
    ros-flake.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, flake-utils, ros-flake }: 
    flake-utils.lib.eachDefaultSystem (system:
      let pkgs = nixpkgs.legacyPackages.${system}; 
          ros = ros-flake.packages.${system}.foxy;
      in
      {
        devShell = pkgs.mkShell {
          buildInputs = [ ros.turtlesim
                          ros.ros2run
                          ros.rmw-fastrtps-dynamic-cpp ];
          RMW_IMPLEMENTATION = "rmw_fastrtps_dynamic_cpp";
        };
      }
    );
}
```

Given that `flake.nix`, one can run `nix develop` and then `ros2 run turtlesim turtlesim_node` and `ros2 run turtlesim turtle_teleop_key`.

A hearty thanks to everyone who has worked on this, particularly @lopsided98. This is such a huge leap forward!

closed 2021-07-20 by lopsided98