lopsided98/nix-ros-overlay #46: Practical usage example

jethrokuan, 2020-09-16:
Hi, thank you for attempting to package ROS with Nix. I'm struggling a little to wrap my head around how to use this in an actual ROS catkin workspace. I've looked at the example, and it does a local import of the `default.nix` and `overlays.nix` file.

My experience with Nix is limited, but the usage of the Emacs overlay is simple:

```
{
  nixpkgs.overlays = [
    (import (builtins.fetchTarball {
      url = https://github.com/nix-community/emacs-overlay/archive/master.tar.gz;
    }))
  ];
}
```

I was hoping that the usage of the this overlay would be similar, which would then expose `rosPackages` under `pkgs`, but I couldn't get this to work. Do you have a sample catkin workspace using this overlay that I could look at? Do you also develop work on ROS packages and build them with Nix?

lopsided98, 2020-09-16:
You should be able to do something like this (not tested):
```
{
  nixpkgs.overlays = [
    (import (builtins.fetchTarball {
      url = "https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz";
    }) + "/overlay.nix")
  ];
}
```

`default.nix` provides a pinned and sometimes patched (while I am waiting for a PR to be merged) version of nixpkgs with the ROS overlay applied. The packages that are uploaded to Cachix are built against this nixpkgs version, so you should use it if you don't want to build things yourself.

I only test against unstable, so you might run into problems if you use a stable version of nixpkgs.

My normal workflow is to create a `shell.nix` that imports this repo:
```
# You could use fetchTarball
{ pkgs ? import ../../../../Projects/ros/nix-ros-overlay { } }:
with pkgs;
with rosPackages.noetic;

mkShell {
  name = "ros-example";

  buildInputs = [
    (buildEnv { paths = [
      catkin
      rosbash
      roslaunch
      rospy
      rosnode
      mavros
      mavros-extras
      angles
      flann
      eigen
      geos
      tf2-geometry-msgs
      tf2-tools
      robot-state-publisher
      actionlib-tools

      mapviz
      mapviz-plugins
      tile-map

      rqt-reconfigure
    ]; })
  ];

  shellHook = ''
    export ROS_HOSTNAME=localhost
    export ROS_MASTER_URI=http://localhost:11311/
  '';
}
```
Then you can use it pretty much like a normal ROS installation. One thing to note is that you will need to delete the `build` and `devel` directories in the catkin workspace every time you update/change the shell environment.

For deployment, I use the NixOS modules in the [`modules/` directory](https://github.com/lopsided98/nix-ros-overlay/tree/master/modules).

jethrokuan, 2020-09-17:
I think I've gotten most things running smoothly, just a few snags.

I have a `shell.nix` that with the current master branch of this repo avaliable in my nix-path as `<nix-ros>`. The shell.nix looks like:

```
with import <nix-ros> {};
with rosPackages.noetic;
with pythonPackages;

mkShell {
  buildInputs = [
    glibcLocales
    (buildEnv { paths = [
                  apriltag
    ]; })
  ];
}
```

With cachix, I'm able to pull down a pre-built package, and this works. However, when I change the ROS distribution:

```
with import <nix-ros> {};
with rosPackages.melodicPython3;
with pythonPackages;

mkShell {
  buildInputs = [
    glibcLocales
    (buildEnv { paths = [
                  apriltag
    ]; })
  ];
}
```

Nix attempts to build the package, and it fails towards the end:

```
Scanning dependencies of target apriltag_demo
[ 96%] Building C object CMakeFiles/apriltag_demo.dir/example/apriltag_demo.c.o
[100%] Linking C executable apriltag_demo
[100%] Built target apriltag_demo
installing
install flags: SHELL=/nix/store/6ffp6xbr6j2ryqw2dafbbvv55ggq16iv-bash-4.4-p23/bin/bash install
[ 93%] Built target apriltag
[100%] Built target apriltag_demo
Install the project...
-- Install configuration: "Release"
CMake Error at cmake_install.cmake:65 (file):
  file INSTALL cannot find
  "/nix/store/476rrbsfxzhzgqkqh4yz0151p6xg23hz-ros-melodic-apriltag-3.1.1-r1/lib/libapriltag.so.3.1.0":
  No such file or directory.

make: *** [Makefile:127: install] Error 1
builder for '/nix/store/aic5a0d9p9x40lav25saf985zayi0w7j-ros-melodic-apriltag-3.1.1-r1.drv' failed with exit code 2
cannot build derivation '/nix/store/xgk9nan7my1xn9y9c93k67zamw2bwil2-ros-env.drv': 1 dependencies couldn't be built
error: build of '/nix/store/xgk9nan7my1xn9y9c93k67zamw2bwil2-ros-env.drv' failed
```

I've noticed this pattern of cmake being unable to find dependencies when trying to build several other libraries as well (franka-control in particular). Am I missing something?

lopsided98, 2020-09-17:
You happened to run into packages that are broken right now. Unfortunately, there are too many packages for me to make sure all of them build, so I end up dealing with them only when I or someone else needs them.

In the case of apriltag, it fails to build due to a common CMake mistake, described here: https://github.com/jtojnar/cmake-snips#assuming-cmake_install_dir-is-relative-path. It works for noetic because the [`apriltag-release` repo has a patch](https://github.com/AprilRobotics/apriltag-release/blob/patches/release/noetic/apriltag/0001-Set-CMAKE_INSTALL_LIBDIR-to-lib.patch) that works around it, because apparently it has become a problem on Ubuntu as well. The real fix is to submit a patch upstream (which I can then apply in this repo).

In the case of franka, that is a bug in the Nix packaging of poco. See https://github.com/NixOS/nixpkgs/pull/98186 for a fix. This kind of problem pops up a lot in this repo because the missing deps usually get manually added to dependent packages in nixpkgs, but don't get automatically added to superflore generated packages.

jethrokuan, 2020-09-18:
Thank you for the prompt and informative replies. I like the experience using this overlay much more than messing around with system deps on mixed versions of Ubuntu (my lab's PC is shared between many users). So as much as possible, I'd like to get the packages I'm using working, and I'm happy to provide as much support as I can.

For now, I'll take a look at the Github CI logs to see which packages fail to build, and how they can be remedied!

closed 2020-09-18 by jethrokuan

lopsided98, 2020-09-20:
I submitted an upstream PR to fix apriltag (https://github.com/AprilRobotics/apriltag/pull/106) and added the patch to this repo.

jethrokuan, 2020-09-21:
Thanks! I've had the apriltag noetic patch backported to melodic and they cut a new release, so that melodic should build fine too without the patch.

One last question. I want to add an overlay on top of `nix-ros-overlay`, to pin versions of some packages (e.g. when the robot firmware is not up-to-date). So I have in my shell.nix for example:

```
{ nix-ros ? import (builtins.fetchTarball {
  name = "nix-ros-overlay";
  url = "https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz";
})}:
let pkgs = nix-ros {
  overlays = [(import ./overlay.nix)];
};
in
with pkgs;
with rosPackages.melodicPython3;
mkShell {
  buildInputs = [
    glibcLocales
    (buildEnv { paths = [
                  apriltag
                  franka-hw
    ]; })
  ];
}
```

how should overlay.nix look like? I tried something like:

```
self: super: {
  rosPackages.melodicPython3.franka-hw = super.rosPackages.melodicPython3.franka-hw.overrideAttrs(oldAttrs: {
    src = "../src/franka-ros/franka-hw";
  });
}
```

but that gives me the error:

```
error: anonymous function at /nix/store/8xyd8mayg7k8y0radyq68sdwgg106yrh-source/pkgs/build-support/buildenv/default.nix:8:2 called without required argument 'name', at /nix/store/8xyd8mayg7k8y0radyq68sdwgg106yrh-source/lib/customisation.nix:69:16
```

lopsided98, 2020-09-21:
I don't think that error has anything to do with the overlay; `buildEnv` just needs a `name` attribute.

On the other hand, I don't think that method of overriding the ROS package set will work (since overlays aren't recursively merged). The ROS package set uses makeExtensible, so you should be able to create an overlay like this:
```
self: super: {
  rosPackages.melodicPython3 = super.rosPackages.melodicPython3.extend (rosSelf: rosSuper: {
    franka-hw = rosSuper.franka-hw.overrideAttrs (oldAttrs: {
      src = "../src/franka-ros/franka-hw";
    });
  });
}
```
I haven't tested this exact example, but I have successfully used the `extend` function [in the NixOS module](https://github.com/lopsided98/nix-ros-overlay/blob/master/modules/ros.nix#L86)

jethrokuan, 2020-09-21:
That works with a small tweak to distro-overlay.nix, but is probably not the right way.

https://gist.github.com/jethrokuan/ccc49403bc61879ceec9289a654eac47

It otherwise complains that the attribute "lib" is missing.

lopsided98, 2020-09-21:
This will make it work without patching:
```
self: super: {
  rosPackages = super.rosPackages // {
    melodicPython3 = super.rosPackages.melodicPython3.extend (rosSelf: rosSuper: {
      franka-hw = rosSuper.franka-hw.overrideAttrs (oldAttrs: {
        src = "../src/franka-ros/franka-hw";
      });
    });
  };
}
```
That problem is also caused by overlays not being recursively merged.