lopsided98/nix-ros-overlay #54: Shell script to setup nix-ros-overlay

flix59, 2020-10-29:
I would like to use this awesome repo (https://github.com/lopsided98/nix-ros-overlay) to develop with ros. The example command

```
nix-shell
-I nix-ros-overlay=https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz
–option extra-substituters ‘https://ros.cachix.org’
–option trusted-public-keys ‘cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY= ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=’
‘<nix-ros-overlay/examples/turtlebot3-gazebo.nix>’
```

works fine. But I need to add an environment variable to the shell and other packages I really would like it to put all of this in an own default.nix that I can just call nix-shell. This is what I came up with so far(props to jonringer):

```
let
ros = import (builtins.fetchTarball https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz) { };
in
with ros.pkgs;
stdenv.mkDerivation {
    name = "ros_overlay_shell";
    PATH_VARIABLE = "path/to/something";
}

```

the nix-shell can start with that script but it didn't include the needed packages, for example, roslaunch couldn't be found.
Is there a way to include the packages in the shell script?
Other approaches I tried are in this discussion in the nix forum (https://discourse.nixos.org/t/nixshell-for-overlay/9728/4)

lopsided98, 2020-10-30:
Have you looked at the source for the turtlebot example: https://github.com/lopsided98/nix-ros-overlay/blob/master/examples/turtlebot3-gazebo.nix?

That shows how to add packages to a shell and set environment variables. You just need to change it to import from `fetchTarball`.

flix59, 2020-10-30:
ok yeah I'm stupid

closed 2020-10-30 by flix59