lopsided98/nix-ros-overlay #62: librealsense cannot be evaluated when `with rosPacakges.melodic` is present

breakds, 2020-12-17:
I am building a `nix-shell` with `nix-ros-overlay`, which needs to have `librealsense` as well. When I added `librealsense` directly in 

```nix
with import (builtins.fetchTarball {
  url = "https://github.com/lopsided98/nix-ros-overlay/archive/19c206638d61d0c8c9466e1ea26908472f79d1ae.tar.gz";
}) {
  nixpkgs = builtins.fetchTarball "https://github.com/NixOS/nixpkgs/archive/66acfa3d16eb599f5aa85bda153a24742f683383.tar.gz";
}; 

with rosPackages.melodic;

mkShell {
  buildInputs = [ librealsense ];
}
```

The evaluation will fail with 

```
error: attribute 'librealsense' missing, at /nix/store/959h823yjbr6dhb5aikdaivcfmr22cph-source/distros/distro-overlay.nix:143:20
(use '--show-trace' to show detailed location information)
```

My best guess is that `rosSuper.librealsense` does not evaluate, but I am not sure how to keep debugging. Instead, I am using this workaround 

```nix
with import (builtins.fetchTarball {
  url = "https://github.com/lopsided98/nix-ros-overlay/archive/19c206638d61d0c8c9466e1ea26908472f79d1ae.tar.gz";
}) {
  nixpkgs = builtins.fetchTarball "https://github.com/NixOS/nixpkgs/archive/66acfa3d16eb599f5aa85bda153a24742f683383.tar.gz";
}; 

let librealsense-unpatched = librealsense;

in with rosPackages.melodic;

mkShell {
  buildInputs = [ librealsense-unpatched ];
}
```

to extract the `librealsense` from `nixpkgs`. My understanding is that since I am pinning to the latest `nixpkgs` where `librealsense` is up to date (2.40.p), I probably do not need the patch. Can you help confirm?

Thanks a lot!

lopsided98, 2020-12-18:
The ROS `librealsense` package is only published for `kinetic`, see: http://wiki.ros.org/librealsense. That is version 1.12.1.

Newer ROS distros have a `librealsense2` package, which is probably what you want: http://wiki.ros.org/librealsense2. It was missing some dependencies in `package.xml`, but I added an override to add them so it builds now.

lopsided98, 2020-12-18:
Just for clarification, there are two different unrelated `librealsense` attributes: one from nixpkgs and one in `rosPackages.<distro>`.   The `with rosPackages.melodic;` statement causes the latter to shadow the former. The override that adds the patch causes the `rosPackages.<distro>.librealsense` attribute to exist for all distros, even though it only successfully evaluates for `kinetic`

breakds, 2020-12-19:
> Just for clarification, there are two different unrelated librealsense attributes: one from nixpkgs and one in rosPackages.<distro>. 

Thanks a lot for the explanation! Now I get it :) I originally thought those are the same one, didn't realize the latter is a "ros package". Will close this issue.

Appreciate the prompt reply!

closed 2020-12-19 by breakds