lopsided98/nix-ros-overlay #64: example how to add the overlay to /etc/nixos/configuration.nix

mgrrx, 2020-12-19:
maybe as a disclaimer: I'm new to nixos. But I've setup a nixos VM and so far I could help myself to setup everything I need.

But since a few days I'm completely stuck trying to add this repo globally as an overlay and installing command line tools like rostopic system wide. It would be great if you could add an example how to setup nixos with your overlay.

3noch, 2020-12-22:
I think what you want is to add `nixpkgs.overlays = [ (import this-repo {}) ];` in your /etc/nixos/configuration.nix (or a module that you import from there).

mgrrx, 2020-12-23:
I've been there already but this breaks with
```
The option value `nixpkgs.overlays.[definition 1-entry 1]' in `/etc/nixos/configuration.nix' is not of type `nixpkgs overlay'.
```
when calling nixos-rebuild

lopsided98, 2020-12-23:
@3noch's example looks right except you need to import `overlay.nix` instead of `default.nix` (which `import nix-ros-overlay` does implicitly). I think replacing that line with `(import (nix-ros-overlay + "/overlay.nix"))` should work.

mgrrx, 2020-12-25:
Thanks, this worked! Can you maybe add it to your README?

lopsided98, 2021-05-24:
I added documentation for this: https://github.com/lopsided98/nix-ros-overlay/blob/staging/docs/nixos.md.

closed 2021-05-24 by lopsided98