lopsided98/nix-ros-overlay #70: Using catkin_make outside of nix-shell

xneomac, 2021-01-08:
I have a following issue concerning the issue #3 and the features brought by the feature of the `buildEnv`. I am building a ros environment with this function, installing it with `nix-env -i` and trying to use `catkin_make` to build ros packages. Building packages works in nix-shell but fail with the installed version because of missing dependencies.

With the installed version running `catkin_make --use-ninja`

```
Base path: /kalifa/releases/newton_fake-latest/sources/src/ros_ws
Source space: /kalifa/releases/newton_fake-latest/sources/src/ros_ws/src
Build space: /kalifa/releases/newton_fake-latest/sources/src/ros_ws/build
Devel space: /kalifa/releases/newton_fake-latest/sources/src/ros_ws/devel
Install space: /kalifa/releases/newton_fake-latest/sources/src/ros_ws/install
####
#### Running command: "cmake /kalifa/releases/newton_fake-latest/sources/src/ros_ws/src -DCATKIN_DEVEL_PREFIX=/kalifa/releases/newton_fake-latest/sources/src/ros_ws/devel -DCMAKE_INSTALL_PREFIX=/kalifa/releases/newton_fake-latest/sources/src/ros_ws/install -G Ninja" in "/kalifa/releases/newton_fake-latest/sources/src/ros_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /kalifa/releases/newton_fake-latest/sources/src/ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /nix/store/n9581zn5dvq6425nkhfixk0hzvn0aqsb-ros-env
-- This workspace overlays: /nix/store/n9581zn5dvq6425nkhfixk0hzvn0aqsb-ros-env
-- Using PYTHON_EXECUTABLE: /nix/store/famvf5klpwbqvh1fal88hvkgrhfi3nbj-python-2.7.18/bin/python
-- Using default Python package layout
-- Could NOT find PY_em (missing: PY_EM) 
CMake Error at /nix/store/n9581zn5dvq6425nkhfixk0hzvn0aqsb-ros-env/share/catkin/cmake/empy.cmake:30 (message):
  Unable to find either executable 'empy' or Python module 'em'...  try
  installing the package 'python-empy'
Call Stack (most recent call first):
  /nix/store/n9581zn5dvq6425nkhfixk0hzvn0aqsb-ros-env/share/catkin/cmake/all.cmake:164 (include)
  /nix/store/n9581zn5dvq6425nkhfixk0hzvn0aqsb-ros-env/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:58 (find_package)

-- Configuring incomplete, errors occurred!
See also "/kalifa/releases/newton_fake-latest/sources/src/ros_ws/build/CMakeFiles/CMakeOutput.log".
Invoking "cmake" failed
```

With nix-shell running `catkin_make --use-ninja`
```
Base path: /home/xneomac/Developer/sultan2/src/ros_ws
Source space: /home/xneomac/Developer/sultan2/src/ros_ws/src
Build space: /home/xneomac/Developer/sultan2/src/ros_ws/build
Devel space: /home/xneomac/Developer/sultan2/src/ros_ws/devel
Install space: /home/xneomac/Developer/sultan2/src/ros_ws/install
####
#### Running command: "ninja build.ninja" in "/home/xneomac/Developer/sultan2/src/ros_ws/build"
####
[0/1] Re-running CMake...
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Using CATKIN_DEVEL_PREFIX: /home/xneomac/Developer/sultan2/src/ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /nix/store/n9581zn5dvq6425nkhfixk0hzvn0aqsb-ros-env;/nix/store/cigh5dfwc0cslx7myygdqy5hjn8kaz1j-ignition-cmake-0.6.1;/nix/store/y03kn18w1jl52w7hf081ih89brk00qky-pkg-config-wrapper-0.29.2;/nix/store/icm74ybhq3zv5w01gzrgadlgwfwy5pap-lndir-1.0.3;/nix/store/p7xxgw0xak0pwah1qma6p6jnrc546s08-patchelf-0.12;/nix/store/lfl1davviy5mai3g80mpy2gkb97rh9j5-gcc-wrapper-9.3.0;/nix/store/14w5kfy0lpz9c5f7vfj8jbnbnmf0g78v-binutils-wrapper-2.31.1;/nix/store/c36777c29wkp4jxjv8pk3frvkdd3w89q-python2.7-autopep8-1.5.4;/nix/store/649h20dykdn6zccxn52wjyg6vr2w1nqs-python2.7-pycodestyle-2.6.0;/nix/store/famvf5klpwbqvh1fal88hvkgrhfi3nbj-python-2.7.18;/nix/store/73m791xz9kcqiyvhq7zpgmlja5hvdnpg-python2.7-toml-0.10.1;/nix/store/dm143vrm1riqng1yfhknr6dfll4jggq1-python2.7-pylint-1.9.5;/nix/store/zpk0w7l4gxaz4qs4l6j4qp8gjx36pbm6-python2.7-astroid-1.6.6;/nix/store/rz12vmh5k4r5ax5vxa7m9hsgvcw02l2n-python2.7-lazy-object-proxy-1.5.1;/nix/store/db5nranmxgvml761wk0kmw0kvdky4mph-python2.7-six-1.15.0;/nix/store/wl1rvgbqx776y4nxzh0nby7l6109zpwb-python2.7-wrapt-1.12.1;/nix/store/33jc6jxdz43rr1dq7p7h0cw1862fdb25-python2.7-enum34-1.1.10;/nix/store/dw5z1m3h4qk67q6w52l5y78cp0vlaa4d-python2.7-singledispatch-3.4.0.3;/nix/store/wxd5d74d8kbizym1b30wcczxg57958kk-python2.7-backports.functools_lru_cache-1.6.1;/nix/store/4z0hckgqmnx1nayyknhlsixrcaa4ii40-python2.7-isort-4.3.21;/nix/store/qr84kwsq698pf8ax02d487w5f3lmwzyn-python2.7-setuptools-44.0.0;/nix/store/cd6cpgdhj90lf4n7nh7fkcbx916nsk17-python2.7-futures-3.3.0;/nix/store/daik8rxcl3jpg64l0nm5klz55z7x4brb-python2.7-mccabe-0.6.1;/nix/store/yr9yz9kl0clp4m4nq0hx5j2pamd6dzwb-python2.7-configparser-4.0.2;/nix/store/2zxv9vvrxf6xidh8xgpgrchzqp6304wd-avahi-0.7;/nix/store/f84rn0bs6qq462v8ijr60y0cvsg99wgh-clang-wrapper-7.1.0;/nix/store/4fiibyqiq5y0dc1ma3far935r2sn58ak-cmake-3.18.2;/nix/store/2byra6ps0n6bpzxwqvry3k94r877rlh2-curl-7.73.0-dev;/nix/store/wc6z0ljl6jcl0h65p4c9xm19nvz4052q-nghttp2-1.41.0-dev;/nix/store/zf0f4qwl6cfw5ah3fbdmn7av1sswyf6i-nghttp2-1.41.0-bin;/nix/store/858hdp1irdixscwa8jzhs11gsj3j5a86-nghttp2-1.41.0-lib;/nix/store/k3wmwf51m3zxz8lz6pbyyya5w0bai34q-zlib-1.2.11-dev;/nix/store/ghrisqgvmz8blq4i9g0cqas1s3p0glx7-zlib-1.2.11;/nix/store/6b54aw7g7mfk52hsrlckhhi40s1vmz0b-libkrb5-1.18-dev;/nix/store/47lx978m7l4k8jraw9pl4dxskv9zc0n4-libkrb5-1.18;/nix/store/0qx3q365dnrna6pb5pvy9akcln27c5nk-openssl-1.1.1h-dev;/nix/store/6cin1mkj6q4wb9dwyvxn071gd2yzp91f-openssl-1.1.1h-bin;/nix/store/lgcr2hnww96958vnw8dymqy17x0klmwf-openssl-1.1.1h;/nix/store/w2flan0mfpjhzry58s3kdbwzx919frcw-libssh2-1.9.0-dev;/nix/store/29ivhjidchfcv03b0885qab96hv1h30s-libssh2-1.9.0;/nix/store/zgf5rl7p0g47259yh56p9008gg9msclh-curl-7.73.0-bin;/nix/store/mrkch665vqn6hs5xzhfja6nrb56wz5mi-curl-7.73.0;/nix/store/0ma9b86c5k12w53n5ps49z4rp0gip2gx-dbus-1.12.20-dev;/nix/store/c4hv8wchh7925h2vdbz1ag0bzbpwkxfw-expat-2.2.10-dev;/nix/store/cfi73kpb94c8sjv769d2cffbrsvmx4rc-expat-2.2.10;/nix/store/ab8x9lnqmbq7csk50z1livp15ibpvsq2-dbus-1.12.20-lib;/nix/store/mwb895frsfjbn0m035gg19zfvv2pvx1s-dbus-1.12.20;/nix/store/nwk8rivbjixsi9f9aghdx3940ga5px0w-glibc-locales-2.32;/nix/store/0pk3sjg92jyvh4n19sqg7wcpw252ni5v-gnused-4.8;/nix/store/xxzdxdy0vp08vcikmik9b194dhgp234a-hostname-1003.1-2008;/nix/store/cnznybpymsgkwxdlrpzfq9ghrq07s0qx-iproute2-5.9.0-dev;/nix/store/7fgxmjkb0nksf8bqwrn45mvivpzfwh58-iproute2-5.9.0;/nix/store/jf3nxgwiwv925ibp5ard9svyphnrqzrh-nginx-1.18.0;/nix/store/4qbjz29bqllw17scjkkls2k5rplz2217-opencv-3.4.8;/nix/store/g82la6r1spr1cnhq3pv22b1v0f6bn86f-openssh-8.4p1;/nix/store/qmr5rhxy942kdm70j8rs751y839zg6hb-protobuf-3.13.0;/nix/store/53gr2zfh5xvgv7n7arhbp3njbmm8is6s-python-2.7.18-env;/nix/store/7kv6l106ckpv70h6gw7y7mmsjn21kl13-python2.7-supervisor-3.3.5;/nix/store/1gifpryibycyx184blar1jkfg0d6mrmk-python2.7-meld3-2.0.1;/nix/store/izh7812h0m8m30p4lxpijs5g3xh3kam2-doctor;/nix/store/v89a8qfy20kx3hk6894yws20zz4m4lwd-boost-1.69.0-dev;/nix/store/3iccb2zby79yyf0xs6i0i0dsf0xf2yw0-boost-1.69.0;/nix/store/rf0w7ich2n7m00fzk1f4lhwj0vx908nm-cachix-0.5.1;/nix/store/zfk3j7g6gq3fgzrh9ag9jgxaa1np7ncy-cargo-watch-7.2.1;/nix/store/vpj11q9d229ljddka7hvv3jpm31jinhj-docker-19.03.13;/nix/store/56ga87ych3hdqcpimgxs3wjfij7f4knp-git-2.29.2;/nix/store/lkvcgsb80dcv3hz17gfaixspqhzfnqim-git-lfs-2.12.1;/nix/store/h4bf4y1lpjavqj55fhl9kkn32g5ns0wj-iputils-20200821;/nix/store/44mpvg182rw7slvlg2f50wp740xzvmpl-just-0.4.4;/nix/store/ik5ya16lfsmayk2rcnby9xz3dgijln0m-killall-1003.1-2008;/nix/store/85v5wvrfl95bchrqwwrdyns3pazhglgj-ninja-1.10.1;/nix/store/8kdyn3rkqdl1pl294b6wp1kmb1kvff3v-nix-2.3.8-dev;/nix/store/cnxvbx9w31g2yv7kbx8rlnsf3li24w4g-boehm-gc-8.0.4-dev;/nix/store/0phlidpg14j1696i6kl7scf0cn1n02cg-boehm-gc-8.0.4;/nix/store/rq4xg5nyfmmavnkcjrdqf9rrh07g53v7-nix-2.3.8;/nix/store/93v8wwl5ix6nq8lyq704rgasnp5fsdqw-nixops-1.7;/nix/store/wf5q4d62skrw7fi3askhp0bdwb99sz5j-nodejs-12.19.0;/nix/store/qdj3i18s7jl98a8b10wz5k98sb7qr7wn-openapi-generator;/nix/store/ygzg3rb738vwrny6dgnbmx9ra7knan66-rsync-3.2.3;/nix/store/yw756amsqnq91pr65yc8qh4imz43k6h8-rust-1.48.0-2020-11-16-7eac88abb;/nix/store/s6by8bi3iwm7zm3n0asjp9yalrxs9fc4-sccache-0.2.13;/nix/store/462imcgmf8aqrll76mlhjwmxm6ayz9yk-toxiproxy-2.1.4;/nix/store/3difzwxgvsidpqzapjkh8giyppk66w08-wget-1.20.3;/nix/store/9vi62ib3qzj4cxb4h9sj4ydcmddgckba-which-2.21;/nix/store/qww3wcvgh3d [... cut ...]
-- Using CATKIN_TEST_RESULTS_DIR: /home/xneomac/Developer/sultan2/src/ros_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /nix/store/kzc80glrc7lgv6d5y68y0jb1403wgnkr-python2.7-nose-1.3.7/bin/nosetests-2.7
-- catkin 0.7.23
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

The big difference I see is the CMAKE_PREFIX_PATH which has all dependencies in nix-shell and only ros-env with the installed version. Is there anything specific to do to make it work with the installed version ?

lopsided98, 2021-01-08:
With Nix, you can generally only build software from within `nix-shell`. Libraries are not supposed to be installed using `nix-env`. See https://nixos.wiki/wiki/FAQ/I_installed_a_library_but_my_compiler_is_not_finding_it._Why%3F for more information.

xneomac, 2021-01-08:
OK, duly noted. Thank you for your quick response.

closed 2021-01-08 by xneomac