lopsided98/nix-ros-overlay #75: ros2doctor not found

akru, 2021-01-21:
**Description**

Currently, some packages like `desktop` or [ros-core](https://github.com/lopsided98/nix-ros-overlay/blob/master/distros/foxy/ros-core/default.nix#L5) from ROS2 distros require a `ros2doctor` package that didn't present in the overlay.

 The reason I guess is this package included in the `ros2cli` set: https://index.ros.org/p/ros2doctor/

lopsided98, 2021-01-23:
`ros2doctor` depends on the `ifcfg` Python package, which is not available in nixpkgs. If you want `ros2doctor`, please submit a PR to nixpkgs to add `ifcfg`.

lopsided98, 2022-10-19:
https://github.com/NixOS/nixpkgs/pull/164515

wentasah, 2023-05-29:
Thanks @movefasta for mentioning this. I've sent a PR to update the ROS 2 example to use `ros-core`, which was previously uninstalable due to missing `ifcfg`. As far as I can see, `ros2doctor` is still missing in *foxy*. It's available in *humble* and *rolling* only.

lopsided98, 2023-09-12:
Foxy is EOL now so I think this can be closed.

closed 2023-09-12 by lopsided98