lopsided98/nix-ros-overlay #82: Is $GAZEBO_PLUGIN_PATH set up correctly?

werner291, 2021-02-19:
Hello,

I'm back with environment variables again. So, I noticed the following:

```
[nix-shell:~/workspace/gazebo-experiment]$ echo $GAZEBO_PLUGIN_PATH
/nix/store/a217cx4cjy6p8ry6477sj5z56wlzqicz-ros-noetic-gazebo-ros-2.9.1-r1/lib
```

...which seems to be missing a bunch of plugins, which can be found by calling `which gazebo`:

```
[nix-shell:~/workspace/gazebo-experiment]$ ls /nix/store/2zzqmlahjr0phrpjxpcdi23lfdizm9p9-gazebo-11.3.0/lib/gazebo-11/plugins
libActorPlugin.so                   libDepthCameraPlugin.so      libJointControlPlugin.so           libMudPlugin.so                   libSonarPlugin.so
libActuatorPlugin.so                libDiffDrivePlugin.so        libJointTrajectoryPlugin.so        libPlaneDemoPlugin.so             libSphereAtlasDemoPlugin.so
libAmbientOcclusionVisualPlugin.so  libElevatorPlugin.so         libJoyPlugin.so                    libPressurePlugin.so              libStaticMapPlugin.so
libArduCopterPlugin.so              libFiducialCameraPlugin.so   libKeyboardGUIPlugin.so            libRandomVelocityPlugin.so        libStopWorldPlugin.so
libArrangePlugin.so                 libFlashLightPlugin.so       libKeysToCmdVelPlugin.so           libRayPlugin.so                   libTimerGUIPlugin.so
libAttachLightPlugin.so             libFollowerPlugin.so         libKeysToJointsPlugin.so           libRaySensorNoisePlugin.so        libTouchPlugin.so
libBlinkVisualPlugin.so             libForceTorquePlugin.so      libLedPlugin.so                    libReflectancePlugin.so           libTrackedVehiclePlugin.so
libBreakableJointPlugin.so          libGimbalSmall2dPlugin.so    libLensFlareSensorPlugin.so        libRegionEventBoxPlugin.so        libTransporterPlugin.so
libBuoyancyPlugin.so                libGpuRayPlugin.so           libLiftDragPlugin.so               libRestUiPlugin.so                libVariableGearboxPlugin.so
libCameraPlugin.so                  libHarnessPlugin.so          libLinearBatteryConsumerPlugin.so  libRestWebPlugin.so               libVehiclePlugin.so
libCartDemoPlugin.so                libHeightmapLODPlugin.so     libLinearBatteryPlugin.so          libRubblePlugin.so                libWheelSlipPlugin.so
libCessnaGUIPlugin.so               libHydraDemoPlugin.so        libLinkPlot3DPlugin.so             libShaderParamVisualPlugin.so     libWheelTrackedVehiclePlugin.so
libCessnaPlugin.so                  libHydraPlugin.so            libLookAtDemoPlugin.so             libSimEventsPlugin.so             libWindPlugin.so
libContactPlugin.so                 libImuSensorPlugin.so        libMisalignmentPlugin.so           libSimpleTrackedVehiclePlugin.so
libContainPlugin.so                 libInitialVelocityPlugin.so  libModelPropShop.so                libSkidSteerDrivePlugin.so
```

I got a bunch of missing plugin errors (`libLiftDragPlugin.so` is essential for me) up to this point, but that path as follows: `[nix-shell:~/workspace/gazebo-experiment]$ export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:/nix/store/2zzqmlahjr0phrpjxpcdi23lfdizm9p9-gazebo-11.3.0/lib/gazebo-11/plugins` seems to make those errors go away.

Is that perhaps missing from the nix files? Am I doing something wrong with my shell.nix file?

[shell.nix](https://github.com/lopsided98/nix-ros-overlay/files/6006749/shell.nix.txt)

lopsided98, 2021-02-19:
Right now, you need to source gazebo's `setup.sh`, but I should probably make this happen automatically. I have this is in my `shellHook`:
```
source '${rosPackages.noetic.gazebo}/share/gazebo-11/setup.sh'
```

werner291, 2021-02-19:
Aha! That's easy enough... Would indeed be easier if this was automatic. Still though, awesome job on making this gigantic software contraption run at all on something that isn't Ubuntu, I really appreciate not having to leave my favorite OS.

wentasah, 2026-08-29:
I think this can be closed.

closed 2026-08-29 by wentasah