lopsided98/nix-ros-overlay #85: How to get rviz?

werner291, 2021-02-26:
Hi, sorry for opening so many issues, I think there's something basic I'm not quite getting.

I have added `rosPackages.noetic.rviz` to the `buildInputs` of the shell, but then I get `rviz: command not found`

I'm running into a lot of issues like this... Where the package is definitely there in my `/nix/store`, but I'm getting `not found` errors. Is there a command I'm missing that sets up paths?

lopsided98, 2021-02-26:
I can't reproduce this. The following shell works correctly for me:
```
with import <nix-ros-overlay> {};

mkShell {
  buildInputs = [
      rosPackages.noetic.rviz
  ];
}
```

For packages with binaries in `bin`, there is nothing ROS specific going on; the package should just get added to `PATH`. Some ROS packages don't install binaries to `bin` and need to be started with `rosrun`, but this is not the case for `rviz`.

werner291, 2021-02-26:
Oh... it's the shellHook in my shell.nix file.

I was trying to implement this: https://github.com/lopsided98/nix-ros-overlay/issues/82#issuecomment-781730061 which *does* give me the Gazebo plugins, but in turn apparently breaks access to rviz.

Commenting the shellHook out gets rviz back... But I still want those plugins. I don't recall if it's possible to concatenate the shellHooks?

lopsided98, 2021-02-26:
The problem is `source devel/setup.bash`. That file contains the environment variable values that were used when you first created your catkin workspace, and it overrides the environment variables from nix-shell. The documentation for `devel/local_setup.bash` makes it sound like it should prevent this problem, but I've found that it doesn't work right either. The only solution I have found so far is to delete the `devel` and `build` folders every time I change the packages in the shell.

werner291, 2021-02-27:
Oh, that works, indeed!

One thing you could do, perhaps, is take a parameter to the catkin workspace directory, then run catkin_make (or whatever part of it generates the environment variables) and source the setup script when building the shell environment.

wentasah, 2026-08-29:
Closing as this seems related to noetic, which is EOL.

closed 2026-08-29 by wentasah