lopsided98/nix-ros-overlay #91: turtlebot3_gazebo turtlebot3_world.launch fails to find libboost_system.so.1.65.1

kjeremy, 2021-03-16:
Running the turtlebot example in the README gives the following:

`roslaunch turtlebot3_gazebo turtlebot3_world.launch`

> started core service [/rosout]
> /nix/store/isy3qy9dc2czy14szwhvm2qi46kjmj9m-ros-env/lib/rosout/rosout: error while loading shared libraries: libboost_system.so.1.65.1: cannot open shared object file: No such file or directory
> process[gazebo-2]: started with pid [40132]
> process[gazebo_gui-3]: started with pid [40136]
> process[spawn_urdf-4]: started with pid [40138]
> [rosout-1] process has died [pid 40129, exit code 127, cmd /nix/store/isy3qy9dc2czy14szwhvm2qi46kjmj9m-ros-env/lib/rosout/rosout __name:=rosout __log:=/home/jkolb/.ros/log/687686cf-8683-11eb-a16e-024247ea46e6/rosout-1.log].

lopsided98, 2021-03-22:
What OS are you using? My first guess would be that you built the packages on a non-NixOS system with sandboxing disabled and the package linked against your system Boost installation. If this is the case, I would recommend enabling sandboxing and rebuilding the packages (run `nix-collect-garbage` to delete them).

kjeremy, 2021-03-22:
Ubuntu 18.04. I already have ROS installed via `apt` so maybe that's the problem. If I set up sandboxing I get the following:

> jkolb@ubuntu:~/nix$ nix-shell -I nix-ros-overlay=https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz   --option extra-substituters 'https://ros.cachix.org'   --option trusted-public-keys 'cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY= ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo='   '<nix-ros-overlay/examples/turtlebot3-gazebo.nix>'
unpacking 'https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz'...
unpacking 'https://github.com/lopsided98/nixpkgs/archive/nix-ros.tar.gz'...
error: defusedxml-0.7.0 not supported for interpreter python2.7

lopsided98, 2021-03-22:
That is a new bug as of about 30 minutes ago when I updated the nixpkgs branch. It looks like the latest version of defusedxml has dropped support for Python 2. I've been meaning to update the example to noetic, but there is some problem that causes Gazebo to crash.

beezow, 2021-03-22:
Hi, I am having the same `defusedxml-0.7.0 not supported for interpreter python2.7` error.  I am running on NixOs 21.05 (unstable).
`error: while evaluating the attribute 'buildInputs' of the derivation 'nix-shell' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/build-support/mkshell/default.nix:28:3:
while evaluating the attribute 'passAsFile' of the derivation 'ros-env' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/build-support/trivial-builders.nix:7:7:
while evaluating the attribute 'rosPackages' at /nix/store/nm0zriy22v172diywsslzalryw6fpn9i-master.tar.gz/distros/build-env/default.nix:30:7:
while evaluating 'propagatePackages' at /nix/store/nm0zriy22v172diywsslzalryw6fpn9i-master.tar.gz/distros/build-env/default.nix:19:23, called from /nix/store/nm0zriy22v172diywsslzalryw6fpn9i-master.tar.gz/distros/build-env/default.nix:25:15:
while evaluating anonymous function at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/lib/lists.nix:645:24, called from /nix/store/nm0zriy22v172diywsslzalryw6fpn9i-master.tar.gz/distros/build-env/default.nix:24:29:
while evaluating the attribute 'propagatedBuildInputs' of the derivation 'ros-melodic-xacro-1.13.9-r1' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/stdenv/generic/make-derivation.nix:192:11:
while evaluating the attribute 'propagatedBuildInputs' of the derivation 'ros-melodic-roslaunch-1.14.10-r1' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/stdenv/generic/make-derivation.nix:192:11:
while evaluating the attribute 'propagatedBuildInputs' of the derivation 'ros-melodic-rosmaster-1.14.10-r1' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/stdenv/generic/make-derivation.nix:192:11:
while evaluating 'getOutput' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/lib/attrsets.nix:482:23, called from undefined position:
while evaluating anonymous function at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/stdenv/generic/make-derivation.nix:157:17, called from undefined position:
while evaluating the attribute 'defusedxml' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/top-level/python-packages.nix:1748:3:
while evaluating 'callPackageWith' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/lib/customisation.nix:117:35, called from /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/top-level/python-packages.nix:1748:16:
while evaluating 'makeOverridable' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/lib/customisation.nix:67:24, called from /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/lib/customisation.nix:121:8:
while evaluating anonymous function at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/development/python-modules/defusedxml/default.nix:1:1, called from /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/lib/customisation.nix:69:16:
while evaluating 'makeOverridablePythonPackage' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/top-level/python-packages.nix:29:37, called from /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/development/python-modules/defusedxml/default.nix:8:1:
while evaluating 'makeOverridable' at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/lib/customisation.nix:67:24, called from /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/top-level/python-packages.nix:31:12:
while evaluating anonymous function at /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/pkgs/development/interpreters/python/mk-python-derivation.nix:30:1, called from /nix/store/agcl2s44by4s5jbpysdhfkr9ckk3cmyz-source/lib/customisation.nix:69:16:
defusedxml-0.7.0 not supported for interpreter python2.7
`
It seems you are already aware though, let me know if you need more logs or anything.  I dont really know how the overlay works.

lopsided98, 2021-03-22:
Unfortunately, I'm not sure there is a fix. Python 2 is EOL, and is being removed from nixpkgs, so we may just have to drop support for melodic before its EOL. If you are stuck on melodic for whatever reason, you can try the `melodicPython3` package set, which mostly works okay in my experience. Otherwise, I'd recommend upgrading to noetic. I'm working on upgrading the turtlebot example to noetic.

beezow, 2021-03-22:
You mentioned that you updated the nixpkgs branch.  Within default.nix, could I pin the nixpkgs archive provided by nixos, rather than your achive?  I tried pinning a nixos-19.09 archive, but got some errors.  What commit of nixpkgs were you using before? Or was nixpkgs modified in some way to make it work?

Here is my new default.nix
```{ nixpkgs ? builtins.fetchTarball {
   # Descriptive name to make the store path easier to identify
        name = "nixos-19.09";
        url = https://github.com/nixos/nixpkgs/archive/75f4ba05c63be3f147bcc2f7bd4ba1f029cedcb1.tar.gz;
#        # Commit hash for nixos-unstable as of 2018-09-12
#        # `git ls-remote https://github.com/nixos/nixpkgs-channels nixos-unstable`
    }
, overlays ? [], ... }@args: import nixpkgs {
  overlays = [ (import ./overlay.nix) ] ++ overlays;
} // args
```

```error: while evaluating the attribute 'buildInputs' of the derivation 'nix-shell' at /nix/store/z2zsci63sx153k8r5xx49an4izs0isbl-nixos-19.09/pkgs/build-support/mkshell/default.nix:28:3:
while evaluating the attribute 'passAsFile' of the derivation 'ros-env' at /nix/store/z2zsci63sx153k8r5xx49an4izs0isbl-nixos-19.09/pkgs/build-support/trivial-builders.nix:7:14:
while evaluating 'propagatePackages' at /mnt/data/projects/daslab/auto_tractor/nix-ros-overlay/distros/build-env/default.nix:19:23, called from /mnt/data/projects/daslab/auto_tractor/nix-ros-overlay/distros/build-env/default.nix:34:21:
while evaluating anonymous function at /mnt/data/projects/daslab/auto_tractor/nix-ros-overlay/distros/build-env/default.nix:20:29, called from undefined position:
while evaluating 'callPackageWith' at /nix/store/z2zsci63sx153k8r5xx49an4izs0isbl-nixos-19.09/lib/customisation.nix:108:35, called from /mnt/data/projects/daslab/auto_tractor/nix-ros-overlay/distros/melodic/generated.nix:3039:26:
while evaluating 'makeOverridable' at /nix/store/z2zsci63sx153k8r5xx49an4izs0isbl-nixos-19.09/lib/customisation.nix:67:24, called from /nix/store/z2zsci63sx153k8r5xx49an4izs0isbl-nixos-19.09/lib/customisation.nix:112:8:
anonymous function at /mnt/data/projects/daslab/auto_tractor/nix-ros-overlay/distros/melodic/robot-state-publisher/default.nix:5:1 called without required argument 'urdfdom-headers', at /nix/store/z2zsci63sx153k8r5xx49an4izs0isbl-nixos-19.09/lib/customisation.nix:69:12
```

Any help/direction would be greatly appreciated!

lopsided98, 2021-03-22:
Previously, I was using commit `978d913d18d30951c0a236ae2633c7c2d6b6d520` from my fork. This contained some fixes for `pcl` (https://github.com/NixOS/nixpkgs/pull/115097) and `poco` (https://github.com/NixOS/nixpkgs/pull/98186, waiting for patch to be accepted upstream) on top of a few weeks old `nixos-unstable` branch. You may not need those fixes depending on which packages you use.

beezow, 2021-03-22:
Thanks,  I switched the pinned commit to `978d913d18d30951c0a236ae2633c7c2d6b6d520` and got everything to build and run again!

lopsided98, 2021-03-24:
I updated the `turtlebot3-gazebo` example to `noetic`. The issues I experienced previously seem to have been fixed.

There is nothing I can do about the melodic Python 2 situation, other than pinning nixpkgs to an old version. I will continue to update melodic, and users can pin nixpkgs to an older version if they need to use it. There is also the `melodicPython3` package set, which builds melodic with Python 3 and may be useful if you are stuck with melodic but can use Python 3.

closed 2021-04-26 by lopsided98