lopsided98/nix-ros-overlay #93: aarch64 cache

3noch, 2021-04-15:
I've discovered that setting up an aarch64 cache with GitHub actions and Cachix is actually really easy.

You need
```
uses: docker/setup-qemu-action@v1
```

and then run your build with

```
nix-build --option system aarch64-linux --extra-platforms aarch64-linux
```

lopsided98, 2021-04-23:
I've been trying this out, and it works but tends to timeout with mass-rebuilds.

3noch, 2021-04-26:
@lopsided98 Does that pose a real problem? Since cachix keeps all intermediate builds, you just need to start the build again to keep going. You also have the option of building it locally (perhaps via qemu binfmt support in NixOS) and then pushing to the cache to fill it.

3noch, 2021-04-26:
But you're absolutely right that binfmt approach is extremely slow. I have an aarch64 device that is *slightly* faster but unimpressively so.

lopsided98, 2021-05-29:
Yeah, it hasn't been too much of a problem over the last month I've been running it.

closed 2021-05-29 by lopsided98