lopsided98/nix-ros-overlay #110: `nix flake show github:lopsided98/nix-ros-overlay` does not work

wucke13, 2021-06-21:
I only get 

```
github:lopsided98/nix-ros-overlay/9fbde27da7c9a15970115c57e1eb1c02a1682dd4
├───nixosModule: unknown
├───overlay: Nixpkgs overlay
└───packages
    ├───aarch64-darwin
error: expected a derivation
```

It would be very cool if nix flake show works, in order to allow easy discovery of available packages.

lopsided98, 2021-06-28:
Nested package sets aren't allowed in flakes, see: https://discourse.nixos.org/t/flake-questions/8741. I'll switch to `legacyPackages`, but that just causes the package sets to be ignored rather than giving an error.

wucke13, 2021-06-28:
So basically this could not work well with flakes any way?

lopsided98, 2021-06-29:
Pretty much. I don't see a reasonable way of organizing this overlay without nested package sets.

wucke13, 2021-06-29:
Thus wondering: would flake utils `flake-utils.lib.flattenTree` help here?

acowley, 2021-07-02:
I apologize that this doesn't solve the reported Issue, but package discovery works quite well in `nix repl` using `ros = builtins.getFlake "github:lopsided98/nix-ros-overlay"` and then tab completion on `ros`.

wucke13, 2022-05-02:
Now it does at least something :)

```console
$ nix flake show github:lopsided98/nix-ros-overlay --legacy
github:lopsided98/nix-ros-overlay/3e26e1c93fcb2487e5b8908129f2920a94812a2f
├───legacyPackages
│   ├───aarch64-darwin
│   ├───aarch64-genode
│   ├───aarch64-linux
│   ├───aarch64-none
│   ├───arm-none
│   ├───armv5tel-linux
│   ├───armv6l-linux
│   ├───armv6l-none
│   ├───armv7a-darwin
│   ├───armv7a-linux
│   ├───armv7l-linux
│   ├───avr-none
│   ├───i686-cygwin
│   ├───i686-darwin
│   ├───i686-freebsd
│   ├───i686-linux
│   ├───i686-netbsd
│   ├───i686-none
│   ├───i686-openbsd
│   ├───i686-windows
│   ├───js-ghcjs
│   ├───mipsel-linux
│   ├───msp430-none
│   ├───powerpc-none
│   ├───powerpc64le-linux
│   ├───riscv32-linux
│   ├───riscv32-none
│   ├───riscv64-linux
│   ├───riscv64-none
│   ├───vc4-none
│   ├───wasm32-wasi
│   ├───wasm64-wasi
│   ├───x86_64-cygwin
│   ├───x86_64-darwin
│   ├───x86_64-freebsd
│   ├───x86_64-genode
│   ├───x86_64-linux
│   ├───x86_64-netbsd
│   ├───x86_64-none
│   ├───x86_64-openbsd
│   ├───x86_64-redox
│   ├───x86_64-solaris
│   └───x86_64-windows
├───nixosModule: unknown
└───overlay: Nixpkgs overlay
```

closed 2022-05-02 by wucke13