lopsided98/nix-ros-overlay #113: Getting rosbag2-bag-v2-plugins to build

acowley, 2021-07-06:
The `rosbag2-bag-v2-plugins` package did not build out of the box for me with the following error,

```
> -- Checking for module 'roscpp'
> --   No package 'roscpp' found
> CMake Error at CMakeLists.txt:28 (message):
>   Failed to find ROS 1 roscpp, cannot build.
```

I was eventually able to get through the configure phase with these changes,

```nix
rosbag2-bag = ros.rosbag2-bag-v2-plugins.overrideAttrs (old: {
            buildInputs = old.buildInputs or [] ++ [ros1.roscpp ros1.cpp-common ros1.rosbag-storage];
            nativeBuildInputs = old.nativeBuildInputs or [] ++ [ros.pluginlib ros.std-msgs ros.rosgraph-msgs];
          });
```

Where I am working with flakes, and have,

```nix
ros = ros-flake.packages.${system}.foxy;
ros1 = ros-flake.packages.${system}.noetic;
```

The configure error I showed was due to `pkg-config` failing to find the things I added to `buildInputs`. Later during the configure phase, various msg definitions are wanted for code generation, so I added those dependencies.

I also had to hand edit the `CMakeLists.txt` file to mention `ros1_rosbag_storage_vendor` rather than `ros1_rosbag_storage`. That name is used to construct file names for cmake to look for which it fails to find as the actual files in the nix store have the `_vendor` part.

The next problem is that the build phase fails with,

```nix
Traceback (most recent call last):
  File "/home/acowley/Projects/ros/bag/rosbag2_bag_v2-release-release-foxy-rosbag2_bag_v2_plugins-0.0.10-1/src/generate_converter_cpp.py", line 19, in <module>
    from ros1_bridge import generate_messages
ModuleNotFoundError: No module named 'ros1_bridge'
```

I am a bit stuck here as the Python parts of `ros1_bridge` seem to be lost at some point.

So my questions are:
1) Have I gone too far into the weeds here? Am I just missing a simpler fix for the original failure of this package not building?
2) If I'm on a more or less good path to fixing this, can you suggest a maneuver to get the `ros1-bridge` python components included in its output?

Thank you!

lopsided98, 2021-07-07:
The core of the problem is that these packages depend on ROS 1 packages, which can't be specified in `package.xml` for a ROS 2 package. Also, it so happens that if `ros1-rosbag-storage-vendor` doesn't find `roscpp`, it just installs an empty package rather than failing. This is why you had to edit `CMakeLists.txt`.

I fixed those issues in #114, but I'm now running into this error:
```
CMake Error at CMakeLists.txt:74 (add_library):
  Target "rosbag2_bag_v2_plugins" links to target
  "std_msgs::std_msgs__rosidl_typesupport_cpp" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
```
I'm not sure what's wrong here, but I'm thinking that [this code](https://github.com/ros2/rosbag2_bag_v2/blob/0.2.0/rosbag2_bag_v2_plugins/CMakeLists.txt#L68) is supposed to call `find_package(std_msgs)`, but for some reason `ament_index_get_resources()` isn't finding `std_msgs`.

wentasah, 2026-08-29:
Closing as `rosbag2_bag_v2` binaries are [no longer part of ROS](https://github.com/ros/rosdistro//blob/82eefa281fd5702f1869408c66606d395cd635ee/jazzy/distribution.yaml#L10467-L10474).

closed 2026-08-29 by wentasah