lopsided98/nix-ros-overlay #116: NixOS: Enabling services.nix fails due to defusedxml-0.7.1

kjeremy, 2021-07-13:
I have the following `/etc/nixos/flake.nix`

```
  inputs = {
    nixpkgs.url = github:NixOS/nixpkgs/nixos-21.05;
    nix-ros-overlay.url = github:lopsided98/nix-ros-overlay;
  };
  outputs = { nixpkgs, nix-ros-overlay, ... }: {

    nixosConfigurations.nixos = nixpkgs.lib.nixosSystem {
      system = "x86_64-linux";
      modules = [
        ./configuration.nix
        nix-ros-overlay.nixosModule
      ];
   };
  };
}
```

In my `configuration.nix` I have
```
services.ros = {
    enable = true;
    distro = "melodic";
  };
```

A `nixos-rebuild switch` fails with: 

> error: defusedxml-0.7.1 not supported for interpreter python2.7

lopsided98, 2021-07-13:
melodic uses Python 2, which is being removed from nixpkgs due to it being EOL. See the discussion here: https://github.com/lopsided98/nix-ros-overlay/pull/99

kjeremy, 2021-07-14:
Is there a workaround? I'm still getting used to nixos.

lopsided98, 2021-07-14:
You really need to move to Python 3, either by using noetic, or if there is a reason you need to stay on melodic, you can use the `melodicPython3` distro, which builds melodic with Python 3. It is not feasible for me to continue to support Python 2; see the issue I linked above for a more detailed explanation.

closed 2021-07-14 by kjeremy