lopsided98/nix-ros-overlay #117: How to nixify a simple catkin c++ project

kjeremy, 2021-07-14:
Is there a good example of taking a boring catkin based c++ project and nixifying it?

mikepurvis, 2021-07-15:
Nix is the packaging layer! Hopefully in _most_ cases there is nothing special needed for Nix-friendliness— the issues that are hacked around for example in https://github.com/lopsided98/nix-ros-overlay/blob/master/distros/distro-overlay.nix tend to be:

- Issues upstream, like missing dependencies, missing patches, etc.
- Issues with shell script shebangs, particularly in generated/templated shell scripts.

Overall, Python shebangs aren't a huge problem in the ROS world, since a lot of sloppiness in that area has had to be tightened up for the Python 3 migration, and for sane compatibility with multi-Python environments like Homebrew.

lopsided98, 2021-07-20:
Like Mike said, you shouldn't need to modify the package much to support Nix. Beyond that, it depends on your specific use case. If you just want to make your package available in this overlay, you can submit it to rosdistro and it will show up here in a week or so assuming all the dependencies are available. If you want to add a Nix expression to your own repo, you can manually write one that is similar to the auto-generated ones in this repo. Unfortunately, I don't know of a public example that does this.

mikepurvis, 2021-07-20:
My nix-tests repo exists but is not a great example. I'm using the nix-ros-overlay flake as an input and applying further overlays, but otherwise I'm starting from scratch with the distro packages; I'm really only reusing the awesome work that's been done already on packaging tools, dependencies, gazebo, etc.

But yeah, there's definitely a gap here for a resource/example applicable to the "ROS expert, Nix noob" group that demonstrates a separate repo utilizing this one and shows how to do things like:

- generate further local/private packages from bloom/superflore.
- add packages that are unreleased, and built directly from local or remote sources.
- override an upstream package to add a patch or select an older tag.
- use `nix develop` to step through manually building a package in the sandbox.

I think I could probably help with most of this (at least the non-superflore parts), but I'm certainly far from an expert.

kjeremy, 2021-07-21:
I am definitely in the camp of "pretty familiar with ROS but Nix newbie". In my case I have a private monorepo with a bunch of catkin paxkages that I would like to nixify.

werner291, 2021-08-14:
+1 here, I dabble a bit in Nix, but I'm kinda stuck when it comes to installing the software I need on my laptop that runs NixOS

werner291, 2021-08-15:
I got this to compile:

```
let 
    tb = builtins.fetchTarball {
        name = "nix-ros-overlay"; 
        url = "https://github.com/lopsided98/nix-ros-overlay/archive/be22697212debf9912e75bb5424285bfbbfbbe41.tar.gz"; 
    };
    pkgs = import tb {};
in
    with pkgs; with rosPackages.noetic; buildRosPackage {
        pname = "ros-noetic-robowflex";
        version = "rev-master";

        src = fetchGit {
            url = "https://github.com/werner291/robowflex.git";
            rev = "aa6299f6b99bf9b9bc9a1c972b89994b45530541";
        };

        buildType = "catkin";
        checkInputs = [ ];
        propagatedBuildInputs = [ 
            moveit-core
            moveit-ros-planning
            
            assimp boost bullet console-bridge eigen eigen-stl-containers fcl geometric-shapes geometry-msgs kdl-parser moveit-msgs octomap octomap-msgs pybind11-catkin random-numbers rosconsole roslib rostime sensor-msgs shape-msgs srdfdom std-msgs tf2-eigen tf2-geometry-msgs trajectory-msgs urdf urdfdom urdfdom-headers visualization-msgs xmlrpcpp 
            ];
        nativeBuildInputs = [ 
            cmake
            catkin pkg-config 
            libyamlcpp
            hdf5
            hdf5-cpp
            rosbag
            ];
            
        configurePhase = ''
            cmake robowflex/robowflex_library -DCMAKE_INSTALL_PREFIX=$out
        '';

        meta = {
            description = ''Wrapper library for MoveIt.'';
            #license = with lib.licenses; [ bsdOriginal ];
        };
    }
```

I am by no means guaranteeing the quality of that nix expression, but it's a start on how to do it?

closed 2021-12-22 by kjeremy