lopsided98/nix-ros-overlay #118: Make pythonOverrides composable?

mikepurvis, 2021-07-15:
I'm hitting an issue where I want to further override some of the nix-ros-overlay python packages in my own flake, and I'd like to do so in a way that doesn't have me vendoring out the whole thing, but this seems to requires some cooperation between the overlay, per:

https://discourse.nixos.org/t/makeextensibleasoverlay/7116/5

Would we consider this?

mikepurvis, 2021-07-16:
Based on the discussion in https://github.com/NixOS/nixpkgs/issues/44426#issuecomment-532208394; we might be able to get python composability for free if this were changed to overrideScope':

https://github.com/lopsided98/nix-ros-overlay/blob/e90355503d47d9a6a5907673bb5c1939fa124d7d/pkgs/default.nix#L3-L4

Will experiment and send a PR if something comes of it.

lopsided98, 2021-07-16:
I don't think the PR that made that work was merged.

mikepurvis, 2021-07-16:
Oh, so I see. :(

I've been using `overrideScope'` at the top level and it seems to work really well for me; that's too bad it never made it in for `pythonPackages`.

lopsided98, 2021-07-18:
I implemented the alternative technique in the linked Discourse thread.

closed 2021-07-18 by lopsided98