lopsided98/nix-ros-overlay #119: Compilation failure, missing dependency to PyQt5?

werner291, 2021-07-16:
Hello, while trying to start up the following shell.nix:

```
let
    ros_tar = (builtins.fetchTarball {
        name = "nix-ros-overlay";
        url = "https://github.com/lopsided98/nix-ros-overlay/archive/e90355503d47d9a6a5907673bb5c1939fa124d7d.tar.gz";
    });
in
    with import ros_tar { };
mkShell {
  buildInputs = with rosPackages.noetic; [
    rviz
  ];
  
  LD_LIBRARY_PATH = with pkgs.xlibs; "${pkgs.libGL}/lib";

  # Commenting this out brings rviz back.
  shellHook = "source devel/setup.bash";
}
```

I got the build error:

```
Scanning dependencies of target executable_autogen
[ 59%] Automatic MOC for target executable
[ 59%] Built target executable_autogen
Scanning dependencies of target librviz_sip
[ 59%] Running SIP generator for rviz_sip Python bindings...
Traceback (most recent call last):
  File "/nix/store/yqsqsqnkxwxyn3j9m5n2w1zv2gswl6f8-ros-noetic-python-qt-binding-0.4.3-r1/share/python_qt_binding/cmake/sip_configure.py", line 57, in get_sip_dir_flags
    sip_dir = config.pyqt_sip_dir
  File "/nix/store/clr06cdrd3gfn35aas9hbq5nq0laf9hm-python3.8-sip-4.19.25/lib/python3.8/site-packages/sipconfig.py", line 193, in __getattr__
    raise AttributeError("\"%s\" is not a valid configuration value or user option" % name)
AttributeError: "pyqt_sip_dir" is not a valid configuration value or user option

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/nix/store/yqsqsqnkxwxyn3j9m5n2w1zv2gswl6f8-ros-noetic-python-qt-binding-0.4.3-r1/share/python_qt_binding/cmake/sip_configure.py", line 92, in <module>
    sip_dir, sip_flags = get_sip_dir_flags(config)
  File "/nix/store/yqsqsqnkxwxyn3j9m5n2w1zv2gswl6f8-ros-noetic-python-qt-binding-0.4.3-r1/share/python_qt_binding/cmake/sip_configure.py", line 72, in get_sip_dir_flags
    raise FileNotFoundError('The sip directory for PyQt5 could not be located. Please ensure' +
FileNotFoundError: The sip directory for PyQt5 could not be located. Please ensure that PyQt5 is installed
make[2]: *** [src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:114: sip/rviz_sip/Makefile] Error 1
make[1]: *** [CMakeFiles/Makefile2:2947: src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
builder for '/nix/store/mbsi61mgsbr1s2mf8na88mylbamb8gz1-ros-noetic-rviz-1.14.8-r1.drv' failed with exit code 2
error: build of '/nix/store/mbsi61mgsbr1s2mf8na88mylbamb8gz1-ros-noetic-rviz-1.14.8-r1.drv' failed
```

mikepurvis, 2021-07-16:
Pretty sure this is a consequence of https://github.com/NixOS/nixpkgs/pull/123378; the patch for `python_qt_binding` is going to need a similar update to the ones that went in with that PR:

https://github.com/lopsided98/nix-ros-overlay/blob/e90355503d47d9a6a5907673bb5c1939fa124d7d/distros/distro-overlay.nix#L235-L239

You could test and send a PR; alternatively setting your base to 21.05 rather than unstable should fix it.

lopsided98, 2021-07-16:
This is fixed in staging, and I have sent a better long term fix upstream: https://github.com/ros-visualization/python_qt_binding/pull/105

closed 2021-07-16 by lopsided98

mikepurvis, 2021-07-16:
For my understanding, what is the staging fix?

werner291, 2021-07-17:
Thanks for the quick fix! Will try it out when I get the chance.

lelongg, 2021-07-28:
I tried the nix-ros-overlay/staging branch and got the same issue (with melodicPython3)

lopsided98, 2021-07-29:
I cannot reproduce this; could you provide more detail about the command you are running and the error?

lelongg, 2021-07-29:
Ok I'll try to make a minimal reproduction case

lelongg, 2021-07-29:
```nix
{
  inputs = {
    nixpkgs.url = "nixpkgs/nixos-21.05";
    flake-utils.url = "github:numtide/flake-utils";
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/staging";
  };
  outputs = { self, ... }@inputs:
    with inputs;
    flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlay ];
        };
      in {
        devShell = with pkgs;
          mkShell { buildInputs = [ rosPackages.melodicPython3.qt-gui-cpp ]; };
      });
}
```

`nix develop` fails with

```
builder for '/nix/store/dl3ddpp72xpc7m4zbza1yq9lvzfzpns1-ros-melodic-qt-gui-cpp-0.4.2-r1.drv' failed with exit code 2; last 10 log lines:
  
  Traceback (most recent call last):
    File "/nix/store/jxb15l3zcx8q103nxj8rdw4ihbcqr1dw-ros-melodic-python-qt-binding-0.4.4-r1/share/python_qt_binding/cmake/sip_configure.py", line 99, in <module>
      sip_dir, sip_flags = get_sip_dir_flags(config)
    File "/nix/store/jxb15l3zcx8q103nxj8rdw4ihbcqr1dw-ros-melodic-python-qt-binding-0.4.4-r1/share/python_qt_binding/cmake/sip_configure.py", line 79, in get_sip_dir_flags
      raise FileNotFoundError('The sip directory for PyQt5 could not be located. Please ensure' +
  FileNotFoundError: The sip directory for PyQt5 could not be located. Please ensure that PyQt5 is installed
  make[2]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/build.make:109: sip/qt_gui_cpp_sip/Makefile] Error 1
  make[1]: *** [CMakeFiles/Makefile2:284: src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
  make: *** [Makefile:149: all] Error 2
cannot build derivation '/nix/store/kkg6s3aq0cnp2af873bkly3p5c204z2b-nix-shell-env.drv': 1 dependencies couldn't be built
error: build of '/nix/store/kkg6s3aq0cnp2af873bkly3p5c204z2b-nix-shell-env.drv' failed
```

lelongg, 2021-07-29:
OK solved by using the `nix-ros-overlay/nixpkgs`

```nix
{
  inputs = {
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";
    flake-utils.url = "github:numtide/flake-utils";
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/staging";
  };
  outputs = { self, ... }@inputs:
    with inputs;
    flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlay ];
        };
      in {
        devShell = with pkgs;
          mkShell { buildInputs = [ rosPackages.melodicPython3.qt-gui-cpp ]; };
      });
}
```