lopsided98/nix-ros-overlay #121: Ompl build fails

werner291, 2021-07-19:
Hello,

The following shell file fails to build:

```
let
    ros_tar = (builtins.fetchTarball {
        name = "nix-ros-overlay";
        url = "https://github.com/lopsided98/nix-ros-overlay/archive/344511c1a3818f6f681ff64cb11c9553a4b6973c.tar.gz";
    });
in
    with import ros_tar { };
mkShell {
  buildInputs = with rosPackages.noetic; [
#     python3Packages.pyqt5
#     rosPackages.noetic.rviz
#     catkin
#     cmake
#     rosbash
#     moveit-ros-move-group
#     moveit-planners-ompl
    ompl
  ];
  
  LD_LIBRARY_PATH = with pkgs.xlibs; "${pkgs.libGL}/lib";

  # Commenting this out brings rviz back.
  shellHook = "source devel/setup.bash";
}
```

With a lot of these types of errors:

```
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:364:25: error: 'dReal' has not been declared
  364 |   dRay (dSpaceID space, dReal length)
      |                         ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:367:32: error: 'dReal' has not been declared
  367 |   void create (dSpaceID space, dReal length) {
      |                                ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:372:19: error: 'dReal' has not been declared
  372 |   void setLength (dReal length)
      |                   ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:374:3: error: 'dReal' does not name a type
  374 |   dReal getLength()
      |   ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:377:13: error: 'dReal' has not been declared
  377 |   void set (dReal px, dReal py, dReal pz, dReal dx, dReal dy, dReal dz)
      |             ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:377:23: error: 'dReal' has not been declared
  377 |   void set (dReal px, dReal py, dReal pz, dReal dx, dReal dy, dReal dz)
      |                       ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:377:33: error: 'dReal' has not been declared
  377 |   void set (dReal px, dReal py, dReal pz, dReal dx, dReal dy, dReal dz)
      |                                 ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:377:43: error: 'dReal' has not been declared
  377 |   void set (dReal px, dReal py, dReal pz, dReal dx, dReal dy, dReal dz)
      |                                           ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:377:53: error: 'dReal' has not been declared
  377 |   void set (dReal px, dReal py, dReal pz, dReal dx, dReal dy, dReal dz)
      |                                                     ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:377:63: error: 'dReal' has not been declared
  377 |   void set (dReal px, dReal py, dReal pz, dReal dx, dReal dy, dReal dz)
      |                                                               ^~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:379:13: error: 'dVector3' has not been declared
  379 |   void get (dVector3 start, dVector3 dir)
      |             ^~~~~~~~
/nix/store/rsbmrrjh9fvi72cb4wjvkzz81gvsh71c-ode-0.12/include/ode/odecpp_collision.h:379:29: error: 'dVector3' has not been declared
  379 |   void get (dVector3 start, dVector3 dir)
```

werner291, 2021-07-19:
Looks like `dReal` is defined in `odecpp_common.h`, but `odecpp_collision.h` doesn't include it for some reason...

lopsided98, 2021-07-19:
ODE is severely outdated (2013) in nixpkgs. Updating to the latest version fixes this issue: https://github.com/NixOS/nixpkgs/pull/130644

closed 2021-07-19 by lopsided98