lopsided98/nix-ros-overlay #128: turtlebot3_gazebo turtlebot3_world.launch fails to find setup.sh

Intey, 2021-08-18:
OS: NixOS
Try to chck sample from README, but catch exceptions below. 
I don't think this is relates to this repository, but maybe someone can suggest where to search fix? 
As i understand, path to setup.sh got from scprit `gazebo` under `ROS_PACKAGE_PATH`:
```
setup_path=$(pkg-config --variable=prefix gazebo)/share/gazebo/
```
where pkg-config returns 
```
/nix/store/kygqmmjhg5wqpx4nmhmn524l4sicl9qm-gazebo-11.5.1/lib/pkgconfig/../../../../../
``` 
I also find out place where this variable defined: `/nix/store/HASH-gazebo-11.5.1/lib/pkgconfig/gazebo`

but i'm stucked with how to change this? 

Error full message:
```
[nix-shell:~]$ roslaunch turtlebot3_gazebo turtlebot3_world.launch 
... logging to /home/intey/.ros/log/58f5493c-0023-11ec-803e-d85ed301e4ef/roslaunch-puppa-39166.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://localhost:37841/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [39180]
process[gazebo_gui-2]: started with pid [39181]
process[spawn_urdf-3]: started with pid [39183]
/nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/gazebo_ros/gzserver: line 31: /nix/store/kygqmmjhg5wqpx4nmhmn524l4sicl9qm-gazebo-11.5.1/lib/pkgconfig/../../../../..//share/gazebo//setup.sh: No such file or directory
/nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/gazebo_ros/gzclient: line 31: /nix/store/kygqmmjhg5wqpx4nmhmn524l4sicl9qm-gazebo-11.5.1/lib/pkgconfig/../../../../..//share/gazebo//setup.sh: No such file or directory
[gazebo-1] process has died [pid 39180, exit code 1, cmd /nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/gazebo_ros/gzserver -e ode /nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/share/turtlebot3_gazebo/worlds/turtlebot3_world.world __name:=gazebo __log:=/home/intey/.ros/log/58f5493c-0023-11ec-803e-d85ed301e4ef/gazebo-1.log].
log file: /home/intey/.ros/log/58f5493c-0023-11ec-803e-d85ed301e4ef/gazebo-1*.log
[gazebo_gui-2] process has died [pid 39181, exit code 1, cmd /nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/intey/.ros/log/58f5493c-0023-11ec-803e-d85ed301e4ef/gazebo_gui-2.log].
log file: /home/intey/.ros/log/58f5493c-0023-11ec-803e-d85ed301e4ef/gazebo_gui-2*.log
^C[spawn_urdf-3] killing on exit
Traceback (most recent call last):
  File "/nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/gazebo_ros/spawn_model", line 239, in <module>
    exit_code = sm.run()
  File "/nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/gazebo_ros/spawn_model", line 183, in run
    success = gazebo_interface.spawn_urdf_model_client(self.args.model, model_xml, self.args.robot_namespace,
  File "/nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/python3.9/site-packages/gazebo_ros/gazebo_interface.py", line 28, in spawn_urdf_model_client
    rospy.wait_for_service(gazebo_namespace+'/spawn_urdf_model')
  File "/nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/python3.9/site-packages/rospy/impl/tcpros_service.py", line 166, in wait_for_service
    raise ROSInterruptException("rospy shutdown")
rospy.exceptions.ROSInterruptException: rospy shutdown
shutting down processing monitor...
... shutting down processing monitor complete
done

samhedin, 2021-08-28:
I am also having this issue. From the examples in the repo, this works
```
nix-shell turtlebot3-gazebo.nix
roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
```
But this does not
```
nix-shell turtlebot3-gazebo.nix
roslaunch turtlebot3_gazebo turtlebot3_world.launch
```

```
/nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/gazebo_ros/gzserver: line 31: /nix/store/kygqmmjhg5wqpx4nmhmn524l4sicl9qm-gazebo-11.5.1/lib/pkgconfig/../../../../..//share/gazebo//setup.sh: No such file or directory
/nix/store/9p7vp211yb8ch03q09r6bxd0k1lhzx1k-ros-env/lib/gazebo_ros/gzclient: line 31: /nix/store/kygqmmjhg5wqpx4nmhmn524l4sicl9qm-gazebo-11.5.1/lib/pkgconfig/../../../../..//share/gazebo//setup.sh: No such file or directory
```
Maybe that was obvious...
Thanks for the overlay, it made me very happy to see that there was a nice way to get ROS working on nixos.

lopsided98, 2021-08-30:
This should be fixed by https://github.com/osrf/gazebo/pull/3076. I have applied that patch to the staging branch.

closed 2021-08-30 by lopsided98