lopsided98/nix-ros-overlay #150: legacyPackages lacks colcon and other tools

d-johnston-fugro, 2021-11-18:
When using the flake for `nix-ros-overlay`, certain tools like `colcon` are not included under `legacyPackages`. I believe this is because you scope the export as `pkgs.rosPackages`.

d-johnston-fugro, 2021-11-18:
```nix
nix-repl> ros = builtins.getFlake "github:lopsided98/nix-ros-overlay"
error: attribute 'colcon' missing

       at «string»:1:1:

            1| ros.legacyPackages.x86_64-linux.noetic.colcon
             | ^
            2|
nix-repl> pkgs = import <nixpkgs> { system = "x86_64-linux"; overlays = [ ros.outputs.overlay ]; }
nix-repl> pkgs.colcon
«derivation /nix/store/a0wxvsxz1dhcpa002s8cvz917vhajd86-colcon-0.6.1.drv»
```

koalp, 2023-11-15:
Hello

What do you think about adding something like 
```nix
  rosPackages = {
    gazebo = super.gazebo;
    colcon = super.colcon;
  };
```
in `default.nix` ?

It's probably what I will do in a fork to be able to use `colcon` in my project.

lopsided98, 2023-11-16:
Yes, I've considered doing this but never got around to it. There are a lot of other packages that might be useful, but this is better than nothing.

closed 2026-03-28 by wentasah