lopsided98/nix-ros-overlay #151: Unable to run example in README.md

midnightexigent, 2021-11-19:
When I try to run

```bash
nix-shell \
  -I nix-ros-overlay=https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz \
  --option extra-substituters 'https://ros.cachix.org' \
  --option trusted-public-keys 'cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY= ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' \
  '<nix-ros-overlay/examples/turtlebot3-gazebo.nix>'
```

I get the following errors 

```log
CMake Error at /nix/store/m33sn6h5kgfbk3z4avwmgxam1dny07i0-cmake-3.21.2/share/cmake-3.21/Modules/CMakeTestCCompiler.cmake:69 (message
):
  The C compiler

    "/nix/store/s5hkav7whndbfz0szshpb46h4idqdq9a-gcc-wrapper-10.3.0/bin/gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /build/catkin-release-release-noetic-catkin-0.8.10-1/build/CMakeFiles/CMakeTmp

    Run Build Command(s):/nix/store/xgp0bgw4rpnbc3vr2qdsdbixp3zy4v1l-gnumake-4.3/bin/make -f Makefile cmTC_e9851/fast && /nix/store/x
gp0bgw4rpnbc3vr2qdsdbixp3zy4v1l-gnumake-4.3/bin/make  -f CMakeFiles/cmTC_e9851.dir/build.make CMakeFiles/cmTC_e9851.dir/build
    make: /bin/sh: No such file or directory
    make: *** [Makefile:127: cmTC_e9851/fast] Error 127

``` 

What am I missing ?

lopsided98, 2021-11-19:
That looks like a problem with your Nix installation. Can you build other Nix packages?

midnightexigent, 2021-11-19:
Indeed, I just tried building an [example](https://nix-tutorial.gitlabpages.inria.fr/nix-tutorial/first-package.html) and got the same error

midnightexigent, 2021-11-20:
I solved the problem by installing `nix` with the installation script they provide instead of the Arch Linux package manager `pacman`

closed 2021-11-20 by midnightexigent