lopsided98/nix-ros-overlay #155: urdf compiling issues, from less than a week,

frett27, 2021-12-04:
We're using the nix ros overlay, with great pleasure, 
from one week ago, updates are breaking compilation for urdf module (on noetic),  
this happend when this library is required.  here it's for the rviz, 

`-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /nix/store/sr0c5qvlq8rnlvjxgx2mwz4x6szc79n3-ros-noetic-urdf-1.13.2-r1/share/urdf/cmake/urdfConfig.cmake:113 (message):
  Project 'urdf' specifies
  '/nix/store/8cs76yqhlc2vnp1w4pp26z0aclx8l3w1-urdfdom-3.0.0/lib/urdfdom/cmake//../../../../../../include'
  as an include dir, which is not found.  It does neither exist as an
  absolute directory nor in
  '${prefix}//nix/store/8cs76yqhlc2vnp1w4pp26z0aclx8l3w1-urdfdom-3.0.0/lib/urdfdom/cmake//../../../../../../include'.
  Check the issue tracker 'https://github.com/ros/urdf/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /nix/store/w1l1h5s0nh0b28m6gsdf3jyzb7r5pr97-ros-noetic-catkin-0.8.10-r1/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:117 (find_package)`

using the git repo for the overlay and coming back in history, the issue seems to stay.

i'm using this typical shell for invocation, 

`with import <nix-ros-overlay> {};
with rosPackages.noetic;

mkShell {
  buildInputs = [
glibcLocales
      (buildEnv { paths = [
		rosbash
		rviz
		moveit-core
		rqt-common-plugins
		tf-conversions
		blas
		xacro
		robot-state-publisher
		joint-state-publisher
		moveit-ros-visualization
                catkin
                industrial-robot-client
                moveit-planners-ompl 
		cv-camera
		map-server
		image-view
		image-common
		compressed-image-transport
		image-transport-plugins
		ros-environment
 ]; } )
  ];
}
`

using the nix-shell and procedure to make the package, using this following actions : 
`use@alexa:~/ros-nix$ nix-shell -I nix-ros-overlay=./nix-ros-overlay '<nix-ros-overlay>' -A rosPackages.noetic.rviz

[nix-shell:~/ros-nix]$ unpack
unpack200    unpackFile   unpackPhase  

[nix-shell:~/ros-nix]$ unpackPhase 
unpacking source archive /nix/store/28j2wn6qn2njdlq2f9wj1q5ihr58h2fh-1.14.10-1.tar.gz
source root is rviz-release-release-noetic-rviz-1.14.10-1
setting SOURCE_DATE_EPOCH to timestamp 1632721861 of file rviz-release-release-noetic-rviz-1.14.10-1/srv/SendFilePath.srv

[nix-shell:~/ros-nix]$ cd rviz-release-release-noetic-rviz-1.14.10-1/

[nix-shell:~/ros-nix/rviz-release-release-noetic-rviz-1.14.10-1]$ configurePhase  
no configure script, doing nothing

[nix-shell:~/ros-nix/rviz-release-release-noetic-rviz-1.14.10-1]$ buildPhase 
no Makefile, doing nothing

[nix-shell:~/ros-nix/rviz-release-release-noetic-rviz-1.14.10-1]$ 
`
how can i solve this issue ? or go to some documentation for understanding the nix ros build process ?

frett27, 2021-12-04:
Seems to be solved on my machine in : modifying the local file : 
/nix/store/sr0c5qvlq8rnlvjxgx2mwz4x6szc79n3-ros-noetic-urdf-1.13.2-r1/share/urdf/cmake/urdfConfig.cmake

2 modifications : 
there are wrong relative path to include : /nix/store/8cs76yqhlc2vnp1w4pp26z0aclx8l3w1-urdfdom-3.0.0/lib/urdfdom/cmake//../../../../../../include 

if changed the link to /nix/store/8cs76yqhlc2vnp1w4pp26z0aclx8l3w1-urdfdom-3.0.0/include
solve this problem, 
i also faced the non replacement of $<$CONFIG in the cmake template

all is fine with these 2 modifications in this file, and solved the compilation issue. Worked then as expected.

i don't know how to make this modification permanent, i'm starting learning nix.
Are there any means to integrate these compile in CI ?  urdf is used in quite all robotic components

lopsided98, 2021-12-04:
This was caused by the update to urdfdom 3.0.0, which has some CMake bugs. I forgot to revert the update in my nixpkgs fork after I discovered the problems. I've started work on a fix to submit upstream, but it's not ready yet.

lopsided98, 2021-12-04:
You may need to pass `--tarball-ttl 0` to Nix or delete `~/.cache/nix/tarballs` to get nixpkgs to update immediately.

closed 2021-12-04 by lopsided98

frett27, 2021-12-04:
hum ? i understood the nix compute de sha256 on the tarball, so if the overlay and pkgs changes the new sources will be downloaded ? i didn't cough your reply ?
in which case i do need to set 0 to TTL or remove the `.cache/nix/tarballs` ?

lopsided98, 2021-12-04:
This overlay impurely downloads my nixpkgs fork using `fetchTarball` by default. My instructions would invalidate the `fetchTarball` cache so you would get the fixed nixpkgs version immediately without waiting for the cache to expire.

frett27, 2021-12-04:
OK, got it , in default.nix : 
{ nixpkgs ? builtins.fetchTarball "https://github.com/lopsided98/nixpkgs/archive/nix-ros.tar.gz"
isn't that better to use a specific commit instead ?