lopsided98/nix-ros-overlay #160: husky-description fails to configure: Unknown CMake command "catkin_run_tests_target".

erdnaxe, 2022-01-07:
When trying to build nix-ros-overlay.rosPackages.noetic.husky-description (current master, https://github.com/lopsided98/nix-ros-overlay/commit/3f30908bc180ecab76d64f71b779c40f8b106949), I am getting the following CMake error:

```
these derivations will be built:
  /nix/store/qbzds5gvf7mh4sfpki1a9ljw4hy8nybs-ros-noetic-husky-description-0.6.0-r2.drv
building '/nix/store/qbzds5gvf7mh4sfpki1a9ljw4hy8nybs-ros-noetic-husky-description-0.6.0-r2.drv'...
/nix/store/ivxlqsbiwk59k3qvj1xmaz68myh19znj-ros-noetic-catkin-0.8.10-r1/etc/catkin/profile.d/05.catkin_make.bash: line 67: complete: command not found
/nix/store/ivxlqsbiwk59k3qvj1xmaz68myh19znj-ros-noetic-catkin-0.8.10-r1/etc/catkin/profile.d/05.catkin_make_isolated.bash: line 65: complete: command not found
unpacking sources
unpacking source archive /nix/store/7ak5d0x1r3x4ag8qx3wxqynzisdklkpx-0.6.0-2.tar.gz
source root is husky-release-release-noetic-husky_description-0.6.0-2
setting SOURCE_DATE_EPOCH to timestamp 1632837671 of file husky-release-release-noetic-husky_description-0.6.0-2/urdf/wheel.urdf.xacro
patching sources
configuring
fixing cmake files...
cmake flags: -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_BUILD_TYPE=Release -DCMAKE_SKIP_BUILD_RPATH=ON -DBUILD_TESTING=OFF -DCMAKE_INSTALL_LOCALEDIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/share/locale -DCMAKE_INSTALL_LIBEXECDIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/libexec -DCMAKE_INSTALL_LIBDIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/lib -DCMAKE_INSTALL_DOCDIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/share/doc/husky_description -DCMAKE_INSTALL_INFODIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/share/info -DCMAKE_INSTALL_MANDIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/share/man -DCMAKE_INSTALL_OLDINCLUDEDIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/include -DCMAKE_INSTALL_INCLUDEDIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/include -DCMAKE_INSTALL_SBINDIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/sbin -DCMAKE_INSTALL_BINDIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/bin -DCMAKE_INSTALL_NAME_DIR=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2/lib -DCMAKE_POLICY_DEFAULT_CMP0025=NEW -DCMAKE_OSX_SYSROOT= -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_STRIP=/nix/store/xiq6j4jsyj351p8q3yw9cg1hdqp9m685-gcc-wrapper-10.3.0/bin/strip -DCMAKE_RANLIB=/nix/store/lbxfixyw1yk099pjyaiy3xj5dl7kxm1g-binutils-2.35.2/bin/ranlib -DCMAKE_AR=/nix/store/lbxfixyw1yk099pjyaiy3xj5dl7kxm1g-binutils-2.35.2/bin/ar -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_INSTALL_PREFIX=/nix/store/j93ksz57xgfsnqnwsxkfwa9jmymlgqsf-ros-noetic-husky-description-0.6.0-r2  -DCATKIN_ENABLE_TESTING=OFF -DSETUPTOOLS_DEB_LAYOUT=OFF
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/xiq6j4jsyj351p8q3yw9cg1hdqp9m685-gcc-wrapper-10.3.0/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/xiq6j4jsyj351p8q3yw9cg1hdqp9m685-gcc-wrapper-10.3.0/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /build/husky-release-release-noetic-husky_description-0.6.0-2/build/devel
-- Using CMAKE_PREFIX_PATH: /nix/store/ivxlqsbiwk59k3qvj1xmaz68myh19znj-ros-noetic-catkin-0.8.10-r1;/nix/store/q2lj6pxd6sn75fa9pjx9x3jpgxnr6zld-cmake-3.21.2;/nix/store/l54hcf62vlwv014vnaf46pz5hzx89f0r-catkin-setup-hook;/nix/store/ja2c6vmg2xz40v6xxz0pjjvls3f1wx59-hook;/nix/store/6rnp1psxb8lhyrgfaif9hvs94anvs3hg-hook;/nix/store/xn9mcs2xjm0akxrrl49aj7l4ax1lbg1y-hook;/nix/store/nr8hvqfbkgb02jpk8l101pis23xvw0hg-gtest-1.11.0-dev;/nix/store/k50ib4p92giysavc76am4gjf3m6rzaj7-gtest-1.11.0;/nix/store/ghbj7zld6a2h3g2ygb81p9ljwd0mqk2l-python3.9-catkin_pkg-0.4.24;/nix/store/3iy7j0mli9sppxa7vlgwzlm7pzlf248x-python3.9-python-dateutil-2.8.2;/nix/store/whr9yl39g4q3vyqk16d76dkxmqqrx9ni-python3.9-six-1.16.0;/nix/store/k0z9n599k02hab8qjjp3ljw065iwjcvg-python3-3.9.6;/nix/store/rz3sgby9rwpf29av40rngl7gk8r7i70v-python3.9-docutils-0.17.1;/nix/store/2x5jhy3sx8za2skbf3v5ciq2dqcsg0rs-python3.9-pyparsing-2.4.7;/nix/store/5jrlh35xw2x784c8w7wchglhj26v1gvf-python3.9-empy-3.3.4;/nix/store/fbr6ycl5p04bwvfb05rhy2pyy9ii69ig-python3.9-nose-1.3.7;/nix/store/dzjk4nvlzcrr41416v1ayvw0rkq7pyfx-python3.9-coverage-5.5;/nix/store/fivxgjz8rplffmzijnw8sgqn9ynidrkl-python3.9-setuptools-57.2.0;/nix/store/53pwmhjbvv5vdnxw7ckh5ikjjqww2ayp-python3.9-cffi-1.14.6-dev;/nix/store/iz8h3a5yv9w2a12wlrf246v9iyc5qqcc-python3.9-pycparser-2.20;/nix/store/x6i5gxhhw5hmjmhg2d4h0x57dqbinvgh-python3.9-cffi-1.14.6;/nix/store/pjba17qx8vh7dln8m5vnkqdbbjahvg9b-patchelf-0.13;/nix/store/xiq6j4jsyj351p8q3yw9cg1hdqp9m685-gcc-wrapper-10.3.0;/nix/store/v7w0pspwq8r2b7k2sndxq3db843z7xm5-binutils-wrapper-2.35.2;/nix/store/73cj10x1jsiq8gp5b9aqcjlkrbyrzbdh-ros-noetic-roslaunch-1.15.13-r1;/nix/store/pq61nd7jbf6isc3193hy0zgldf21qvzc-python3.9-paramiko-2.7.2;/nix/store/6jaq6wg440307yvn84naikgr156g618r-python3.9-bcrypt-3.2.0;/nix/store/lzrlpwrad4k97krd3lc93aj2azkrxxhj-python3.9-cryptography-3.4.8-dev;/nix/store/lvpdqfcdln2lp444myyk8sij705bchn3-python3.9-packaging-20.9;/nix/store/ri618r0vw2p2skqq0s79x8lzh7yfcn9g-python3.9-cryptography-3.4.8;/nix/store/0dkppm1wgr4np2v7djc21gxf33zcnwrr-python3.9-pynacl-1.4.0;/nix/store/ps76bmwvgy213hjpx9r0bv33xif47mln-python3.9-pyasn1-0.4.8;/nix/store/aa0ir4vg2w35jaxsvwdm7nibi8bkh23f-python3.9-PyYAML-5.4.1.1;/nix/store/m5pc57jzz5ixqhsv7y618ajpx6ji5pqi-python3.9-rospkg-1.1.10;/nix/store/7wyd0kbwvh6qcdr6h8lwbfhrkcvrb71q-ros-noetic-rosclean-1.15.8-r1;/nix/store/x5i7y62s8z8d0hx0fl4hgph7yh8ix2bj-ros-noetic-rosgraph-msgs-1.11.3-r1;/nix/store/j1gp3jwfa9r3rsj7mx1dhg4xgskzlgd1-ros-noetic-message-runtime-0.4.13-r1;/nix/store/qm86w7mb8ybz04s0jgpl62b8rq635ii0-ros-noetic-cpp-common-0.7.2-r1;/nix/store/4db89kadal0gm3kws2p4dic1zqyvc8dc-boost-1.73.0-dev;/nix/store/4hfakp74xh6wi0dlb8ihna2wbgm0v4bp-boost-1.73.0;/nix/store/8jsbl9lzwfh28440hhzryf4z5j09hdw3-console-bridge-1.0.1;/nix/store/ij9dczh77y9zm66dm2pc9y5xahn7iw2g-ros-noetic-genpy-0.6.15-r1;/nix/store/6v240z6174dlasz9xjynvm8pq9jragbj-ros-noetic-genmsg-0.5.16-r1;/nix/store/3wgx40p3w0hhxqykhxs9ybl15ff8xff2-ros-noetic-roscpp-serialization-0.7.2-r1;/nix/store/1gidgybvyqlj2kv2m8140ncnrswk0him-ros-noetic-roscpp-traits-0.7.2-r1;/nix/store/aazldnp1mp1zkb3l2fva5yj1yld6g53c-ros-noetic-rostime-0.7.2-r1;/nix/store/sxllrq38wyk95mbmzi2q3i83vxq7agzn-ros-noetic-std-msgs-0.5.13-r1;/nix/store/531cxsfwgy1hxvngv9x4cz4l13pj9f9h-ros-noetic-roslib-1.15.8-r1;/nix/store/a1rri8xwjs6n1al80wgrzb38sic187ln-ros-noetic-ros-environment-1.3.2-r1;/nix/store/jbnlg3z0qsdymgr7qp8y7ip5npfiifwh-ros-noetic-rospack-2.6.2-r1;/nix/store/d3c4y6ssahavbjgisvvldc9kdnfjryyk-pkg-config-wrapper-0.29.2;/nix/store/lby6miz8gww41fjjmcns4bip19w2ricd-python3.9-rosdep-0.17.1;/nix/store/0iq16bwhhl45xs9ikv0as8qwcih7kh5h-python3.9-rosdistro-0.7.5;/nix/store/18liax2xxa7zrgnqmb54kpfhga35znh1-tinyxml-2-6.0.0;/nix/store/zj5lf88l8g97vwar6p8xdwkwh6ixznxf-ros-noetic-rosmaster-1.15.13-r1;/nix/store/1740kyhy48xgdr90s18lj9wpnzarn81n-python3.9-defusedxml-0.7.1;/nix/store/1dmmnlw26j2vz40w0xdwa469dksb0piy-ros-noetic-rosgraph-1.15.13-r1;/nix/store/r8xcz5bxln4r2pmn1p4ygawb9alzdy3g-python3.9-netifaces-0.11.0;/nix/store/5k2cjw19jxbqjw86k5vz3msf7mird70z-ros-noetic-rosout-1.15.13-r1;/nix/store/r5g57llsp0v0b295xkjbja03bxh2j21d-ros-noetic-roscpp-1.15.13-r1;/nix/store/zgg8dr9isv9a9xikh03sikagwk6c78p5-ros-noetic-rosconsole-1.14.3-r1;/nix/store/bd1pr57ys2yqaabpsp6yskflrnx1sypd-apr-1.7.0-dev;/nix/store/n0p6xr2b42lf2vfyf4zi56rs999a6 [... cut ...]
See also "/build/husky-release-release-noetic-husky_description-0.6.0-2/build/CMakeFiles/CMakeOutput.log".
builder for '/nix/store/qbzds5gvf7mh4sfpki1a9ljw4hy8nybs-ros-noetic-husky-description-0.6.0-r2.drv' failed with exit code 1
error: build of '/nix/store/qbzds5gvf7mh4sfpki1a9ljw4hy8nybs-ros-noetic-husky-description-0.6.0-r2.drv' failed
```

I am new to catkin and I believe there is an issue with this package configuration (https://github.com/clearpath-gbp/husky-release/blob/release/noetic/husky_description/0.6.0-2/CMakeLists.txt). Have you already seen something similar?

erdnaxe, 2022-01-07:
I believe that this issue is solved by https://github.com/husky/husky/pull/202. Is there a reason why nix-ros-overlay does not set `CATKIN_ENABLE_TESTING` during building?

lopsided98, 2022-01-07:
Tests often assume things about the environment that are not true with Nix. There are so many packages that it is infeasible to fix all these issues, so I disable the tests. Unfortunately, some packages break with tests disabled, but that is usually easier to fix.

erdnaxe, 2022-01-07:
Thank you for the explanation. It makes sense.

> Unfortunately, some packages break with tests disabled, but that is usually easier to fix.

Would it be possible to manually activate tests on some of nix-ros-overlay packages, or do you prefer to wait for upstream to release the fix?

lopsided98, 2022-01-07:
Normally, I would apply the patch in the overlay, but in this case the patch is difficult to apply (needs to be filtered then stripped, but `fetchpatch` does strip then filter). Since the tests pass in this case, the best solution seems to be to just enable them.

closed 2022-01-07 by lopsided98