lopsided98/nix-ros-overlay #163: Missing commit

lelongg, 2022-01-18:
I was depending on https://github.com/lopsided98/nixpkgs/tarball/cb84e61c57ad07123a977e124b55348201ad4b39 and this commit does no longer exist. Is there a way to retrieve it ? Does someone have it lying around on his computer ? Would be really helpful

lopsided98, 2022-01-18:
Sorry, it seems to have been garbage collected on my local repo as well.

lelongg, 2022-01-18:
Argh too bad... I can only cry in despair now... Thanks for your time !

closed 2022-01-18 by lelongg

lopsided98, 2022-01-18:
I was trying to figure out how old it was, which would perhaps allow you to figure out which upstream nixpkgs commit it was based on. I looked through my Hydra evaluations, but none of them use that commit.

lelongg, 2022-01-18:
The commit in my repo that set it was from Aug 11, 2020