lopsided98/nix-ros-overlay #196: Packages don't build due to Setuptools and Distutils collision

hollmmax, 2022-07-28:
Some packages in `noetic` don't build with the following error:
```
error: invalid command '<pkg>'
CMake Error at /nix/store/<catkin_store>/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  
  execute_process(/tmp/<pkg_derivation>/build/catkin_generated/env_cached.sh
  "/nix/store/<python3_store>/bin/python3"
  "/nix/store/<catkin_store>/share/catkin/cmake/interrogate_setup_dot_py.py"
  "<pkg>"
  "/tmp/<pkg_derivation>/setup.py"
  "/tmp/<pkg_derivation>/build/catkin_generated/setup_py_interrogation.cmake")
  returned error code 1
Call Stack (most recent call first):
  /nix/store/<catkin_store>/share/catkin/cmake/catkin_python_setup.cmake:46 (safe_execute_process)
  CMakeLists.txt:23 (catkin_python_setup)
 ```
 (Nix paths redacted for readability's sake)

The error also includes a warning from python suggesting a fix:
```
UserWarning: Distutils was imported before Setuptools, but importing Setuptools also replaces the `distutils` module in `sys.modules`. This may lead to undesirable behaviors or errors. To avoid these issues, avoid using distutils directly, ensure that setuptools is installed in the traditional way (e.g. not an editable install), and/or make sure that setuptools is always imported before distutils.
```

Replacing `distutils.core` with `setuptools` in `setup.py` as recommended fixes the problem.

Packages encountered so far:
- angles
- roslint
- laser-geometry
- resource-retriever

cc  @wentasah

lopsided98, 2022-08-01:
I think I already fixed this in staging. I submitted a patch upstream to import setuptools before distutils (https://github.com/ros/catkin/pull/1176), but I haven't followed up on it because it turns out that the real problem is related to how setuptools is packaged in nixpkgs. setuptools' monkey patching still works correctly on other distros even if you import distutils before setuptools.

hollmmax, 2022-08-03:
I can't tell what change fixed it, but it does work in staging.
Thank you for the fix, closing.

closed 2022-08-03 by hollmmax

wentasah, 2022-08-03:
I think it's [this commit](https://github.com/lopsided98/nix-ros-overlay/commit/6b09de5c1acde28f358a052bcbf9eb7cb481cdce) :-)