lopsided98/nix-ros-overlay #199: Turtlebot shell fails to build due to gazebo

wcole3, 2022-08-04:
The shell in the example script fails to build with the errors resulting from compiling at c++ 14 standard.  By pulling down the source and manually adding `cmakeFlags = [ "-DCMAKE_CXX_STANDARD=17" ];` to the default.nix files in distros/noetic/gazebo-plugins and distros/noetic/gazebo-ros allows the command `nix-shell examples/turtlebot3-gazebo.nix` to correctly build the shell.  

Launching turtlebot works as expects from that shell.  Not sure how to address this, but we have be able to reproduce this on a few different machines.

TilCreator, 2022-08-15:
Can reproduce, the fix also works! Can you please open a PR for this?

lopsided98, 2022-08-17:
Updating Gazebo and its dependencies fixed this.

closed 2022-08-17 by lopsided98