lopsided98/nix-ros-overlay #205: [missing package] asr-mild-base-laserscanner

TilCreator, 2022-09-01:
I want to use the `asr_mild_base_laserscanner` package and can't quite figure out why it's missing. My best guess is that `tf` is not building because `bcrypt` is not available for `python2` anymore?
What would be a good way of getting that package to work? As far as I read the `tf` package should just be a compatibility layer pointing to `tf2` which builds?

(sry, I'm quite new to ROS)

lopsided98, 2022-09-02:
That package doesn't have a release repository published in rosdistro, only documentation: https://github.com/ros/rosdistro/blob/d5e4803ce937ac83f5ece3dc35427bae9cd82616/melodic/distribution.yaml#L431. I don't think this package is available in Ubuntu either.

TilCreator, 2022-09-02:
I found it on ROS index: https://index.ros.org/p/asr_mild_base_laserscanner/github-asr-ros-asr_mild_base_laserscanner/#melodic
And thought that it should be available here as well then.
How would I go about making that package available?

lopsided98, 2022-09-02:
I think that appears because there is a `doc` key in the YAML, but the maintainers of the package need to release it with bloom. Unfortunately, that project looks unmaintained, so that probably isn't going to happen.

You can also either build it yourself as part of a ROS workspace (not managed by Nix), or manually package it with Nix.

closed 2022-10-19 by lopsided98