lopsided98/nix-ros-overlay #208: [missing package] moveit-commander

demathelin, 2022-09-23:
Hi ! When I put "franka-ros" in my buildInputs, I have this error:
```
nix-shell --pure
error: Function called without required argument "moveit-commander" at /nix/store/haxaf2yfhb6ail4vjc7i5chl3i6sdspb-source/distros/noetic/franka-example-controllers/default.nix:5
(use '--show-trace' to show detailed location information)
```

Here my default.nix:

```nix

let

pkgs = (import (builtins.fetchTarball {
   
    url = "https://github.com/NixOS/nixpkgs/archive/nixos-21.11.tar.gz";
  }) {}).pkgs ;

ros_overlay = ((import (builtins.fetchGit{
    url="https://github.com/lopsided98/nix-ros-overlay.git";
    rev="7559a8cf6ef1ee4324d9b0857b24c89475786ab5";
    ref="staging";
}){}));

in

pkgs.stdenv.mkDerivation rec {
	name = "ros_test";
  buildInputs = [
	ros_overlay.rosPackages.noetic.catkin
	pkgs.git
	ros_overlay.pythonPackages.rosdep
	ros_overlay.rosPackages.noetic.roscpp
	pkgs.eigen

  ros_overlay.rosPackages.noetic.actionlib-msgs 
  ros_overlay.rosPackages.noetic.control-msgs 
  ros_overlay.rosPackages.noetic.actionlib
  ros_overlay.rosPackages.noetic.sensor-msgs
  ros_overlay.rosPackages.noetic.franka-ros
  ];

  
}
```

Thanks for all your work !

demathelin, 2022-09-23:
I have checked that it was solved in [this commit] (https://github.com/lopsided98/nix-ros-overlay/commit/1cb25a71b48e59098a7c612c6d141b1707c1a728)

closed 2022-09-23 by demathelin

WinstonHartnett, 2023-11-10:
I ran into this again with 5f92658. `moveit` fails to build with:

```
       error: evaluation aborted with the following error message: 'Function called without required argument "moveit-commander" at /nix/store/dc7sviyvcgif1ss22kgx5pvvx28dks32-source/distros/noetic/moveit/default.nix:5'
```

wentasah, 2023-11-10:
That's because [pyassimp](https://pypi.org/project/pyassimp/) (`python3-pyassimp` in ROS) is not packaged in Nixpkgs. Somebody should package it. You can work around it, by packaging it yourself and adding that package to your nixpkgs overlay. However, fixing the problem for everybody by maintaining the package in `nixpkgs` would be better.