lopsided98/nix-ros-overlay #218: install from repo via flakes

brothermechanic, 2022-10-12:
Hello
Help me as new user, how to install packages via flakes
(via channels works, but i need flakes instead)
i get this error
```
$ nix profile install github:lopsided98/nix-ros-overlay#gazebo
error: flake 'github:lopsided98/nix-ros-overlay' does not provide attribute 'packages.x86_64-linux.gazebo', 'legacyPackages.x86_64-linux.gazebo' or 'gazebo'
```

koalp, 2022-10-12:
Hello. 
I think that you need to specify which ros distribution you want to use.

As an example, I use
```
nix shell -L github:lopsided98/nix-ros-overlay#humble.gazebo --command gazebo
```
to run gazebo from the `humble` distribution.

brothermechanic, 2022-10-13:
> Hello. I think that you need to specify which ros distribution you want to use.
> 
> As an example, I use
> 
> ```
> nix shell -L github:lopsided98/nix-ros-overlay#humble.gazebo --command gazebo
> ```
> 
> to run gazebo from the `humble` distribution.

Cool!
It works! Tank you!

lopsided98, 2022-10-14:
I need to fix this for real though, it just happens to work in this case since gazebo is part of the top-level package set and the distro specific package set. With other packages like colcon, there isn't a work-around, see: https://github.com/lopsided98/nix-ros-overlay/issues/150

closed 2022-10-14 by lopsided98