lopsided98/nix-ros-overlay #220: Mixing ROS1 and ROS2 distros?

PAJohnson, 2022-10-19:
I'm working on a codebase that's a mix of ROS1 and ROS2. I'd like to use packages like [ros1_bridge](https://github.com/ros2/ros1_bridge) or [slim_bridge](https://github.com/blackcoffeerobotics/slim_bridge) with this overlay. Both of those packages require a ros1 and ros2 installation to be available at compilation time. 

Is there a way to accomplish that with this repo?

lopsided98, 2022-10-19:
I haven't tried, but you might be able to do this by overriding the bridge package to add the relevant ROS 1 packages to its `buildInputs`.

PAJohnson, 2023-07-11:
Your suggestion worked. For reference, this modification of `ros2-basic.nix` allowed me to run Example 1 from the ros1_bridge readme:

```nix
# Example of using two ROS distros in the same environment

{ pkgs ? import ../. {} }:
with pkgs;
with rosPackages.foxy;

let
  ros1-deps-for-bridge = (buildEnv {
      paths = [         
        rosPackages.noetic.rosmsg
        rosPackages.noetic.ros-environment
        rosPackages.noetic.rosbash
        rosPackages.noetic.roslaunch
        rosPackages.noetic.rospy-tutorials
      ];
    });

  ros1-bridge-fixed = ros1-bridge.overrideAttrs ({
    propagatedBuildInputs ? [], ...
  }: {
    propagatedBuildInputs = propagatedBuildInputs ++ [ ros1-deps-for-bridge ];
    version = "b30614c";
    src = fetchFromGitHub {
      owner = "ros2";
      repo = "ros1_bridge";
      rev = "b30614c";
      hash = "sha256-S9ZN3719WVtNGMhhhJ8EBc7L3N1x7jL0Udiwq9FCmvM=";
    };
  });
in
mkShell {
  RMW_IMPLEMENTATION = "rmw_fastrtps_dynamic_cpp";
  nativeBuildInputs = [
    (buildEnv {
      paths = [
        ros-environment
        ros2run
        ros1-bridge-fixed
        rmw-fastrtps-dynamic-cpp
        demo-nodes-cpp
      ];
    })
  ];
}
```

The src override for the bridge was necessary to fix some compilation issues related to static assertions in the bridge codebase.

closed 2023-07-11 by PAJohnson