lopsided98/nix-ros-overlay #222: `nix shell .#humble.ros2cli .#humble.ros2run` does not make `ros2 run` available.

kjeremy, 2022-11-02:
I would expect that the above would make the `ros2 run` command available on the shell but it does not.

```sh
jkolb@nixos:~/ara/nix-ros-overlay]$ nix shell .#humble.ros2cli .#humble.ros2run

[jkolb@nixos:~/ara/nix-ros-overlay]$ ros2 run -h
usage: ros2 [-h] [--use-python-default-buffering] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'run' (choose from 'daemon', 'extension_points', 'extensions')

```

lopsided98, 2022-11-06:
`ros2cli` finds `ros2run` and other command packages dynamically using environment variables, which are set by setup hooks. `nix shell` doesn't run setup hooks, so I don't know how I could make this work. I could wrap the `ros2` command to include a hardcoded set of common commands, but this wouldn't solve the general case. `buildEnv` will also generate wrappers to allow a set of packages to all find each other without setup hooks, but this doesn't work with `nix shell`.

lopsided98, 2022-11-06:
This problem is not specific to ROS; Python libraries don't work with `nix shell` either:
```
$ nix shell .#python3 .#python3.pkgs.numpy
$ python3
Python 3.10.6 (main, Aug  1 2022, 20:38:21) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'numpy'
>>> 
```

kjeremy, 2022-11-06:
What's the best way to run these commands then?

lopsided98, 2022-11-06:
You pretty much have to use a shell file like this: https://github.com/lopsided98/nix-ros-overlay/blob/staging/examples/ros2-basic.nix. If you want to use flakes, this shell file has to be part of your own flake, as a `devShell` output.

Unfortunately, flakes have no real equivalent to `nix-shell -p`.

kjeremy, 2022-11-11:
Somewhat related. I have the following which has a ROS1 equivalent and works out of the box like `nix run mypkg.run mypkg_node` as an example. In ROS2 this does not work. Is there a recommended way to combine the `mkShell` with a `writeShellScriptBin` in the below code?

```nix
# Trivial helpers for ROS 2 packages.  These get jammed into each package's
# passthru as shorthand for running nodes/launch files inside minimal
# environments.

{ buildEnv
, ros2cli
, ros2run
, ros2launch
, ros-environment
, writeShellScriptBin
, pkg
}:
{
  # Doesn't work. See: https://github.com/lopsided98/nix-ros-overlay/issues/222
  run = let env = buildEnv { paths = [ ros-environment ros2cli ros2run pkg ]; }; in
    writeShellScriptBin "run" ''exec ${env}/bin/ros2 run '${pkg.pname}' "$@"'';

  launch = let env = buildEnv { paths = [ ros-environment ros2launch pkg ]; }; in
    writeShellScriptBin "launch" ''exec ${env}/bin/ros2launch '${pkg.pname}' "$@"'';
}
```

lopsided98, 2022-11-11:
I may have fixed this in master, could you test it again?

kjeremy, 2022-11-14:
Yes this now works for me! with 

```nix
  run = let env = buildEnv { paths = [ ros2run pkg ]; }; in
    writeShellScriptBin "run" ''exec ${env}/bin/ros2 run '${pkg.pname}' "$@"'';

  launch = let env = buildEnv { paths = [ ros2launch pkg ]; }; in
    writeShellScriptBin "launch" ''exec ${env}/bin/ros2 launch '${pkg.pname}' "$@"'';
```

kjeremy, 2022-11-14:
The above now works but it still fails on the command line

lopsided98, 2023-04-08:
Could you provide more detail if this is still an issue?

closed 2024-04-07 by lopsided98