lopsided98/nix-ros-overlay #227: rviz does not build in noetic and melodicPython3

lelongg, 2022-11-22:
rviz fails to build with the following error

```
-- Found PythonInterp: /nix/store/xcaaly5shfy227ffs8nipxrd49b56iqq-python3-3.10.8/bin/python3 (found suitable version "3.10.8", minimum required is "3.10") 
CMake Warning at /nix/store/9fcvq10d1i5xy2v5gkhjyycv322w94j2-ros-noetic-python-qt-binding-0.4.4-r1/share/python_qt_binding/cmake/sip_helper.cmake:28 (message):
  SIP binding generator NOT available.
Call Stack (most recent call first):
  src/python_bindings/sip/CMakeLists.txt:42 (include)

-- Python binding generators: 
CMake Error at src/python_bindings/CMakeLists.txt:9 (message):
  No Python binding generator found.

-- Configuring incomplete, errors occurred!
``` 

reproducible with `nix build github:lopsided98/nix-ros-overlay#noetic.rviz `

lopsided98, 2022-12-04:
There's not much I can do about this. The latest versions of Qt5 are no longer compatible with SIP 4, so I switched to Shiboken/PySide2 (see https://github.com/lopsided98/nix-ros-overlay/issues/224). This fixed rqt, but rviz is not compatible with PySide2, see: https://github.com/ros-visualization/rviz/pull/1526

natto1784, 2022-12-18:
Thanks for maintaining all of the distros! Do you suggest anything that might help with building rviz? Maybe overriding qt5.qtbase?

natto1784, 2022-12-22:
Some issues you can follow: https://github.com/ros-visualization/rviz/issues/1777 https://github.com/ros-visualization/python_qt_binding/issues/121

kjeremy, 2023-02-07:
I'm running into this too. Maybe we can pin rviz to an older version from nix-ros-overlay?

natto1784, 2023-02-07:
Using [this patch](https://aur.archlinux.org/cgit/aur.git/tree/restore-sip4-support.patch?h=python-pyqt5-sip4) along with the old python-qt-binding override works with SIP4. I tested it as a global overlay for all distros and rviz compiled! Let's see if we can use this here for noetic and melodic specifically while using shiboken/pyside2 for other distros.

closed 2023-02-26 by lopsided98