lopsided98/nix-ros-overlay #230: ros2 run does not work with python packages.

rshanor, 2022-12-12:
I installed `rqt-robot-monitor` into a nix shell, but I am unable to run the executable.

- `ros2 pkg list | grep rqt_robot_monitor` shows that the package is expected.
- `ros2 run rqt_robot_monitor rqt_robot_monitor` reports `No executable found`
- I am able to run the executable using `python3 -m rqt_robot_monitor.robot_monitor`

I observer this behavior with multiple python packages.

Happy to help if you can point me in the right direction, thanks!

lopsided98, 2022-12-12:
There appear to be three problems here:

1. `ament_python` packages don't install `local_setup.sh` scripts, which prevents them from appearing in `AMENT_PREFIX_PATH` if `buildEnv` is not used. I suspect that no one considered the possibility that ament_python packages would be installed into a prefix all by themselves.
2. If `buildEnv` is used, the package is in `AMENT_PREFIX_PATH` (because the environment directory is included), but `ros2 run` fails because the executable is in `bin` instead of `lib/rqt_robot_monitor`. [`setup.cfg` appears to tell setuptools to install to `lib/rqt_robot_monitor`](https://github.com/ros-visualization/rqt_robot_monitor/blob/069f44ff20e13274364f3f569decea0de3e5713a/setup.cfg#L4), but for some reason this doesn't work with Nix.
3. If you just run the binary in `bin`, it fails due to missing Qt wrapping. This is easy to fix.

rshanor, 2022-12-13:
Hmm, when I try to build anything inside my nix shell, I get errors like `/nix/store/rlq70vd42nadlgyi5il7nzs9sa3k1rjz-ros-rolling-lifecycle-msgs-1.3.1-r1' in the environment variable AMENT_PREFIX_PATH doesn't contain any 'local_setup.*' files.`. This seems to be true for both `ament_python` and `ament_cmake` packages.

To get around #2, I could do something hacky like copy from bin to lib as a postInstall. Otherwise I am kind of lost where to look.

lopsided98, 2022-12-13:
That warning is harmless and kind of bogus. It occurs because `.colcon_install_layout` doesn't exist (because the packages weren't installed with colcon), and doesn't have anything to do with the `local_setup.*` files.

hacker1024, 2023-01-11:
> To get around https://github.com/lopsided98/nix-ros-overlay/pull/2, I could do something hacky like copy from bin to lib as a postInstall. Otherwise I am kind of lost where to look.

Do you have an example of this working?

```nix
postInstall = ''
  cp -r $out/bin/* $out/lib/
'';
```

```
$ ros2 run my_package my_executable
No executable found
```

hacker1024, 2023-06-30:
> [setup.cfg appears to tell setuptools to install to lib/rqt_robot_monitor](https://github.com/ros-visualization/rqt_robot_monitor/blob/069f44ff20e13274364f3f569decea0de3e5713a/setup.cfg#L4), but for some reason this doesn't work with Nix.

Should this be brought up in the Nixpkgs issue tracker?

wentasah, 2023-06-30:
I noticed this problem as well and investigated the reason. `colcon` builds python packages by calling:

    setup.py egg_info ... build ... install ...

whereas `buildPythonPackage` (used by `ament_python` packages in nix-ros-overlay) calls:

    setup.py bdist_wheel

and then installs the wheel. This procedure ignores the content of `setup.cfg`. Not sure why. I guess that bringing it to nixpkgs Python maintainers would be useful. Would you file the issue?

I have a half finished patch for nix-ros-overlay, which does the above `postInstall` automatically for all `ament_python` packages. I'll try to clean it up and post a PR here. If nixpkgs maintainers have a better solution, we might adopt it later.

wentasah, 2023-06-30:
I prepared #270. Currently, it's marked as draft, because I did only a few quick tests with it. Once I rebuild my projects with this change without problems, I'll mark it as ready.

Elston10, 2023-07-02:
i created a ros2 package and i ran the command ros2 run my_py_pkg py_node where py_node is the executable in linux shell but it is showing that no executables are found

Elston10, 2023-07-02:
from setuptools import setup
f
package_name = 'my_py_pkg'

setup(
    name=package_name,
    version='0.0.0',
    packages=[package_name],
    data_files=[
        ('share/ament_index/resource_index/packages',
            ['resource/' + package_name]),
        ('share/' + package_name, ['package.xml']),
    ],
    install_requires=['setuptools'],
    zip_safe=True,
    maintainer='victus',
    maintainer_email='victus@todo.todo',
    description='TODO: Package description',
    license='TODO: License declaration',
    tests_require=['pytest'],
    entry_points={
        'console_scripts': ['py_node=my_py_pkg.my_first_node:main'
        ],
    },
) this is my setup.py file 
[develop]
script_dir=$base/lib/my_py_pkg
[install]
install_scripts=$base/lib/my_py_pkg
this is my setup.cfg file

PR #272 (2023-09-07): Build ROS Python packages like colcon
[It seems](https://github.com/colcon/colcon-core/blob/99c03262ad7edd4af5954c424c9248e0c4fe7307/colcon_core/task/python/build.py#L91) that colcon uses the deprecated `setup.py`-style build and installation commands to build Python ROS packages.

This changes the `ament_python` variant of `buildRosPackage` to use those commands.

Fixes #230, #273.
Closes #270.

Tested with `ros-core` from Rolling, Foxy and Noetic.
closed 2023-09-07 by lopsided98