lopsided98/nix-ros-overlay #231: Adding Python module packages

hacker1024, 2022-12-13:
One of my ROS packages requires Tkinter. Is there a way to add packages such as these to the Python environment used by ROS?

lopsided98, 2022-12-13:
Yes, it should mostly work the same as anything else in Nix. What have you tried?

hacker1024, 2022-12-15:
Well, normally if I want to add packages to a Python derivation, I'd use the `withPackages` function. It's not clear to me how to modify the Python derivation used by ROS, however. Does it just use a Python derivation from Nixpkgs? If so, an overlay can be used, but for which version?

lopsided98, 2023-01-29:
Yeah, `withPackages` expects you to run to only run the Python binary in the environment it returns, which won't work with ROS. I normally just add Python packages directly to the shell `buildInputs` or the ROS `buildEnv`, without using `withPackages`.

hacker1024, 2023-01-30:
Thanks, that works well.

closed 2023-01-30 by hacker1024