lopsided98/nix-ros-overlay #238: ROS packages with nixGL requirement

lvjonok, 2023-01-27:
Thank you for your tool! I am discovering my possibilities to have ROS workspace and I have some problems.

Since I am not _nixOS_ user, I am required to use [nixGL](https://github.com/guibou/nixGL). Despite such commands work as expected ```nix run github:guibou/nixGL#nixGLIntel -- gzclient --verbose```.
I have no idea how to get it working with ```ros2 launch ...```. I looked down source files and **gazebo_ros** package during its launch simply executes **gzserver** and **gzclient**. I believe it is not worthy to change source files of gazebo package each time.

What could you suggest as a way to run with such prefix?

lvjonok, 2023-01-27:
Update:

it works even with ```nixGL gazebo```, and as simplification I used ```alias gazebo="nixGL gazebo"```. Probably now I need just help to auto alias in my **default.nix** file

lopsided98, 2023-01-27:
Have you tried just running `nixGL ros2 launch ...`? The `nixGL` wrappers just set some environment variables, so I think that should work unless `ros2 launch` resets the environment.

closed 2023-02-16 by lvjonok