lopsided98/nix-ros-overlay #240: [gazebo] gazebo in rolling  error while loading libsodium.so

brothermechanic, 2023-01-28:
Hi
I have an error with gazebo

```
$ nix profile install github:lopsided98/nix-ros-overlay#rolling.gazebo
$ nixGL gazebo
/nix/store/6izs9zfkhbrqbil6vkl9zaxf9mnhmzhr-gazebo-11.11.0/bin/.gazebo-wrapped: error while loading shared libraries: libsodium.so.23: cannot open shared object file: No such file or directory
```

movefasta, 2023-01-28:
this solution might be useful:
```
devShells.ros = 
            pkgs.mkShell {
              buildInputs = (with ros; [ ros2run rmw-fastrtps-dynamic-cpp ]);
              RMW_IMPLEMENTATION = "rmw_fastrtps_dynamic_cpp";
              shellHook = ''
                export LD_LIBRARY_PATH=${(with pkgs; lib.makeLibraryPath [ libsodium ])}
              '';
            };
```

lopsided98, 2023-01-29:
Probably https://github.com/lopsided98/nix-ros-overlay/pull/228, I guess I should just merge that.

closed 2023-02-05 by brothermechanic

lopsided98, 2023-04-08:
I've confirmed that #228 fixes this issue.