lopsided98/nix-ros-overlay #241: Multiple Mismatched Checksums

jeff-hykin, 2023-01-30:
For almost all of the packages I'm getting something like the following:

```
trying https://github.com/ROBOTIS-GIT-release/turtlebot3-release/archive/release/noetic/turtlebot3_description/1.2.5-1.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 6609k    0 6609k    0     0  2679k      0 --:--:--  0:00:02 --:--:-- 4201k
error: hash mismatch in fixed-output derivation '/nix/store/kda2h88qffxarhizkx1qzqwwas3d7fss-1.2.5-1.tar.gz.drv':
         specified: sha256-tCIoOVLRiPSrvfgzCeVAA/P2LSvvvCEWQduFoyZWlgs=
            got:    sha256-EmXI+RGeY9SfMyjADcaUI+YHpEdlxPrvFUxmY2osw94=
error: 1 dependencies of derivation '/nix/store/cwcmvkrig1dhqqd2gj1ds2gwil63vk0s-ros-noetic-turtlebot3-descript
```

I created a fork and patched some of the hashes, but after about 7 of them I realized something bigger is probably wrong for this to be happening

lopsided98, 2023-01-30:
Yeah, its this: https://github.blog/changelog/2023-01-30-git-archive-checksums-may-change/, although I don't know why its still happening even though they apparently reverted it.

It really is my fault though, as you are supposed to use `fetchFromGitHub` to avoid this exact problem, but it was a bit more difficult to properly generate the code for it with superflore so I never implemented it.

lopsided98, 2023-01-30:
I'm going to leave things as they are for now, since the change is being reverted. When I get a chance I'll try to implement `fetchFromGitHub`, but no guarantees on when that will happen.

jeff-hykin, 2023-01-31:
> When I get a chance I'll try to implement fetchFromGitHub, but no guarantees on when that will happen.

No worries, I might be able to. Although I'm not sure how the current nix files are being generated. I see the `superflore-gen-nix` but I have no idea where the source code from that command is coming from since the [superflore repo](https://github.com/lopsided98/superflore) doesn't mention anything like `fetchurl` or `superflore-gen-nix`. 

> https://github.blog/changelog/2023-01-30-git-archive-checksums-may-change/

Wow I'm honeslty suprised more of the nixpkgs stuff I use isnt breaking. I always grab straight from tar files.

On my fork I've updated hashes for ~600 files using a script, but that was before I knew tar files were the problem here. I can probably just use the same script to switch them all to `fetchFromGithub`. If you're generating the files though, the script I made is a pretty hacky solution so I doubt it would work long term.

lopsided98, 2023-01-31:
The entry point is here: https://github.com/lopsided98/superflore/blob/nixos-support/superflore/generators/nix/run.py

jeff-hykin, 2023-01-31:
Thanks! I went ahead and made [a fork](https://github.com/jeff-hykin/superflore/) that adds `fetchFromGithub` when its possible to, but otherwise keeps the old functionality. It does depend on `nix-prefetch`, which is the only reliably way I know to get the sha256 for `fetchFromGithub`.

I did some dryrun's and everything seems to be working (at least as much as it was before); I confirmed the outputs include `fetchFromGithub` and work. The generator is skipping a lot of packages though, so I wonder if I need to force a reevaluation of all packages.

I'm not sure about the AUTH token stuff, it seems to me like the dry run is good enough for me to generate nix-ros-overlay.

wentasah, 2026-08-29:
I think, these mismatched checkums are gone for long time. Closing.

closed 2026-08-29 by wentasah