lopsided98/nix-ros-overlay #243: Building `mcap-vendor` fails

koalp, 2023-02-04:
In the [mcap_vendor cmake](https://github.com/ros2/rosbag2/blob/8044dba28aa7f99247ccf29861f349969be07117/mcap_vendor/CMakeLists.txt#L35), an external source is fetched and it makes the nix build fail.

Do you know what the best way to fix it is ?

### reproduce 
```
nix build github:lopsided98/nix-ros-overlay\#humble.mcap-vendor
```

movefasta, 2023-03-22:
To reach purity of build evaluation nix disables networking access during build time. Fetching remote resources while building makes build not reproducible.

You can do following:
1) Create package mcap in nix-ros-overlay non-ROS packages (or in nixpkgs, it is more recommended way) and contribute it
2) Create git patch for CmakeLists.txt file to disable fetching or write in CmakeLists condition for fetching cmake code block with custom environment variable (for example, DISABLE_FETCH=0 by default) and contribute it to mcap_vendor repo
3) Override mcap-verdor package attributes in nix-ros-overlay with mcap and lz4 in nativeBuildInputs and environment variable DISABLE_FETCH=1

@lopsided98 can correct me if this way is wrong

lopsided98, 2023-03-27:
I just fixed this specific case, but for future reference, my solution uses the `patchVendorUrl` and `patchVendorGit` functions (search the repo to find many examples) to fetch the resources and patch `CMakeLists.txt` to fetch from a store path instead. Unfortunately, this requires manual updates each time the vendor package is updated.

closed 2023-03-27 by lopsided98