lopsided98/nix-ros-overlay #244: Cmake cannot find `tf2-kdl`, it's not in propagatedBuildInputs

movefasta, 2023-02-06:
Hello.
I tried to build moveit-core on Humble and got an issue with cmake, that cannot found package
```
nix log /nix/store/vrwn8d3ligjzf3zqa1cslyzca4l2hri3-ros-humble-moveit-core-2.5.4-r1.drv
...
CMake Error at CMakeLists.txt:31 (find_package):
  By not providing "Findtf2_kdl.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "tf2_kdl", but
  CMake did not find one.
```

To debug this i'm run nix shell with `inputsFrom = [ ros-humble.moveit-core ];` attribute and found that tf2-kdl is not presented in propagatedBuildInputs. Meanwhile another moveit-core dependencies (like tf2-eigen, that cmake successfully found) are defined there:

```
$ env | grep propagatedBuildInputs
propagatedBuildInputs=... /nix/store/r63apzy6wbbq089sl28abhipwjvvznip-ros-humble-std-msgs-4.2.3-r1
/nix/store/y47bkw473bih729lb4b8bzkgsmf024d3-ros-humble-tf2-0.25.2-r1
/nix/store/q0rbsbyjd90c3vsiigd8cc3gzscq09qb-ros-humble-tf2-eigen-0.25.2-r1
/nix/store/p1cxzjc233jwphvcwdbivbgvbp4cz3vc-ros-humble-tf2-geometry-msgs-0.25.2-r1 ...
```

According to [humble/moveit-core/default.nix#L18](https://github.com/lopsided98/nix-ros-overlay/blob/staging/distros/humble/moveit-core/default.nix#L18), `tf2-kdl` is in `checkInputs` and missing in `propagatedBuildInputs`.

movefasta, 2023-02-06:
it was fixed in
https://github.com/ros-planning/moveit2/pull/1817

movefasta, 2023-02-06:
Well, issue was fixed, but fix was not included to releases yet. What should i do? Overlay nix-ros-ovelay with master branch of moveit-core?

lopsided98, 2023-02-07:
Overriding the version won't fix the problem, since it won't change the dependencies. Your best bet is probably to override the package to add `tf2-kdl` to `propagatedBuildInputs` while you wait for them to make a new release.

movefasta, 2023-02-07:
Thanks for advice.
I tried
```
          moveit-core-patched = ros.humble.moveit-core.overrideAttrs (old: {
                propagatedBuildInputs = old.propagatedBuildInputs ++ [ ros.humble.tf2-kdl ];
              });
```
`moveit-core` was built successfully, but when i tried build `moveit-ros-planning` ('moveit-core''s dependent) it starts to build moveit-core again and drops with same error. I thought, that `moveit-ros-planning` is using standard overlays's version and tried to override `moveit-core` from `buildPropagatedInputs` of `moveit-ros-planning`:
```
          moveit-ros-planning = ros.humble.moveit-ros-planning.overrideAttrs (old: {
            propagatedBuildInputs = (pkgs.lib.lists.remove "moveit-core" old.propagatedBuildInputs) ++ [ moveit-core-patched ];
          });
```

This is not helps... so why `moveit-ros-planning` is trying build `moveit-core` again?

movefasta, 2023-02-08:
Thanks. That's works! Final overriding is here
```
humble-overrided = ros.humble.overrideScope (final: prev: {
    moveit-core = prev.moveit-core.overrideAttrs (old: { propagatedBuildInputs = old.propagatedBuildInputs ++ [ prev.tf2-kdl ];});
    moveit-kinematics = prev.moveit-kinematics.overrideAttrs (old: { propagatedBuildInputs = old.propagatedBuildInputs ++ [ final.moveit-ros-planning ];});
  });
```

closed 2023-02-08 by movefasta