lopsided98/nix-ros-overlay #259: Gazebo broken after python2 colcon updates

movefasta, 2023-04-08:
Hello!

Commit https://github.com/lopsided98/nix-ros-overlay/commit/194b4a68bac0cd8910d4685c48698fa598f84758 breaked Gazebo package
https://github.com/lopsided98/nix-ros-overlay/actions/runs/4623604897/jobs/8177599340#step:7:1099
```
  error: builder for '/nix/store/w6x4j67j7hawi4q3lcz6h966mw7864cb-ignition-common3-3.14.1.drv' failed with exit code 2;
         last 10 log lines:
         > /build/ignition-common3_3.14.1-source/graphics/src/Image.cc: In member function 'void ignition::common::Image::SavePNGToBuffer(std::vector<unsigned char>&)':
         > /build/ignition-common3_3.14.1-source/graphics/src/Image.cc:171:8: error: 'memcpy' is not a member of 'std'; did you mean 'wmemcpy'?
         >   171 |   std::memcpy(buffer.data(), memBuffer, sizeInBytes);
         >       |        ^~~~~~
         >       |        wmemcpy
         > [ 70%] Building CXX object graphics/src/CMakeFiles/ignition-common3-graphics.dir/ImageHeightmap.cc.o
         > make[2]: *** [graphics/src/CMakeFiles/ignition-common3-graphics.dir/build.make:160: graphics/src/CMakeFiles/ignition-common3-graphics.dir/Image.cc.o] Error 1
         > make[2]: *** Waiting for unfinished jobs....
         > make[1]: *** [CMakeFiles/Makefile2:419: graphics/src/CMakeFiles/ignition-common3-graphics.dir/all] Error 2
         > make: *** [Makefile:146: all] Error 2
         For full logs, run 'nix log /nix/store/w6x4j67j7hawi4q3lcz6h966mw7864cb-ignition-common3-3.14.1.drv'.
  error: 1 dependencies of derivation '/nix/store/wqbqhgl3wpgyyf6hwl706ar4z2m8hld0-ignition-fuel-tools4-4.5.0.drv' failed to build
  error: 1 dependencies of derivation '/nix/store/a0jg5x8iyaf9iig5i40a7c08gai51fkg-gazebo-11.11.0.drv' failed to build
```

I checked out, one commit before gazebo is builds.
`nix shell github:lopsided98/nix-ros-overlay?ref=77e6334b05dc747e0edccdf9579b0d76947f7956#humble.gazebo` is worked.

lopsided98, 2023-04-08:
That is not the commit that broke it, and this is a common issue affecting a lot of packages recently. This particular case has been fixed in newer versions of ign-common, and I created an upstream PR to backport it to version 3.

movefasta, 2023-04-08:
Link https://github.com/gazebosim/gz-common/pull/501

closed 2023-04-08 by movefasta