lopsided98/nix-ros-overlay #260: Gazebo not building?

apppppppple, 2023-04-20:
Tried to use the gazebo package from the overlay, but it wouldn't build, giving the error `ogre1_9 has been removed, use ogre instead`. Replacing `gazebo` in my package list with `(gazebo.override { ogre1_9 = ogre; })` worked, but then cachix didn't work, so it was building from source. Tried applying the overlay to nixpkgs 22.11 (the rest of my system is on unstable), and using the gazebo from that did build, but still from source, and when ran, gave an error about mismatched glibc versions (I believe it was expecting a newer version than it found). Any advice on what to do here? Is it a problem with the overlay or am I just doing something wrong?

note: For debugging, I also tried applying the overlay to the version of nixpkgs in the flake for this repo, and I tried using the module. Both of them gave the same ogre1_9 error.

ShoosGun, 2023-04-28:
I am facing the same issue, gazebo and rviz are having this issue with ogre1_9, and by changing it to ogre it has to build from source. The strange thing is that I tested the example command `nix develop github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo` and gazebo worked, so I am not sure if it is an issue with how I have my flake setup:

- flake.nix:
```nix
{
  description = "A very basic flake";

  inputs.nixros.url = "github:lopsided98/nix-ros-overlay";
  inputs.flake-utils.url = "github:numtide/flake-utils";

  outputs = { self, nixpkgs, nixros, flake-utils }:
    with nixpkgs.lib;
    with flake-utils.lib;
    eachSystem allSystems (system: let
      pkgs = import nixpkgs {
        inherit system;
        overlays = [ nixros.overlays.default ];
      };
    in {
      legacyPackages = pkgs.rosPackages;

      devShells = {
        teste = import ./teste.nix { inherit pkgs; };
      };
    }) // {
  };

    nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```
- flake.lock:
```lock
{
  "nodes": {
    "flake-utils": {
      "inputs": {
        "systems": "systems"
      },
      "locked": {
        "lastModified": 1681202837,
        "narHash": "sha256-H+Rh19JDwRtpVPAWp64F+rlEtxUWBAQW28eAi3SRSzg=",
        "owner": "numtide",
        "repo": "flake-utils",
        "rev": "cfacdce06f30d2b68473a46042957675eebb3401",
        "type": "github"
      },
      "original": {
        "owner": "numtide",
        "repo": "flake-utils",
        "type": "github"
      }
    },
    "flake-utils_2": {
      "locked": {
        "lastModified": 1676283394,
        "narHash": "sha256-XX2f9c3iySLCw54rJ/CZs+ZK6IQy7GXNY4nSOyu2QG4=",
        "owner": "numtide",
        "repo": "flake-utils",
        "rev": "3db36a8b464d0c4532ba1c7dda728f4576d6d073",
        "type": "github"
      },
      "original": {
        "owner": "numtide",
        "repo": "flake-utils",
        "type": "github"
      }
    },
    "nixpkgs": {
      "locked": {
        "lastModified": 1682566018,
        "narHash": "sha256-HPzPRFiy2o/7k7mtnwfM1E6NVZHiFbPdmYCMoIpkHO4=",
        "owner": "NixOS",
        "repo": "nixpkgs",
        "rev": "8e3b64db39f2aaa14b35ee5376bd6a2e707cadc2",
        "type": "github"
      },
      "original": {
        "id": "nixpkgs",
        "type": "indirect"
      }
    },
    "nixpkgs_2": {
      "locked": {
        "lastModified": 1678244258,
        "narHash": "sha256-iecrLqFb1QDgBKmC0h065ZBV6CZsigcz+1aHCv6rvfg=",
        "owner": "lopsided98",
        "repo": "nixpkgs",
        "rev": "61852b7faa8b47aad422adca0fea90fe007e9ead",
        "type": "github"
      },
      "original": {
        "owner": "lopsided98",
        "ref": "nix-ros",
        "repo": "nixpkgs",
        "type": "github"
      }
    },
    "nixros": {
      "inputs": {
        "flake-utils": "flake-utils_2",
        "nixpkgs": "nixpkgs_2"
      },
      "locked": {
        "lastModified": 1680981731,
        "narHash": "sha256-czIXVpxwnsqAmHMj+XJdcqiwvmGoq9nPuWCkeZ0sLik=",
        "owner": "lopsided98",
        "repo": "nix-ros-overlay",
        "rev": "650b5972c994c6d13fdac70869c19c52c056d8de",
        "type": "github"
      },
      "original": {
        "owner": "lopsided98",
        "repo": "nix-ros-overlay",
        "type": "github"
      }
    },
    "root": {
      "inputs": {
        "flake-utils": "flake-utils",
        "nixpkgs": "nixpkgs",
        "nixros": "nixros"
      }
    },
    "systems": {
      "locked": {
        "lastModified": 1681028828,
        "narHash": "sha256-Vy1rq5AaRuLzOxct8nz4T6wlgyUR7zLU309k9mBC768=",
        "owner": "nix-systems",
        "repo": "default",
        "rev": "da67096a3b9bf56a91d16901293e51ba5b49a27e",
        "type": "github"
      },
      "original": {
        "owner": "nix-systems",
        "repo": "default",
        "type": "github"
      }
    }
  },
  "root": "root",
  "version": 7
}
```
- teste.nix:
```nix
# Run:
# roslaunch turtlebot3_gazebo turtlebot3_world.launch
# roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

{ pkgs ? import ../. {} }:
with pkgs;
with rosPackages.noetic;
with pythonPackages;

mkShell {

  buildInputs = [
    glibcLocales
    (buildEnv { paths = [
      rosbash
      rostopic
      roslaunch
      std-msgs
      ublox
      navigation
      joy
      teleop-twist-keyboard
      teleop-twist-joy
      #teb-local-planner
      twist-mux
      rqt-tf-tree
      #(rviz.override {ogre1_9 = ogre; })
      diffbot-control
      #turtlebot3-description
      #turtlebot3-teleop
      #turtlebot3-gazebo
      #(gazebo.override { ogre1_9 = ogre; })
      #(gazebo-plugins.gazebo.override { ogre1_9 = ogre; })
      xacro
    ]; })
  ];

  ROS_HOSTNAME = "localhost";
  ROS_MASTER_URI = "http://localhost:11311";
  TURTLEBOT3_MODEL = "burger";
}
```

lopsided98, 2023-04-28:
You are using a newer version of nixpkgs than I test with (https://github.com/lopsided98/nixpkgs/tree/nix-ros). I will try to fix this when I update nixpkgs, but last time I checked Gazebo fails at runtime with newer Ogre versions, see: #161

ShoosGun, 2023-05-01:
I changed the nixpkgs url to the one you mentioned, and it gave the same result of ogre1_9 being removed. But I saw that your lock file was last updated two months ago, so I changed the rev to the one in the lock file (github:lopside98/nixpkgs/61852b7faa8b47aad422adca0fea90fe007e9ead), and is now compiling from source, but I think it is using ogre1_9, so perhaps this can be used as a temporary fix in my case.

lopsided98, 2023-05-01:
Sorry, yeah, I forgot I updated it and never checked what new packages failed to build.

lopsided98, 2023-05-01:
I assume that it was removed because nothing in nixpkgs uses it anymore.

ShoosGun, 2023-05-01:
Found now another issue while building: collision with the ros-noetic-python-qt-binding.
```bash
warning: Using saved setting for 'extra-substituters = https://ros.cachix.org' from ~/.local/share/nix/trusted-settings.json.
warning: Using saved setting for 'extra-trusted-public-keys = ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' from ~/.local/share/nix/trusted-settings.json.
warning: ignoring untrusted substituter 'https://ros.cachix.org'
error: builder for '/nix/store/fsfwgcw3r9xvwk7yjgc32swfysf3xigv-ros-env.drv' failed with exit code 255;
       last 2 log lines:
       > building
       > error: collision between `/nix/store/a9yg3jax0mj9y07c6785wfhyz91v9bgr-ros-noetic-python-qt-binding-0.4.4-r1/lib/python3.10/site-packages/python_qt_binding/__pycache__/binding_helper.cpython-310.pyc' and `/nix/store/ljrswlz02fc1a7qvhjw8yhcxkcp4byka-ros-noetic-python-qt-binding-0.4.4-r1/lib/python3.10/site-packages/python_qt_binding/__pycache__/binding_helper.cpython-310.pyc'
       For full logs, run 'nix log /nix/store/fsfwgcw3r9xvwk7yjgc32swfysf3xigv-ros-env.drv'.
error: 1 dependencies of derivation '/nix/store/6cpx9cpmnzm098xqalgl02xsgc2x3qcr-nix-shell-env.drv' failed to build
```

```bash
nix log /nix/store/fsfwgcw3r9xvwk7yjgc32swfysf3xigv-ros-env.drv
@nix { "action": "setPhase", "phase": "buildPhase" }
building
error: collision between `/nix/store/a9yg3jax0mj9y07c6785wfhyz91v9bgr-ros-noetic-python-qt-binding-0.4.4-r1/lib/python3.10/site-packages/python_qt_binding/>
```
The changes I did to the flake where (apart from adding rviz as a package):
```nix
inputs = {
    nixpkgs.url = "github:lopsided98/nixpkgs/61852b7faa8b47aad422adca0fea90fe007e9ead";
    flake-utils.url = "github:numtide/flake-utils/3db36a8b464d0c4532ba1c7dda728f4576d6d073";
    nixros.url = "github:lopsided98/nix-ros-overlay";
};
```
I really don't know how the collision is happening.

lopsided98, 2023-05-01:
I switched Gazebo to use Ogre 1.10, which allows it to build and run, but has the bug where adding multiple light sources causes it to freeze (discussed in #161). Gazebo fails to build with Ogre 13.

I believe that collision occurs because rviz requires a patched version of Qt5, so you end up with two versions of `python-qt-binding` in your environment. If you put rviz in a separate `buildEnv` from the rest (or just outside the `buildEnv`), it should work.

ShoosGun, 2023-05-02:
Got it to get rviz from the cache, but now the issue is a mismatch with the qt lib versions 
```bash
rviz
[ INFO] [1682992662.770088495]: rviz version 1.14.19
[ INFO] [1682992662.770105880]: compiled against Qt version 5.15.8
[ INFO] [1682992662.770111971]: compiled against OGRE version 1.9.0 (Ghadamon)
Cannot mix incompatible Qt library (5.15.7) with this library (5.15.8)
Aborted (core dumped)
```
Is there a way to specify the qt lib version of rviz, like how it was done for ogre? I could only find mentions of qt5 on the default.nix from rviz, but I am not sure where is the file of qt5.

lopsided98, 2023-05-02:
I think you must be mixing nixpkgs versions somehow.

ShoosGun, 2023-05-02:
I think I found the source of the issue. I am running nixos with kde, and I think it is using qt-5.15.7. I ran strace on rviz and I saw that both qt-5.15.8 and qt-5.15.7 are being mentioned, with qt-5.15.7 appearing when it tries to grab configs from kde. Perhaps there is a way to isolate the nix develop for it to not try to search kde configs, which might be why it is loading qt-5.15.7. I also find these issues, which are probably related to this somehow: https://github.com/NixOS/nixpkgs/issues/30551 and https://github.com/supercollider/supercollider/issues/5777

closed 2023-06-19 by lopsided98

AchmadFathoni, 2024-07-30:
> I think I found the source of the issue. I am running nixos with kde, and I think it is using qt-5.15.7. I ran strace on rviz and I saw that both qt-5.15.8 and qt-5.15.7 are being mentioned, with qt-5.15.7 appearing when it tries to grab configs from kde. Perhaps there is a way to isolate the nix develop for it to not try to search kde configs, which might be why it is loading qt-5.15.7. I also find these issues, which are probably related to this somehow: [NixOS/nixpkgs#30551](https://github.com/NixOS/nixpkgs/issues/30551) and [supercollider/supercollider#5777](https://github.com/supercollider/supercollider/issues/5777)

I have similar problem as #437, my current solution is  `unset QT_LIBRARY_PATH`.