lopsided98/nix-ros-overlay #262: Adding local packages to rosPackages

ShoosGun, 2023-05-08:
I am trying to add a local ros package to be used as the ones in rosPackages, in the overlay. I was able to get it to acknowledge that it exists, but it couldn't build saying that there was no argument called `buildRosPackage`. Here is the flake.nix and the default.nix of the package:

flake.nix:
```nix
{
  description = "ROS Development Enviroment";

  inputs = {
    nixpkgs.url = "github:lopsided98/nixpkgs/61852b7faa8b47aad422adca0fea90fe007e9ead";
    flake-utils.url = "github:numtide/flake-utils";
    nixros.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, nixros, flake-utils }:
    with nixpkgs.lib;
    with flake-utils.lib;
    eachSystem allSystems (system: let
      pkgs = import nixpkgs {
        inherit system;
        overlays = [
          nixros.overlays.default
          (final: prev: {
            kydas-driver = prev.callPackage ./pkgs/kydas-driver {}; #the pkg I want to add
          })
        ];
      };
    in {
      legacyPackages = pkgs.rosPackages ;

      devShells = {
      #dev shells
      };
    }) // {
  };

    nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

pkgs/kydas_driver/default.nix:
```nix
{ lib, buildRosPackage, fetchFromGitHub, catkin }:
buildRosPackage {
  pname = "kydas-driver";
  version = "1.0.0";

  src = fetchFromGitHub {
    owner = "ShoosGun";
    repo = "kydas_driver";
    rev = "25be2f6a6a30a0947a53a68387915744c8115bfd";
  };

  buildType = "catkin";
  buildInputs = [ catkin ];
  #propagatedBuildInputs = [ industrial-robot-client ];
  nativeBuildInputs = [ catkin ];

  meta = {
    #description = '''';
    license = with lib.licenses; [ bsdOriginal ];
  };
}
```

I am not familiar with extending/overriding overlays, so that is probably why I couldn't get it working.

wentasah, 2023-05-09:
I use the following approach:
```nix
{
  description = "ROS Development Enviroment";

  inputs = {
    nixpkgs.url = "github:lopsided98/nixpkgs/61852b7faa8b47aad422adca0fea90fe007e9ead";
    flake-utils.url = "github:numtide/flake-utils";
    nixros.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, nixros, flake-utils }:
    let my-ros-distro-overlay = (final: prev: {
          kydas-driver = prev.callPackage ./pkgs/kydas-driver {}; #the pkg I want to add
        });
        my-ros-overlay = (self: super: {
          # Apply our overlay to multiple ROS distributions
          rosPackages = super.rosPackages // {
            noetic = super.rosPackages.noetic.overrideScope my-ros-distro-overlay;
            foxy = super.rosPackages.foxy.overrideScope my-ros-distro-overlay;
            humble = super.rosPackages.humble.overrideScope my-ros-distro-overlay;
      };
    });

    in
    with nixpkgs.lib;
    with flake-utils.lib;
    eachSystem allSystems (system: let
      pkgs = import nixpkgs {
        inherit system;
        overlays = [
          nixros.overlays.default
          my-ros-overlay
        ];
      };
    in {
      legacyPackages = pkgs.rosPackages ;

      devShells = {
      #dev shells
      };
    }) // {
  };

    nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```
You create your overlay once and that apply it to multiple ROS distributions as you need.

With that, when I run `nix build .#noetic.kydas-driver`, it starts building but fails due to missing dependencies. Hopefully, you can fix those yourself.

If there is some better way of doing this, I'd be happy to learn it.

hacker1024, 2023-06-30:
If you want to add your packages to every ROS version (like how Nixpkgs handles Python and NodeJS), you can use this function in an overlay:

```nix
rosOverlay: rosPackages:
  rosPackages // builtins.mapAttrs
    (rosDistro: rosPkgs:
      if rosPkgs ? overrideScope
      then rosPkgs.overrideScope rosOverlay
      else rosPkgs)
    rosPackages;
```

```
self: super: {
  rosPackages = appendDistroOverlay (rosSelf: rosSuper: { /* ... */ });
}
```

movefasta, 2024-02-20:
> I use the following approach:

Does it work on ROS2 packages?

If i try to build ROS2 package with content like that (got it from random simple interfaces package from nix-ros-overlay):

```nix
{ lib, buildRosPackage, fetchurl, action-msgs, ament-cmake, rosidl-default-generators, rosidl-default-runtime }:
buildRosPackage {
  pname = "ros-test-example-interfaces";
  version = "0.9.1-r1";

  src = fetchurl {
    url = "https://github.com/ros2-gbp/example_interfaces-release/archive/release/foxy/example_interfaces/0.9.1-1.tar.gz";
    name = "0.9.1-1.tar.gz";
    sha256 = "74a5cd21a531a90d5117d482943d18ad44f0b5b3679ec28a3331f32f660aa77d";
  };

  buildType = "ament_cmake";
  buildInputs = [ ament-cmake rosidl-default-generators ];
  propagatedBuildInputs = [ action-msgs rosidl-default-runtime ];
  nativeBuildInputs = [ ament-cmake rosidl-default-generators ];

  meta = {
    description = ''Contains message and service definitions used by the examples.'';
    license = with lib.licenses; [ asl20 ];
  };
}
```

I stuck at this error:
```
       error: attribute 'pythonOnBuildForHost' missing

       at /nix/store/g1g3pmyvkjsxmbb21hng61081ar8snj3-source/distros/ament-cmake-core-setup-hook/default.nix:5:37:

            4|   name = "ament-cmake-core-setup-hook";
            5|   substitutions.python_executable = python.pythonOnBuildForHost.interpreter;
             |                                     ^
            6| } ./setup-hook.sh
```

Doesn't understand why this attribute is missing

wentasah, 2024-02-20:
I use it with ROS 2 distros.

It looks like you're crosscompiling the package for another architecture or OS. Is it the case?

movefasta, 2024-02-20:
> I use it with ROS 2 distros.
> 
> It looks like you're crosscompiling the package for another architecture or OS. Is it the case?

Nope... Just now i'm init new repo with flake.nix (from your comment above):
```nix
{
  description = "ROS Development Enviroment";

  inputs = {
    nixpkgs.url = "github:lopsided98/nixpkgs/61852b7faa8b47aad422adca0fea90fe007e9ead";
    flake-utils.url = "github:numtide/flake-utils";
    nixros.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, nixros, flake-utils }:
    let my-ros-distro-overlay = (final: prev: {
          pkg = prev.callPackage ./pkg.nix {}; #the pkg I want to add
        });
        my-ros-overlay = (self: super: {
          # Apply our overlay to multiple ROS distributions
          rosPackages = super.rosPackages // {
            noetic = super.rosPackages.noetic.overrideScope my-ros-distro-overlay;
            foxy = super.rosPackages.foxy.overrideScope my-ros-distro-overlay;
            humble = super.rosPackages.humble.overrideScope my-ros-distro-overlay;
      };
    });

    in
    with nixpkgs.lib;
    with flake-utils.lib;
    eachSystem allSystems (system: let
      pkgs = import nixpkgs {
        inherit system;
        overlays = [
          nixros.overlays.default
          my-ros-overlay
        ];
      };
    in {
      legacyPackages = pkgs.rosPackages ;

      devShells = {
      #dev shells
      };
    }) // {
  };

    nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

and pkg.nix from `ros-humble-action-msgs`:

```nix
{ lib, buildRosPackage, fetchurl, ament-cmake, ament-lint-auto, ament-lint-common, builtin-interfaces, rosidl-default-generators, rosidl-default-runtime, unique-identifier-msgs }:
buildRosPackage {
  pname = "ros-humble-action-msgs";
  version = "1.2.1-r1";

  src = fetchurl {
    url = "https://github.com/ros2-gbp/rcl_interfaces-release/archive/release/humble/action_msgs/1.2.1-1.tar.gz";
    name = "1.2.1-1.tar.gz";
    sha256 = "dbea032efe7fdd74415f1f6fefd2eb1c1f891cd27025ea54684992833ff9d4c6";
  };

  buildType = "ament_cmake";
  buildInputs = [ ament-cmake rosidl-default-generators ];
  checkInputs = [ ament-lint-auto ament-lint-common ];
  propagatedBuildInputs = [ builtin-interfaces rosidl-default-runtime unique-identifier-msgs ];
  nativeBuildInputs = [ ament-cmake rosidl-default-generators ];

  meta = {
    description = ''Messages and service definitions common among all ROS actions.'';
    license = with lib.licenses; [ asl20 ];
  };
}
```

Result the same...

```
nix build .\#legacyPackages.x86_64-linux.humble.pkg                                                                                               
warning: Git tree '/home/movefasta/Documents/nix-overlay-test' is dirty
error:
       … while calling the 'derivationStrict' builtin

         at /derivation-internal.nix:9:12:

            8|
            9|   strict = derivationStrict drvAttrs;
             |            ^
           10|

       … while evaluating derivation 'ros-humble-control-msgs-4.4.0-r1'
         whose name attribute is located at /nix/store/8ac1azrh8xv8sh7f2hbx4kqzv8bf270q-source/pkgs/stdenv/generic/make-derivation.nix:294:7

       … while evaluating attribute 'buildInputs' of derivation 'ros-humble-control-msgs-4.4.0-r1'

         at /nix/store/8ac1azrh8xv8sh7f2hbx4kqzv8bf270q-source/pkgs/stdenv/generic/make-derivation.nix:337:7:

          336|       depsHostHost                = lib.elemAt (lib.elemAt dependencies 1) 0;
          337|       buildInputs                 = lib.elemAt (lib.elemAt dependencies 1) 1;
             |       ^
          338|       depsTargetTarget            = lib.elemAt (lib.elemAt dependencies 2) 0;

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: attribute 'pythonOnBuildForHost' missing

       at /nix/store/g1g3pmyvkjsxmbb21hng61081ar8snj3-source/distros/ament-cmake-core-setup-hook/default.nix:5:37:

            4|   name = "ament-cmake-core-setup-hook";
            5|   substitutions.python_executable = python.pythonOnBuildForHost.interpreter;
             |                                     ^
            6| } ./setup-hook.sh
```

My system:
NixOS 24.05.20240207.d1238f3 (Uakari) x86_64
Kernel: 6.1.77

movefasta, 2024-02-21:
> I use it with ROS 2 distros.
> 
> It looks like you're crosscompiling the package for another architecture or OS. Is it the case?

My test overlay repo
https://github.com/movefasta/nix-test-overlay

movefasta, 2024-02-23:
@wentasah
Can you give me some working reference example with local ros2 package and flake?

wentasah, 2024-02-24:
You example is almost correct. You just need to use compatible nixpkgs version. Changing your flake as follows:
```diff
--- a/flake.nix
+++ b/flake.nix
@@ -2,7 +2,7 @@
   description = "ROS Development Enviroment";
 
   inputs = {
-    nixpkgs.url = "github:lopsided98/nixpkgs/61852b7faa8b47aad422adca0fea90fe007e9ead";
+    nixpkgs.follows = "nixros/nixpkgs";
     flake-utils.url = "github:numtide/flake-utils";
     nixros.url = "github:lopsided98/nix-ros-overlay";
   };
```
and running `nix build .#legacyPackages.x86_64-linux.humble.pkg` gives:

    result -> /nix/store/9bil3v66fcawdy50wwndmfmdc1bxd2lw-ros-humble-control-msgs-4.4.0-r1

Your original flake used almost one year old nixpkgs:

    github:lopsided98/nixpkgs/61852b7faa8b47aad422adca0fea90fe007e9ead (2023-03-08 02:57:38)

Current nix-ros-overlay uses:

    github:lopsided98/nixpkgs/f8e2ebd66d097614d51a56a755450d4ae1632df1 (2024-02-07 01:22:34)

movefasta, 2024-02-29:
@wentasah thank you very much, it works!

closed 2024-04-07 by lopsided98