lopsided98/nix-ros-overlay #264: MoveIt2 move_group, RViz2, etc. can't find class plugins

okvik, 2023-05-18:
After some patching and cherry picking issues I almost have a workable ROS 2 Humble environment with MoveIt2 and RViz2.

However, the `move_group` package that expects to load its capabilities from plugins have trouble finding them, even though the plugin objects themselves and the resource index stuff seem to be in place. Strangely enough the MoveIt2 planning pipelines, which are also plugins, get loaded fine... 

I'm not exactly sure on how to continue debugging this. Did anyone encounter this with other packages before? I can't seem to find many references to relevant problems with plugins in past issues.

Later today I'll (try to) provide a way to reproduce this easily, but for now here's a log of running `ros2 launch moveit_resources_fanuc_moveit_config demo.launch.py` (Fanuc instead of Panda here because Panda config package is broken in the Humble release of `MoveIt`... 

```
[INFO] [launch]: All log files can be found below /home/kvik/.ros/log/2023-05-18-16-22-50-727021-vpcd-dell-2379303
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [rviz2-1]: process started with pid [2379304]
[INFO] [static_transform_publisher-2]: process started with pid [2379306]
[INFO] [robot_state_publisher-3]: process started with pid [2379308]
[INFO] [move_group-4]: process started with pid [2379310]
[INFO] [ros2_control_node-5]: process started with pid [2379312]
[INFO] [spawner-6]: process started with pid [2379314]
[INFO] [spawner-7]: process started with pid [2379316]
[static_transform_publisher-2] [WARN] [1684419771.395976573] []: Old-style arguments are deprecated; see --help for new-style arguments
[ros2_control_node-5] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class mock_components::GenericSystem. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[ros2_control_node-5]          at line 253 in /nix/store/4is2qsw3j6an766xhbzik16zzhrf6sg9-ros-humble-class-loader-2.2.0-r3/include/class_loader/class_loader/class_loader_core.hpp
[static_transform_publisher-2] [INFO] [1684419771.435410062] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'base_link'
[robot_state_publisher-3] [INFO] [1684419771.474847917] [robot_state_publisher]: got segment base
[robot_state_publisher-3] [INFO] [1684419771.474940903] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1684419771.474970662] [robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1684419771.474988278] [robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1684419771.475001192] [robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1684419771.475013926] [robot_state_publisher]: got segment link_4
[robot_state_publisher-3] [INFO] [1684419771.475028027] [robot_state_publisher]: got segment link_5
[robot_state_publisher-3] [INFO] [1684419771.475053199] [robot_state_publisher]: got segment link_6
[robot_state_publisher-3] [INFO] [1684419771.475066758] [robot_state_publisher]: got segment tool0
[ros2_control_node-5] [INFO] [1684419771.568447574] [resource_manager]: Loading hardware 'FanucFakeSystem' 
[ros2_control_node-5] [INFO] [1684419771.578953792] [resource_manager]: Initialize hardware 'FanucFakeSystem' 
[ros2_control_node-5] [WARN] [1684419771.579015382] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-5] <state_interface name="velocity"> 
[ros2_control_node-5]   <param name="initial_value">0.0</param> 
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1684419771.579046827] [resource_manager]: Successful initialization of hardware 'FanucFakeSystem'
[ros2_control_node-5] [INFO] [1684419771.579105848] [resource_manager]: 'configure' hardware 'FanucFakeSystem' 
[ros2_control_node-5] [INFO] [1684419771.579117408] [resource_manager]: Successful 'configure' of hardware 'FanucFakeSystem'
[ros2_control_node-5] [INFO] [1684419771.579133298] [resource_manager]: 'activate' hardware 'FanucFakeSystem' 
[ros2_control_node-5] [INFO] [1684419771.579142027] [resource_manager]: Successful 'activate' of hardware 'FanucFakeSystem'
[ros2_control_node-5] [INFO] [1684419771.635727695] [controller_manager]: update rate is 100 Hz
[ros2_control_node-5] [INFO] [1684419771.636125580] [controller_manager]: RT kernel is recommended for better performance
[move_group-4] [INFO] [1684419771.839521105] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00520669 seconds
[move_group-4] [INFO] [1684419771.839565106] [moveit_robot_model.robot_model]: Loading robot model 'fanuc'...
[move_group-4] [INFO] [1684419772.020825563] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-4] [INFO] [1684419772.020999817] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-4] [INFO] [1684419772.022911549] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-4] [INFO] [1684419772.023932304] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-4] [INFO] [1684419772.023959775] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-4] [INFO] [1684419772.024778985] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-4] [INFO] [1684419772.024806193] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-4] [INFO] [1684419772.025873725] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-4] [INFO] [1684419772.026822991] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-4] [WARN] [1684419772.029829951] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-4] [ERROR] [1684419772.029874774] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[move_group-4] [INFO] [1684419772.034603960] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'ompl'
[move_group-4] [INFO] [1684419772.318970527] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-4] [INFO] [1684419772.426790493] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.path_tolerance' was not set. Using default value: 0.100000
[move_group-4] [INFO] [1684419772.426982403] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.resample_dt' was not set. Using default value: 0.100000
[move_group-4] [INFO] [1684419772.427007566] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.min_angle_change' was not set. Using default value: 0.001000
[move_group-4] [INFO] [1684419772.427090255] [moveit_ros.fix_workspace_bounds]: Param 'ompl.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-4] [INFO] [1684419772.427130172] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_bounds_error' was set to 0.100000
[move_group-4] [INFO] [1684419772.427145480] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-4] [INFO] [1684419772.427176694] [moveit_ros.fix_start_state_collision]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[... 78 lines cut ...]
[... cut ...]
[rviz2-1] [INFO] [1684419783.089468291] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-1] [INFO] [1684419783.089534574] [moveit_ros_visualization.motion_planning_frame]: group manipulator
[rviz2-1] [INFO] [1684419783.089545966] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[rviz2-1] [INFO] [1684419843.116869728] [move_group_interface]: Ready to take commands for planning group manipulator.

```

wentasah, 2023-05-18:
Hi, I've no experience with MoveIt. I'd need to wait for your `.nix` files so that I can try to reproduce the issue you're reporting.

okvik, 2023-05-18:
@wentasah Here's a flake that can reproduce the `move_group` capability plugins issue. `nix develop` it and run

```
ros2 launch moveit_resources_fanuc_moveit_config demo.launch.py
```
You should get the same log as above. Take note of the lines like:
```
[move_group.move_group]: Exception while loading move_group capability 'move_group/ApplyPlanningSceneService': MultiLibraryClassLoader: Could not create object of class type move_group::ApplyPlanningSceneService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
```

```
{
  description = "";

  inputs = {
    nixpkgs.url = "github:nixos/nixpkgs/nixos-unstable";
    flake-utils.url = "github:numtide/flake-utils";
    nix-ros-overlay.url = "github:okvik/nix-ros-overlay/reproduce-moveit-plugin-issue";
  };

  outputs = { self, nixpkgs, flake-utils, nix-ros-overlay }: flake-utils.lib.eachDefaultSystem (system:
    let
      pkgs = import nixpkgs {
        inherit system;
        config = {};
        overlays = [
          nix-ros-overlay.overlays.default
        ];
      };
      ros = pkgs.rosPackages.humble;
    in {
      devShells.default = pkgs.mkShell {
        name = "ROS 2 development";
        LC_NUMERIC = "en_US.UTF-8";
        buildInputs = [
          (ros.buildEnv {
            name = "ROS 2 Environment";
            paths = [              
              ros.ros-environment
              ros.ros-base
              ros.ros2-control
              ros.rviz2
              ros.moveit
              ros.moveit-resources
              ros.moveit-resources-panda-description
              ros.moveit-resources-panda-moveit-config
            ];
          })
        ];
      };
    }
  );
}
```

wentasah, 2023-05-19:
Now I can reproduce your problem with the missing capabilities. 

You mentioned that "plugin objects are in place". As I don't have any moveit experience, I don't really know where should these objects be located. I tried searching all dependencies for files with names similar to what's printed in the error messages, but it didn't find anything. Can you tell me which objects it is supposed to load?

okvik, 2023-05-19:
The plugins are described in the plugin description file (you may have a different hash but look for `move-group`.

```
/nix/store/5092rwrx983f3s8vg0k4ifik9zasdxjd-ros-humble-moveit-ros-move-group-2.5.4-r1/share/moveit_ros_move_group/default_capabilities_plugin_description.xml
```

This file lists the plugin classes (in this case MoveIt Move Group "capabilities") and points to the shared library object that contains them. This `.so` file itself is located in the same package at:

```
/nix/store/5092rwrx983f3s8vg0k4ifik9zasdxjd-ros-humble-moveit-ros-move-group-2.5.4-r1/lib/libmoveit_move_group_default_capabilities.so
```
It is my understanding that the `pluginlib` (a library responsible for plugin loading throughout ROS2) searches through plugin description files it knows about when it gets asked to load a plugin class, hopefully finding a match and loading it's object from the shared object file. The resource index (located at `./share/ament_index/resource_index` is there, I think, only to speed up the process of searching for viable plugins, so the search doesn't have to go through all the plugin `.xml` descriptions.

All of these things seem to be in place for the said `move-group` package but loading the plugins still doesn't work for this specific package while it does for others.

okvik, 2023-05-19:
I've discovered a program in the move group package that lists the available capabilities. This one seems to find the plugins just fine?!

`ros2 run moveit_ros_move_group list_move_group_capabilities`

okvik, 2023-05-19:
I found the same issue reported upstream, so this might not be a problem with Nix packaging here: https://github.com/ros-planning/moveit2/issues/1907

Thieso, 2024-04-18:
I encountered the same issue but have also not found a solution

Thieso, 2024-04-22:
Ok so I think this is a path issue. On `humble` and by using `buildEnv` I do not encounter this issue. On `iron` I cannot use `buildEnv` due to #391 which makes `AMENT_PREFIX_PATH` be very long. So I assume fixing #391 would also fix this issue potentially.

nilp-dromeda, 2024-05-18:
> Ok so I think this is a path issue. On `humble` and by using `buildEnv` I do not encounter this issue. On `iron` I cannot use `buildEnv` due to #391 which makes `AMENT_PREFIX_PATH` be very long. So I assume fixing #391 would also fix this issue potentially.

Were you able to fix this? I am also getting the same error.

Thieso, 2024-05-21:
Sadly, I was not able to fix this. As I am not a MoveIt expert, it is unclear to me, where exactly this originates and [this issue](https://github.com/moveit/moveit2/issues/1907) also does not mention a solution except reinstallation. I did not have more time to look into this.

Takishima, 2024-05-23:
So looking into this a little more, it seems that the `pluginlib` or the `class_loader` packages are unable to read the symbols from the MoveIt library.

Essentially, going back up the callstack from

```
[move_group-4] [INFO 1716497236.304083998] [pluginlib.ClassLoader]: Exception raised by low-level multi-library class loader when attempting to create instance of class move_group/MoveGroupStateValidationService. (createUniqueInstance() at /tmp/install/include/pluginlib/pluginlib/class_loader_imp.hpp:178)
```

The `ClassLoader * getClassLoaderForClass(const std::string & class_name)` function located at `multi_library_class_loader.hpp` from the `class_loader` returns `nullptr`.

This is in turns due to the  `template<class Base> std::vector<std::string> getAvailableClasses() const` from `class_loader.hpp` from the same package returning an empty `std::vector<std::string>`.

The debug output (which I hacked) of some of the `class_loader_core` functions also seems to indicate this:
```
move_group-4] Warning: class_loader.impl: Attempting to load library /tmp/install/lib/libmoveit_move_group_default_capabilities.so on behalf of ClassLoader handle 0x1293650...
[move_group-4]          at line 433 in /home/damien/code/ros2/src/class_loader/src/class_loader_core.cpp
[move_group-4] Loading 'move_group/ApplyPlanningSceneService'...
[move_group-4] Warning: class_loader.impl: Successfully loaded library /tmp/install/lib/libmoveit_move_group_default_capabilities.so into memory (handle = 0x12c33f0).
[move_group-4]          at line 474 in /home/damien/code/ros2/src/class_loader/src/class_loader_core.cpp
[move_group-4] Warning: class_loader.impl: Though the library /tmp/install/lib/libmoveit_move_group_default_capabilities.so was just loaded, it seems no factory metaobjects were registered. Checking factory graveyard for previously loaded metaobjects...
[move_group-4]          at line 482 in /home/damien/code/ros2/src/class_loader/src/class_loader_core.cpp
```

In a normal run  (outside of Nix), we get something like this:
```
[move_group-4] [INFO 1716496724.767593825] [pluginlib.ClassLoader]: Loading library: /home/damien/code/ros2/install/lib/libmoveit_move_group_default_capabilities.so (loadLibraryForClass() at /home/damien/code/ros2/install/include/pluginlib/pluginlib/class_loader_imp.hpp:654)
[move_group-4] Warning: class_loader.impl: Attempting to load library /home/damien/code/ros2/install/lib/libmoveit_move_group_default_capabilities.so on behalf of ClassLoader handle 0x5e99297e6850...
[move_group-4]          at line 433 in /home/damien/code/ros2/src/class_loader/src/class_loader_core.cpp
... [omitted]
[move_group-4] Warning: class_loader.impl: Registering plugin factory for class = move_group::ApplyPlanningSceneService, ClassLoader* = 0x5e99297e6850 and library name /home/damien/code/ros2/install/lib/libmoveit_move_group_default_capabilities.so.
[move_group-4]          at line 215 in /home/damien/code/ros2/install/include/class_loader/class_loader/class_loader_core.hpp
[move_group-4] Warning: factoryMapMap.size() = 7 (N10move_group19MoveGroupCapabilityE)
[move_group-4]          at line 68 in /home/damien/code/ros2/src/class_loader/src/class_loader_core.cpp
[move_group-4] Warning: class_loader.impl: Registration of move_group::ApplyPlanningSceneService complete (Metaobject Address = 0x5e99297ea780)
... [omitted]
[move_group-4] Warning: class_loader.impl: Successfully loaded library /home/damien/code/ros2/install/lib/libmoveit_move_group_default_capabilities.so into memory (handle = 0x5e99297e76a0).
[move_group-4]          at line 474 in /home/damien/code/ros2/src/class_loader/src/class_loader_core.cpp
[move_group-4] Warning: class_loader.impl: Library /home/damien/code/ros2/install/lib/libmoveit_move_group_default_capabilities.so generated new factory metaobjects on load. Destroying graveyarded objects from previous loads...
[move_group-4]          at line 491 in /home/damien/code/ros2/src/class_loader/src/class_loader_core.cpp
```

Takishima, 2024-05-24:
Rather, the init functions from the plugins defined within `libmoveit_move_group_default_capabilities.so` are not called when this gets `dlopen`-ed by the class loader.

However, when loading the library manually:
```
#include <dlfcn.h>

int main()
{
  const char * library_path = "/tmp/install/lib/libmoveit_move_group_default_capabilities.so";
  void * handle = dlopen(library_path, RTLD_LAZY);
  if (handle == nullptr) {
    return 1;
  }
  dlclose(handle);

  return 0;
}
```
Those get called.

Takishima, 2024-05-24:
I think I found the culprit. Still need to run some more tests to be 100% sure, but I think I figured it out.

By default Nix builds derivation with `bindnow` in `NIX_HARDENING_ENABLE`. This in turns silently adds `-Wl,-z,now` to any linker command.

This then wreaks havoc with `pluginlib` which loads libraries with `dlopen()` and relies on the fact that libraries are loaded lazily in order to attribute the symbols it loads to a particular shared library.

Takishima, 2024-05-24:
I stand corrected, it is a bit more than just that.

```
[INFO] [1716561773.091904226] [pluginlib.ClassLoader]: [search path for 'pilz_industrial_motion_planner']: '/nix/store/XXX-env/bin/pilz_industrial_motion_planner/liblibpilz_industrial_motion_plannerd.so'
[INFO] [1716561773.091906893] [pluginlib.ClassLoader]: Iterating through all possible paths where pilz_industrial_motion_planner could be located...
[INFO] [1716561773.091909363] [pluginlib.ClassLoader]: Checking path /nix/store/XXX-env/lib/libpilz_industrial_motion_planner.so 
[INFO] [1716561773.091916802] [pluginlib.ClassLoader]: Library pilz_industrial_motion_planner found at explicit path /nix/store/XXX-env/lib/libpilz_industrial_motion_planner.so.
[INFO] [1716561773.091920312] [pluginlib.ClassLoader]: Loading library: /nix/store/XXX-env/lib/libpilz_industrial_motion_planner.so
```

The issue is that `libpilz_industrial_motion_planner.so` is actually dependent on `libmoveit_move_group_interface.so.2.5.5 `, but only within Nix.

Outside of Nix:
```
> ldd /opt/ros/humble/lib/libpilz_industrial_motion_planner.so | grep move_group
```

Within of Nix:
```
> ldd /nix/store/yvsf95mf3w0m9x1c0dxr4bj0hsr874v6-paragon-core-buildDeps-env/lib/libpilz_industrial_motion_planner.so | grep move_group
	libmoveit_move_group_interface.so.2.5.5 => /opt/ros/humble/lib/libmoveit_move_group_interface.so.2.5.5 (0x0000746c6098d000)
	libmoveit_move_group_default_capabilities.so.2.5.5 => .../install/lib/libmoveit_move_group_default_capabilities.so.2.5.5 (0x0000746c60200000)
	libmoveit_move_group_capabilities_base.so.2.5.5 => .../install/lib/libmoveit_move_group_capabilities_base.so.2.5.5 (0x0000746c6089b000)
```

Thieso, 2024-05-25:
Cool, thanks for looking into this :) So turning off `NIX_HARDENING_ENABLE` fixes the issue? 
I am a bit confused that it works in humble for me but not in iron

Takishima, 2024-05-25:
Unfortunately no. I think that my theory is wrong after looking into it some more...

The one key difference I can see in the linker commands between normal and Nix build is that Nix does not have `-Wl,--as-needed` which means a lot of the dependencies stay explicit, which in turns make the linker load more than one shared library when loading the plugins.

Takishima, 2024-05-25:
No clue where that comes from yet though.

Thieso, 2024-05-25:
Hm okay, sadly this is a bit too deep for my CS knowledge, so I cannot help you there.

nilp-dromeda, 2024-05-25:
I am actively looking into this issue as well. Will keep you guys updated with any breakthrough.

Takishima, 2024-05-25:
Outside of Nix you get: 
```
patchelf --print-needed libmoveit_planning_scene_interface.so.2.5.5 
libmoveit_msgs__rosidl_typesupport_cpp.so
librclcpp.so
librcl.so
librmw.so
librcutils.so
libtracetools.so
libstdc++.so.6
libgcc_s.so.1
libc.so.6
ld-linux-x86-64.so.2
```

While inside we get:

```
patchelf --print-needed /nix/store/XXX/lib/libmoveit_planning_scene_interface.so.2.5.5
libmoveit_move_group_default_capabilities.so.2.5.5
libmoveit_move_group_capabilities_base.so.2.5.5
libstd_srvs__rosidl_typesupport_introspection_c.so
libstd_srvs__rosidl_typesupport_fastrtps_c.so
libstd_srvs__rosidl_typesupport_introspection_cpp.so
libstd_srvs__rosidl_typesupport_fastrtps_cpp.so
libstd_srvs__rosidl_typesupport_cpp.so
libstd_srvs__rosidl_generator_py.so
libstd_srvs__rosidl_typesupport_c.so
libstd_srvs__rosidl_generator_c.so
libstatic_transform_broadcaster_node.so
libmoveit_constraint_sampler_manager_loader.so.2.5.5
libmoveit_plan_execution.so.2.5.5
libmoveit_default_planning_request_adapter_plugins.so.2.5.5
libmoveit_cpp.so.2.5.5
libmoveit_planning_pipeline.so.2.5.5
libmoveit_trajectory_execution_manager.so.2.5.5
libmoveit_planning_scene_monitor.so.2.5.5
libmoveit_ros_occupancy_map_monitor.so.2.5.5
libmoveit_robot_model_loader.so.2.5.5
libmoveit_kinematics_plugin_loader.so.2.5.5
libmoveit_rdf_loader.so.2.5.5
libmoveit_collision_plugin_loader.so.2.5.5
libcollision_detector_bullet_plugin.so.2.5.5
libmoveit_butterworth_filter.so.2.5.5
librclcpp_lifecycle.so
librcl_lifecycle.so
liblifecycle_msgs__rosidl_typesupport_introspection_c.so
liblifecycle_msgs__rosidl_typesupport_fastrtps_c.so
liblifecycle_msgs__rosidl_typesupport_introspection_cpp.so
liblifecycle_msgs__rosidl_typesupport_fastrtps_cpp.so
liblifecycle_msgs__rosidl_typesupport_cpp.so
liblifecycle_msgs__rosidl_generator_py.so
liblifecycle_msgs__rosidl_typesupport_c.so
liblifecycle_msgs__rosidl_generator_c.so
librsl.so
libmoveit_collision_distance_field.so.2.5.5
libmoveit_collision_detection_bullet.so.2.5.5
libBulletDynamics.so.3.25
libBulletCollision.so.3.25
libLinearMath.so.3.25
libBulletSoftBody.so.3.25
libmoveit_dynamics_solver.so.2.5.5
libkdl_parser.so
libmoveit_constraint_samplers.so.2.5.5
libmoveit_distance_field.so.2.5.5
libmoveit_kinematics_metrics.so.2.5.5
libmoveit_planning_interface.so.2.5.5
libmoveit_planning_request_adapter.so.2.5.5
libmoveit_planning_scene.so.2.5.5
libmoveit_kinematic_constraints.so.2.5.5
[... 182 lines cut ...]
librosidl_typesupport_introspection_cpp.so
librosidl_typesupport_introspection_c.so
libstatistics_msgs__rosidl_typesupport_cpp.so
libbuiltin_interfaces__rosidl_typesupport_cpp.so
librosidl_typesupport_cpp.so
libstatistics_msgs__rosidl_generator_py.so
libbuiltin_interfaces__rosidl_generator_py.so
libpython3.11.so.1.0
libstatistics_msgs__rosidl_typesupport_c.so
libbuiltin_interfaces__rosidl_typesupport_c.so
libstatistics_msgs__rosidl_generator_c.so
libbuiltin_interfaces__rosidl_generator_c.so
librosidl_typesupport_c.so
librosidl_runtime_c.so
librcpputils.so
librcutils.so
libdl.so.2
libatomic.so.1
libtracetools.so
libstdc++.so.6
libm.so.6
libgcc_s.so.1
libc.so.6
ld-linux-x86-64.so.2
```

nilp-dromeda, 2024-05-27:
> Cool, thanks for looking into this :) So turning off `NIX_HARDENING_ENABLE` fixes the issue? I am a bit confused that it works in humble for me but not in iron

I am in humble, and it does not work for me? How were you able to get it to function on humble?

nilp-dromeda, 2024-05-27:
I am obtaining this error:

`[ros2_control_node-5] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class mock_components::GenericSystem. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[ros2_control_node-5]          at line 253 in /nix/store/QJKWYVARM2R71DKBM2L5YRVH6P7F3FVR-ros-humble-class-loader-2.2.0-r3/include/class_loader/class_loader/class_loader_core.hpp`

Is any one also obtaining this error?

nilp-dromeda, 2024-05-27:
If you run the following command:

```console
LD_PRELOAD=/nix/store/xxx/lib/libmoveit_move_group_default_capabilities.so.2.5.5 nixGL ros2 launch <robot-moveit-config> demo.launch.py
```

The simulation launches and finds all the required capabilties. So, it looks like the shared object files are not in the correct place or the symlinks to the shared object files for the move_group_default_capabilities.so are incorrect.

Takishima, 2024-05-27:
I think the issue is that the `moveit_ros_move_group` is added to `ament_target_dependencies` calls in places where it is not necessary.
Particularly in:
- `moveit-ros-planning-interface`: particularly in `planning_scene_interface/CMakeLists.txt` and `move_group_interface/CMakeLists.txt`
- `pilz-industrial-motion-planner`: in `CMakeListst.txt`

I think that this is not apparent outside of Nix, because some of the libraries MoveIt is linking to add `-Wl,--as-needed` to the linker command whereas the Nix derivations do not have that flag.

nilp-dromeda, 2024-05-27:
Were you able to apply these changes through `overridingAttrs` of the `propgatedBuildInputs` for `pilz-industrial-motion-planner` and `moveit-ros-planning-interface` packages (`move_group_interface` is not a package either, its just added as `subdirectory` call in the CMakeLists.txt file)? Or did you have to manually patch the CMakeList files?

Takishima, 2024-05-28:
I had to patch the CMake files since you need to change the calls to `ament_target_dependencies`

okvik, 2024-05-30:
Do you plan on upstreaming these patches by any chance? If not, posting them here would be greatly appreciated. Thanks.

Takishima, 2024-05-31:
As much as I'd like to post the patches here or upstream them right now, I first have to wait for the official permission from the company.
As soon as this process is done, I'll post them.

khaninger, 2024-08-08:
The LD_PRELOAD trick worked for me, but that was causing issues with other shell use.  Using patchelf to insert the dependencies worked for me; 
```
        moveit-ros-move-group = prev.moveit-ros-move-group.overrideAttrs (old: {
          postFixup = (old.postFixup or '''') + ''
            patchelf --add-needed libmoveit_move_group_default_capabilities.so $out/lib/$
            patchelf --add-needed libmoveit_move_group_default_capabilities.so $out/lib/$
          '';
        });
```

liarokapisv, 2024-11-03:
I am trying to understand the changes in order to fix something similar with pluginlib and composable nodes.
Why is adding the plugin library to the dependencies of more places than necessary an issue?