lopsided98/nix-ros-overlay #266: ros-noetic-teb-local-planner does not build

muellerbernd, 2023-06-02:
Hello. I am currently trying to convert my Archlinux ros noetic setup into the nix world. Everything works fine but I don't get the teb-local-planner to work. On Arch everything works fine.
When I build the teb-local-planner it crashes with the following error:
```bash
> CMake Error at cmake_modules/FindSUITESPARSE.cmake:131 (MESSAGE):
>   Unable to find SuiteSparse
> Call Stack (most recent call first):
>   CMakeLists.txt:35 (find_package)
```
As far as I know Suitesparse is a dependency of g2o, which gets build with the ros-noetic-libg2o package, which itself is already a dependency of teb-local-planner. 
You can  reproduce my setup with the config given in [my fork](https://github.com/muellerbernd/nix-ros-overlay/blob/staging/examples/teb-local-planner.nix) of this package.

wentasah, 2023-06-02:
Hi! There seems to be at least two problems. 

First, `teb-local-planner` doesn't declare dependency on `suitesparse` in `package.xml` and therefore, the dependency is missing also in `.nix` files. 

Second, even if the dependency is added manually to the .nix files, `cmake` in `teb-local-planner` does not find it. I guess this is because [`teb-local-planner` hardcodes locations](https://github.com/rst-tu-dortmund/teb_local_planner-release/blob/14705fe0c2709f8db83f2cfdd405f9f956ea0133/cmake_modules/FindSUITESPARSE.cmake#L60-L68) where to search for the library and this does not work for nix. I guess the referenced file would have to be patched to work with Nix. 

Unfortunately, I won't have time for it in the upcoming days.

muellerbernd, 2023-06-02:
@wentasah could you give me a hint on how this should look like?

wentasah, 2023-06-02:
Usually the `FindXXX.cmake` files allow to specify the location of the library via some variable, e.g., `SUITESPARSE_ROOT`. It doesn't work here, because the file uses `FIND_PATH` instead of `FIND_LIBRARY` to find the libraries. But I've found a hack to find it even without modification:
```diff
diff --git a/distros/noetic/teb-local-planner/default.nix b/distros/noetic/teb-local-planner/default.nix
index e61d2dc3a..37c1711b3 100644
--- a/distros/noetic/teb-local-planner/default.nix
+++ b/distros/noetic/teb-local-planner/default.nix
@@ -4,3 +4,3 @@
 
-{ lib, buildRosPackage, fetchurl, base-local-planner, catkin, cmake-modules, costmap-2d, costmap-converter, dynamic-reconfigure, geometry-msgs, interactive-markers, libg2o, mbf-costmap-core, mbf-msgs, message-generation, message-runtime, nav-core, nav-msgs, pluginlib, roscpp, std-msgs, tf2, tf2-eigen, tf2-geometry-msgs, tf2-ros, visualization-msgs }:
+{ lib, buildRosPackage, fetchurl, base-local-planner, catkin, cmake-modules, costmap-2d, costmap-converter, dynamic-reconfigure, geometry-msgs, interactive-markers, libg2o, mbf-costmap-core, mbf-msgs, message-generation, message-runtime, nav-core, nav-msgs, pluginlib, roscpp, std-msgs, tf2, tf2-eigen, tf2-geometry-msgs, tf2-ros, visualization-msgs, suitesparse }:
 buildRosPackage {
@@ -16,5 +16,8 @@ buildRosPackage {
   buildType = "catkin";
-  buildInputs = [ catkin cmake-modules message-generation tf2-eigen tf2-geometry-msgs ];
+  buildInputs = [ catkin cmake-modules message-generation tf2-eigen tf2-geometry-msgs suitesparse ];
   propagatedBuildInputs = [ base-local-planner costmap-2d costmap-converter dynamic-reconfigure geometry-msgs interactive-markers libg2o mbf-costmap-core mbf-msgs message-runtime nav-core nav-msgs pluginlib roscpp std-msgs tf2 tf2-ros visualization-msgs ];
   nativeBuildInputs = [ catkin ];
+  cmakeFlags = [
+    "-DCMAKE_INCLUDE_PATH=${suitesparse}/lib"
+  ];
 
```
If you do this, you end up with another problem:
```
-- Searching for g2o ...
-- Found g2o headers in: /nix/store/9dwfgpbqh35whq5fpjx9d64lk21hp61p-ros-noetic-libg2o-2020.5.3-r1/include/g2o
CMake Error at cmake_modules/FindG2O.cmake:85 (message):
  Could not find libg2o!
Call Stack (most recent call first):
  CMakeLists.txt:36 (find_package)
```
g2o is available, as can be seen from found headers, but two if its several libraries are missing.

wentasah, 2023-06-02:
The above problem can be solved by adding `openblas` as `libg2o` build input:
```diff
diff --git a/distros/noetic/libg2o/default.nix b/distros/noetic/libg2o/default.nix
index 03ab711b1..8ff4b572d 100644
--- a/distros/noetic/libg2o/default.nix
+++ b/distros/noetic/libg2o/default.nix
@@ -4,3 +4,3 @@
 
-{ lib, buildRosPackage, fetchurl, boost, catkin, cmake, eigen, libGL, libGLU, suitesparse }:
+{ lib, buildRosPackage, fetchurl, boost, catkin, cmake, eigen, libGL, libGLU, suitesparse, openblas }:
 buildRosPackage {
@@ -17,3 +17,3 @@ buildRosPackage {
   buildInputs = [ cmake ];
-  propagatedBuildInputs = [ boost catkin eigen libGL libGLU suitesparse ];
+  propagatedBuildInputs = [ boost catkin eigen libGL libGLU suitesparse openblas ];
   nativeBuildInputs = [ cmake ];
diff --git a/distros/noetic/teb-local-planner/default.nix b/distros/noetic/teb-local-planner/default.nix
index e61d2dc3a..37c1711b3 100644
--- a/distros/noetic/teb-local-planner/default.nix
+++ b/distros/noetic/teb-local-planner/default.nix
@@ -4,3 +4,3 @@
 
-{ lib, buildRosPackage, fetchurl, base-local-planner, catkin, cmake-modules, costmap-2d, costmap-converter, dynamic-reconfigure, geometry-msgs, interactive-markers, libg2o, mbf-costmap-core, mbf-msgs, message-generation, message-runtime, nav-core, nav-msgs, pluginlib, roscpp, std-msgs, tf2, tf2-eigen, tf2-geometry-msgs, tf2-ros, visualization-msgs }:
+{ lib, buildRosPackage, fetchurl, base-local-planner, catkin, cmake-modules, costmap-2d, costmap-converter, dynamic-reconfigure, geometry-msgs, interactive-markers, libg2o, mbf-costmap-core, mbf-msgs, message-generation, message-runtime, nav-core, nav-msgs, pluginlib, roscpp, std-msgs, tf2, tf2-eigen, tf2-geometry-msgs, tf2-ros, visualization-msgs, suitesparse }:
 buildRosPackage {
@@ -16,5 +16,8 @@ buildRosPackage {
   buildType = "catkin";
-  buildInputs = [ catkin cmake-modules message-generation tf2-eigen tf2-geometry-msgs ];
+  buildInputs = [ catkin cmake-modules message-generation tf2-eigen tf2-geometry-msgs suitesparse ];
   propagatedBuildInputs = [ base-local-planner costmap-2d costmap-converter dynamic-reconfigure geometry-msgs interactive-markers libg2o mbf-costmap-core mbf-msgs message-runtime nav-core nav-msgs pluginlib roscpp std-msgs tf2 tf2-ros visualization-msgs ];
   nativeBuildInputs = [ catkin ];
+  cmakeFlags = [
+    "-DCMAKE_INCLUDE_PATH=${suitesparse}/lib"
+  ];
 ```
Then it compiles, but the final link fails with:
```
/nix/store/22p5nv7fbxhm06mfkwwnibv1nsz06x4b-binutils-2.40/bin/ld: devel/lib/libteb_local_planner.so: undefined reference to `int boost::math::sign<double>(double const&)'
```

muellerbernd, 2023-06-02:
nice thanks. For that problem I already have a patch because I had the same problem on arch. 
```patch
diff --color -Naur teb_local_planner-release-release-noetic-teb_local_planner/include/teb_local_planner/h_signature.h teb_local_planner-release-release-noetic-teb_local_planner_new/include/teb_local_planner/h_signature.h
--- A/include/teb_local_planner/h_signature.h   2020-05-29 18:12:46.000000000 +0200
+++ B/include/teb_local_planner/h_signature.h   2023-01-20 19:56:56.259209108 +0100
@@ -51,6 +51,7 @@
 #include <functional>
 #include <vector>
 #include <iterator>
+#include <boost/math/special_functions/sign.hpp>

 namespace teb_local_planner

```

muellerbernd, 2023-06-02:
It now compiles and everything seems to work fine. I am running a Simulation and the simulated robot uses teb as local planner. Pathplanning works and the interaction with move_base also works. Huge thanks @wentasah

kjeremy, 2023-06-05:
Can you file an issue upstream at https://github.com/rst-tu-dortmund/teb_local_planner